kizniche/Mycodo #1: Sensor calibration

kizniche, 2015-08-18:


boomstick8x, 2015-09-29:
Its a very good idea, cause my two dht22 sensors are really different. Hummidity difference ~4% and temperature ~1C, or maybe it normal?

kizniche, 2015-09-29:
I have two sensors like this as well. This isn't high on the priority list though, so I'm not sure when I'll get a chance to research calibration methods.

Myke974, 2015-09-29:
@boomstick8x I'll use the lower RH one for PID relay control.

boomstick8x, 2015-09-29:
@kizniche callibration methods? Hmm, interesting...
Im thinking about taking weather alcohol or usual mercury thermometer and make some tests to find the error of each sensor.
I think calibration is just possibility to add or subtract temperature error for each sensor in Mycodo software. And for RH calibration I can use psychrometer (dont have one yet, but its easy to use two thermometers with some DIY)

@Myke974 You are using lower RH because of smaller saturate chance?

kizniche, 2015-09-29:
The standard method is to get the environment condition to a known low and a high, so you can determine the offset at both the low and high of the measurement spectrum. This will allow a linear regression to be performed. If it's not a linear change, a more complex function will need to be calculated.

Myke974, 2015-09-29:
[image]

closed 2017-05-24 by kizniche