kizniche/Mycodo #2: Email notification

kizniche, 2015-08-18:
During a critical failure, event, or measured condition.

kizniche, 2015-09-20:
Feature completed in commits https://github.com/kizniche/Mycodo/commit/6547b18204125ea6508e09e8bd35281bcc8d2b5a https://github.com/kizniche/Mycodo/commit/e026370fd0693bd3594f53fd3c33d9a89a4a9aad

kizniche, 2015-09-30:
Reopening because the Daily Max (maximum number of notifications that can be sent in a single day) and Wait Time (how long to wait between sending the same notification) settings still need to be implemented. Currently, a notification will go out whenever the conditional is true and has a trigger to notify, which can cause a flood of emails (depending on how often the conditional is checked).

boomstick8x, 2015-10-06:
Hi Kyle, sorry for asking question here, but I have question about email notification.
Under the `Settings` tab in `Email Notification` part in `User` line I must write my email address and my password in `Password` line, am I right?
I have a problem, the gmail block signing in and telling me that someone `untrue` try to sign in in my account. Did you faced the same problem?

kizniche, 2015-10-06:
Depending on your smtp server, he user will either be the user name or the email address. In most cases it should be the whole email address. And the password is the password for that account.

I've only used gmail for the email notification, but I've never had gmail give me any notification about anything. I've heard some have had to authorize sending from remote programs (Mycodo) but I haven't experienced this.

boomstick8x, 2015-10-07:
Thanks for reply, Kyle. I will try to win gmail in this battle:)

kizniche, 2015-10-20:
Notification wait time implemented for HT sensors in https://github.com/kizniche/Mycodo/commit/f1e19bea40e1e540af19380472fd9ebb598f39f3

Once I copy this method to the other sensors and relay conditionals, the notification system will be fully operational. I'll implement Daily_Max in the future, as it's not nearly as important as the wait time and will be harder to implement.

kizniche, 2015-10-20:
Finished implementing wait time for sensor and relay conditionals. Daily_Max will be implemented in the future, but the main functionality is complete.

closed 2015-10-20 by kizniche