kizniche/Mycodo #3: Enhancement

Eville3357, 2015-08-18:
enable bright PI light functionality with camera.

https://www.pi-supply.com/bright-pi-v1-0-code-examples/

kizniche, 2015-08-18:
Feature added in [v3.5.68](https://github.com/kizniche/Mycodo/commit/c7425e745fed1bc9e4cb092ed456fd7b1dc7282b)

closed 2015-08-18 by kizniche

Eville3357, 2015-08-18:
The feature is not working with streaming or still pics.  Used commands sudo i2cset -y 1 0x70 0x00 0x5a to activate white light and sudo i2cset -y 1 0x70 0x00 0x00 to turn lights off.

kizniche, 2015-08-18:
The web server uses user www-data with shell_exec() to run the pre- and post-commands. Since www-data does not have sudo access, sudo can't be used (unless you give www-data sudo acces, which I don't recommend). However, /dev/i2c-1 has group access by the group i2c:

```
$ ls -la /dev/i2c-1
crw-rw---- 1 root i2c 89, 1 Aug 17 14:25 /dev/i2c-1
```

So, if the user www-data is added to the i2c group, it will have read/write access to /dev/i2c-1. I added www-data to the i2c group with the following command:

`sudo usermod -a -G i2c www-data`

Also, the full path needs to be entered, so the following command should work after adding www-data to the i2c group.

`/usr/sbin/i2cset -y 1 0x70 0x00 0xa5`