kizniche/Mycodo #6: Add time/date timers

kizniche, 2015-09-02:
Turn relay x [on/off] at xx:xx for x seconds (0 to remain on/off)

[selection] repeat x times
[selection] repeat every x [seconds] thereafter)

MisterMeeseeks, 2015-11-23:
This would be phenomenal... I can't get the timers to work, so doing timers based on clock times would be great. Something like "Turn Relay ___ on at ___ and off at ___".

Myke974, 2015-11-23:
"Turn Relay ___ ON only between ___ and ____"

kizniche, 2015-11-23:
What's difficult about implementing this is that it must only trigger once for each time, and if there's a delay in polling the time, it may pass without actually triggering. There needs to be a way to verify that it was triggered and only trigger it once.

kizniche, 2015-11-23:
If time is past On_Time, activate Timer_Relay and set On_Trigger = True and Off_Trigger = False
If time is past Off_Time, deactivate Timer_Relay, set Off_Trigger = True and On_Trigger = False

kizniche, 2016-02-01:
half complete with commit https://github.com/kizniche/Mycodo/commit/3372814c599ba64f1620db8d7a69f9938904adad

Still need to add functionality to execute every x days. 1 day would be every day, 2 days would be every other day, etc.

closed 2016-02-01 by kizniche

MountAthos, 2016-02-02:
Merged revisions and will test now.
If tests are all ok here, then i will request to merge to main repository
and you can decide if you want to have all changes made for SHT support and
some other things related to it.

I will start to work on UI (and then will offer it too)
I have forked to my account the MyCoDo repository