kizniche/Mycodo #7: Enhancement (small screen / terminal support)

Myke974, 2015-09-02:
[image]
I have a small TFT screen, connected to the Pi 
the objectif is to be able to see mycodo/gaphs and at same time you can look throuh the glass inside the fruting chamber.

this base screeen  no tactlie , just cheap TFT is only 7''
and the webpage won't fit ! I tryed to reduce the wepage size in chromium with CTRL + +
 but all became too small/tight
Ideal, I think, would be to have some graph from terminal directly
I'm sure this is easy to read the sensors log file in real time to show a graph in a terminal window

boomstick8x, 2015-09-18:
Just thinking around, is your display resolution 480 х 234?
Could you try this?
sudo apt-get install fbi
and than
fbi -d /dev/fb0 -a /var/www/mycodo/images/*.png
Does it works? I can't try by myself because don`t have any display for RPi now.

kizniche, 2016-07-02:
Latest version supports small screens.

Automatic login -> autostart firefox in fullscreen mode

closed 2016-07-02 by kizniche