kizniche/Mycodo #18: favicon for easy searching in browser tabs

boomstick8x, 2015-09-18:
Few times I run with my eyes more than usually to find Mycodo tab in browser. I can try to make favicon for the project. May be you have an ideas about graphics?

p.s. What Mycodo means? I think first part 'Myco' shortcut from mycology, but the last 'do' I don't understand?

[image]

kizniche, 2015-09-18:
Nice idea. I've added a fav icon.

Back in 2009, when this was just a command line program, I wanted to have myco (fungus, mycology) in the the name. I just came up with Mycodo (like sudo), and I just never changed it. There's not much meaning beyond the myco part, but it's grown on me since then.

closed 2015-09-18 by kizniche

boomstick8x, 2015-09-18:
Yep! I see favicon! Thank you!

Can I ask you a question here about graphs?
What is green line on graph named DP? Digitall Pressure?
(If so why green line generating on DHT22 sensor?)

Myke974, 2015-09-18:
Green Line is the Dew Point ! :D
https://en.wikipedia.org/wiki/Dew_point

boomstick8x, 2015-09-18:
Ops... Digital Pressure... What I am thinking about.. :`D
Thanks!