kizniche/Mycodo #22: user guest with no password or show password on index.php or public graph page/tab

Myke974, 2015-09-23:
Turn ON or OFF public/anonymous direct access to the graph/tab page : perhaps a special graph page with a link to, from the login/index.php page.
Turn ON or OFF password showing for user guest in the main login page : index.php
Turn ON or OFF password need for user guest  in the main login page : index.php

kizniche, 2015-09-23:
Currently, to disable guest access, you can delete the guest user or change the password.

I'm not sure what you mean with the second comment. Do you want the guest password visible? This will not be possible, as the password is saved as a hash, and it can't be converted back to the actual password (it may be, but it's not meant to be and it wouldn't be easy).

Does your third comment mean to disable a guest password entirely, so all you have to do is enter the user name guest to log in?

Myke974, 2015-09-23:
OK

Turn ON or OFF password showing for user guest . ; yes ! make the password visible or affect some default visible password : by default I mean the default user guest with default password would be writed on the main page.

Yes : disable a guest password entirely or just show the default password somewhere on the page.

kizniche, 2015-09-23:
Well, when the database is created, you're prompted to enable or disable the guest login. If you choose to enable it, it asks for a password. So, there's no real mystery of what the password is unless you forget what you set. I did this so there is never a default password. Default passwords are the bane of the security professionals (unless you are working strictly on a LAN).

So, it's not possible to display the password, as it's stored as a hash. This is for the user's protection, because if the security of the password file is compromised, the perpetrator cannot get the password, only the hash.

Myke974, 2015-09-23:
OK then what about a dialog box where admin can decide to write the guest password in and this $word  will be shown/writed on the main page:
"If you want to visit : use user guest with password $word."

kizniche, 2015-09-23:
Ah, like a MOTD, but before login, not after. I can add a setting to add a custom message at the login page.

kizniche, 2015-09-23:
Added in commit https://github.com/kizniche/Mycodo/commit/41bcdbdcb5512658a26f77bae63ff9931fb4f4fe

closed 2015-09-23 by kizniche