kizniche/Mycodo #24: Owner individualisation of mycodo

Myke974, 2015-09-24:
If I have multiples Raspb.pi with mycodo on each / multiples setup / multi chambers : it'll be great to be abble to personalise a bit to know immediatly on which one you are.
Perhaps a functionnality to upload a banniere with fixed valors that'll appear in see-through the header (please find example in attached picture)
[image]
I used 50% opacity.
Perhaps a small logo at right but I'm totally aware when you add sensors with logs ticked they 'll need this place !

kizniche, 2015-09-24:
I could have a setting to header text, and I could bring back gravatar support, which would display a different image based on the email of the user logged in.

kizniche, 2016-10-08:
Fixed with commit 3b1a448

closed 2016-10-08 by kizniche