kizniche/Mycodo #25: Improve backup/update process

kizniche, 2015-09-24:
Potential features:
Choose what to back up
Choose which commit to update to
Choose which commit to roll back to

kizniche, 2015-09-25:
Testing by @boomstick8x:

First, I checked diff/patch. It was used on the og formatting I was mentioned earlier, it looks I got all pieces resolved after a few bumps. The main thing was to find a proper replacement for the explode function. I attach here the diff file for you to see what it was changed, and the resulting relay log file. I only could check the relay and HT-sensor logs for correctness. The diff was made from the home folder, based on a backup:

diff -Naur --exclude="log" --exclude="config" --exclude="*.pyc" --exclude="img" Mycodo-backups/Mycodo-2015-09-24_13-33-22/3.5 Mycodo/3.5 > test.patch

and checked with

patch -p0 --dry-run < test.patch

end patched with

patch -p0 < test.patch

This diff/patch should work on the whole Mycodo directory, but custom changes would be difficult to follow.
After this, I googled the following page:

http://stackoverflow.com/questions/10414769/git-pull-keeping-local-changes

This looked easy to follow, so I tried it. In fact, after a first try, the real run took about 20 minutes...
Here is the sequence of commands I used from the terminal (somehow my .bash_history file was not updated, but anyway, the whole terminal output is attached):

root@chz:/home/pi/Mycodo# git  log --oneline
root@chz:/home/pi/Mycodo# git reset --hard 4024e62
root@chz:/home/pi/Mycodo# /etc/init.d/mycodo restart
root@chz:/home/pi/Mycodo# cd ..
root@chz:/home/pi# patch -p0 --dry-run < test.patch
root@chz:/home/pi# patch -p0  < test.patch
root@chz:/home/pi# /etc/init.d/mycodo restart
root@chz:/home/pi# cd Mycodo
root@chz:/home/pi/Mycodo# git status
root@chz:/home/pi/Mycodo# git stash
root@chz:/home/pi/Mycodo# git pull
root@chz:/home/pi/Mycodo# /etc/init.d/mycodo restart
root@chz:/home/pi/Mycodo# git stash pop
root@chz:/home/pi/Mycodo# /etc/init.d/mycodo restart

In short:
I started from my patched version of log files
then first reverted 2 commits back
then  applied the patch
then stashed the modified files
then pulled the newest version from github (coincidentally you just pushed an update in this small timeframe, I am worried about the database update, but everything still looks working, I'll do a proper update soon).
then reapplied my modified files.

The log formatting shows this nicely.

As for the conflict changes, I had none in this case. I suppose to be able to automate it, some ideas are needed, but I think regular users face no problem, and whoever applies custom code, should be at least knowledgeable enough to resolve it manually if the update process gives him a flag.

boomstick8x, 2015-09-25:
I feel a little lame here. I don`t have 'Mycodo-backups' folder at my home directory and in case of this think your diff/patch command doesn`t work properly. Am I wrong?

kizniche, 2015-09-25:
I assume you're referring to mycodo-wrapper.c?

Either will work. The service command is the same as executing the init.d script. I chose to execute the init.d script directly to bypass the use of another program. The service command is just more convenient when working in the terminal. I also wanted to use the full path... the service command may not necessarily be in the same location across linux distributions, but the script will always reside in /etc/init.d/

boomstick8x, 2015-09-25:
Im asking about service command because we need to stop service while patching(I think so), but if we stop it(not restart), then I cant run it with `/etc/init.d/mycodo start`
p.s. or we doesnt need to stop process?

kizniche, 2015-09-25:
There generally isn't a problem with leaving the daemon running while patching/updating unless the database changes. The daemon is running in memory, so even the daemon (mycodo.py) can be changed without repercussions.

However, if the database is being rewritten (updated) when the daemon is trying to read it, there could be unanticipated consequences (no idea what could happen, but worst case would be the daemon crashing or relays turning on or off when they're not supposed to). To avoid this, the update script stops the daemon before the update and starts it after it finishes.

kizniche, 2015-09-25:
What do you mean by "I cant run it with /etc/init.d/mycodo start"

boomstick8x, 2015-09-25:
I mean that it will start not in background process
----------
this I get after patching
```
root@mycodo:/home/pi# /etc/init.d/mycodo restart
/etc/init.d/mycodo: line 146: /var/www/mycodo/cgi-bin/mycodo-client.py: No such file or directory
Checking if daemon has stopped (please wait 10 seconds)..........root     10450  2.7  1.4  23772 12488 pts/1    Sl   21:11   0:01 python /var/www/mycodo/cgi-bin/mycodo.py
 Daemon is still running. Attempting to stop... mycodo.py (pid 10450) has been killed after 1 seconds.
Because daemon had to be killed, turning off all GPIOs
/etc/init.d/mycodo: line 197: /var/www/mycodo/cgi-bin/GPIO-initialize.py: No such file or directory
chown: cannot access `/var/www/mycodo/cgi-bin/mycodo-wrapper': No such file or directory
chmod: cannot access `/var/www/mycodo/cgi-bin/mycodo-wrapper': No such file or directory
```

kizniche, 2015-09-25:
I have no idea why you are getting those errors. The files should exist. I think I will need to know first why those files don't exist to fix the problem.

boomstick8x, 2015-09-25:
Ok, I will investigate this.

boomstick8x, 2015-09-25:
It because my HEAD was somewhere unknown, and after `git reset --hard origin/master` all files come to their places.

kizniche, 2015-09-25:
I'm not sure I'm following you. Is it moving the files? Deleting them and creating them elsewhere?

boomstick8x, 2015-09-25:
I really dont know why this happens, but after `git fetch --all` and `git reset --hard origin/master` they are on their places, I will try to find the moment when they are gone(will try whole update process few times more with hope this will helpful)
---
I think  it`s not about update, but when I am restarting daemon with `/etc/init.d/mycodo restart` and the daemon is not running I got this errors:
```
Traceback (most recent call last):
  File "/var/www/mycodo/cgi-bin/mycodo-client.py", line 253, in <module>
    menu()
  File "/var/www/mycodo/cgi-bin/mycodo-client.py", line 85, in menu
    c = rpyc.connect("localhost", 18812)
  File "/usr/local/lib/python2.7/dist-packages/rpyc/utils/factory.py", line 89, in connect
    s = SocketStream.connect(host, port, ipv6 = ipv6, keepalive = keepalive)
  File "/usr/local/lib/python2.7/dist-packages/rpyc/core/stream.py", line 132, in connect
    return cls(cls._connect(host, port, **kwargs))
  File "/usr/local/lib/python2.7/dist-packages/rpyc/core/stream.py", line 102, in _connect
    s.connect(sockaddr)
  File "/usr/lib/python2.7/socket.py", line 224, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 111] Connection refused
Checking if daemon has stopped (please wait 10 seconds).......... No daemon running.
```

kizniche, 2015-09-25:
That error is normal. If the daemon is not running, the connection is refused because the threaded communication server of the daemon is not alive to accept the connection.

kizniche, 2015-09-25:
mycodo-client.py is attempting to make a connection with the daemon to instruct it to shut down, but because the daemon is not running, it cannot connect.

boomstick8x, 2015-09-25:
Is it okay that I dont have directory Mycodo-backups/Mycodo-2015-09-24_13-33-22/3.5?

kizniche, 2015-09-25:
I recently changed the backup process. It should now back up the contents of 3.5/ to Mycodo-backups/Mycodo-2015-09-24_13-33-22/ and there shouldn't be a 3.5 directory. This was done so there would [hopefully] be less confusion about what the backup is... which is only the contents of /var/www/mycodo/

kizniche, 2015-09-25:
Oops. I said that @boomstick8x was the person who provided the information in https://github.com/kizniche/Mycodo/issues/25#issuecomment-143227398, however it was @zsole2. I've updated the comment with the correct attribution. Sorry for the confusion.

boomstick8x, 2015-09-25:
1. move head at '4024e62 remove display of password in settings (insecure)'
2. run backup from web update
3. make patch with:
`diff -Naur --exclude="log" --exclude="config" --exclude="*.pyc" --exclude="img" Mycodo-backups/Mycodo-XXX-XX-XX_XX-XX-XX/3.5 Mycodo/3.5 > test.patch`
Agree with all `Reversed (or previously applied) patch detected!  Assume -R? [n]`
4. git stash
5. git pull
6. /etc/init.d/mycodo restart
7. git stash pop
8. /etc/init.d/mycodo restart
9. and finally i got `The log formatting shows this nicely' :)

Thank you for confusion, now I know a little more about git :)

kizniche, 2015-09-25:
To my understanding this process is supposed to allow updating to the latest Mycodo version as well as add custom code from the previous version to the updated version.

How does the patch know what is custom code and what is actually being updated?

kizniche, 2015-09-25:
Sorry about all that, even though you know a bit more. I can see now that it seemed like I was requesting you to investigate this. I was merely trying to put the information here that @zsole2 provided in an email form his efforts to preserve custom code after an update has happened (or is this process to revert back to a previous state?).

boomstick8x, 2015-09-25:
Im using git only few days, but like you write in testing comment, we do:
1. `git stash` - to save uncommited changes to stack
2. `git pull` - do update to newest version
3. `git stash apply` - to apply our last saved changes to updated version(`git stash pop` will also delete changes from stack after applying)
4. check all is on right places
5. `git stash drop` - to drop last saved stash (if we dont use `git stash pop` before)

**making patch file with git:**
0. `git stash` - to stash our local changes
1. `git stash show -p > name.patch` - to stash output in a patch file
add `--binary` option after `-p` if we also need to stash binary files `git stash show -p --binary > name.patch`)
2. `git apply --stat name.patch` - to view all looks good
3. `git apply --check name.patch` - to verify no errors
4. `git apply name.patch` - to apply our patch

p.s. `git stash list` - to see our stack

kizniche, 2015-09-25:
@boomstick8x, Thanks for that explanation and for the work to figure it out!

I'll probably add this to the manual after I confirm everything works as expected.

I'm hesitant to add it as an automated feature because it's such a niche area where someone would be adding custom code. There's also the possibility of conflicts if the patch tries to change something that was already updated or if the custom code breaks the system, so I don't want someone accidentally using this option without knowing fully what it's for and the risks involved.

boomstick8x, 2015-09-25:
Dont know about helpful of my work to your project, but for me it was very interesting, I like to learn something new.
If you except that I could somehow help to project I am waiting and always ready :)

zsole2, 2015-09-26:
Guys,
Isn't the purpose of these project to learn and accomplish something new and useful?
Personally I did not know about git anything before Mycodo, so already learned a lot. My whole investigation was for finding out why I could not copy back a backup and go forward from there.

Now with all this info, an automated restore is feasible, at least for a regular user where no custom code is used. In this case, simple way reverting back to a working state would be helpful. As in my case was, some bug prevented the humidity PID working, leading me to wanting to restore a backup. 

The update script can be modified to include a restore function, and if 'git status' shows any sign of modified files, the script would stop and inform the (probably advanced) user to do it by hand, for which the manual could show a simple guideline, as @boomstick8x summarized quite nicely. The regular user would not notice anything, other than his install went back to an earlier working state, and he can try the update later, when the bug possibly fixed.

kizniche, 2015-09-27:
I agree that the function should exist, but I want to make sure that it's foolproof in it's implementation.

For reverting to a previous state, `git reset --hard x`, where x is the commit ID, should reset everything to that commit. If I incorporate the commit ID in the backup folder title, it would be easy to identify which databases should be restored after the `git reset`. This would be very easy to implement and there's a reasonable assurance that it can't be incorrectly used. I could add a "Revert" button next to each commit ID (that there is a backup of) of the output of the "Git Commits" button in the Data tab.

I think the use of `git stash` should be reserved for advanced users and would be most appropriate in the manual.

zsole2, 2015-09-28:
I agree completely, on all points.

kizniche, 2015-09-28:
After mush testing, I think I have a working restore system in place, with commit https://github.com/kizniche/Mycodo/commit/5da6d738e5f9f209e0d164d3862e821dd2ddec90

Please test it if you are inclined and see if you can find any problems.

boomstick8x, 2015-09-29:
I testing near two hours, but...
Can I use `Update` button in `Settings Tab` for update to any new commit?

kizniche, 2015-09-29:
@boomstick8x Yes, and I would recommend always doing a restore from the latest version. I should put in the documentation that before doing a restore, make sure you run an update to have the latest code that carries out the restore.

boomstick8x, 2015-09-29:
@kizniche I cant understand... In commits Im at `5da6d73 v3.5.79: Add backup restore feature to revert back to a previously-backed up system and databases (found in 'Git Commits' of the Data tab)`, but when I press `Check for update` button it tells `You are running the latest version`.
I feel that I miss something...

kizniche, 2015-09-29:
You might have gone back to a broken version. You may have to do a fresh clone and copy your logs and databases back in to correct it.

boomstick8x, 2015-09-29:
I will create new, now just testing...
When i create fresh install, after login page, everytime I got an error `Error: Can't find/open /home/pi/Mycodo/3.5/log/auth.log`

kizniche, 2015-09-29:
Ah, I had thought that may happen a while ago, but I rarely do fresh clones anymore. I'll put a line in the init.d script to create it if it doesn't exist.

boomstick8x, 2015-09-29:
And even now, after fresh install I cant update :( Maybe it case of something like cloning repository url... but before everything worked... Dont know what to think.

boomstick8x, 2015-09-29:
And even now, after fresh install I cant update.
I think it because my copy of repository. It cant find update because my current version is already up to date to my repository.

zsole2, 2015-09-29:
After cloning anew, the backup/restore appears to be working. 
The restore log does not appear correctly ... but I saw your bug for that
Is there a specific reason for moving the backups directory to /var ? 
What if someone is so behind that the current commit is not visible, therefore the user does not see the red highlight and gets confused? Maybe you should show at least that many lines from git log...
In a related suggestion, it would be convenient if the Lines input-box could remember the previous number of lines, defaulting to 30, as in the code now.

I'll try to test more later.

kizniche, 2015-09-29:
I moved the backups to /var because it was too difficult to accurately determine what the correct user was where the backups were previously. As root is the user performing the backups, there is no good way to determine the correct user other than just searching all the user directories in /home. Having a standard location also alleviates a lot of other problems.

Currently, the commits will show the 30 (default) commits you are ahead and an unlimited number you are behind. So, the current commit (red) will always be visible. You \just made me realize I did not limit the display of the commits ahead of HEAD, but this may not be a bad thing, as you will always see ALL the commits ahead.

zsole2, 2015-09-30:
Good point on the user...
Also agree on the new commits, I did not look into the code just fired a question.
Thanks for implementing the Lines input.

kizniche, 2015-10-18:
Last feature of backup system in place ("create backup" button) with commit https://github.com/kizniche/Mycodo/commit/9f81b574342124d4c6204d44c20a0ccad7525260

I'll add the patch notes to the manual soon.

kizniche, 2015-10-28:
Finished adding stash/patch information to the manual. Closing this issue. Thanks for the work, guys.

closed 2015-10-28 by kizniche