kizniche/Mycodo #28: Add Notes/Journal entries Section

Myke974, 2015-09-26:
Wouldn't it be awesome to be abble to write some notes anytime to keep log of anything : 
- Grow journal
- PID settings/tuning
- personal notes and comments
Very important to have automatic time stamps affected when writing the notes/

kizniche, 2015-09-26:
How do you think this should be implemented?

Myke974, 2015-09-26:
I think its own tab should be a good choice good : like the data tabs, the entries could be separated into labels.
Some default labels editables by the users who got rights.

zsole2, 2015-09-26:
I fully support this!
Since the mycodo database for storing the parameters, it is practically no way to know looking back what was changed and when, and the result of this change was relevant or not. Now we only now from the daemon log that the database was reloaded at a certain time. 
I think a journal tab (or journal button on the Data tab) would be ideal to store the PID parameters with timestamps automatically, helping for tuning the PID loop looking back days/weeks later. 
Of course text box would allow to add any personal comments, too.

Myke974, 2015-09-27:
If we can add pictures too (I'm thinking Grow-log) !
The result would be a Web Page showing all the notes with timestamps and color to differentiate between labels and the possibility to hide/show labels.

kizniche, 2015-09-28:
I added the relay/sensor value history logging in commit https://github.com/kizniche/Mycodo/commit/6cb09a7f47664fbb2ce5adbce53e6d8982149b20

kizniche, 2015-10-01:
Considering how the sensor/relay change (delta) logs are set up, what do you think is the best way to implement a text logging system?

I was thinking about if all the sensor and relay change logs were combined, but they have different numbers of columns, so it would be difficult to determine what each value was. This would have worked well with the history feature, as you could see everything in relation to each other, along with text notes. But, because the columns pose a problem, the logs will have to remain separate.

Maybe a notes button on the Data tab, that will load a new page in the Data tab with an input field, a "new note" button, all notes with a timestamp, along with "edit" and "delete" buttons next to each entry. The edit button would overwrite the note with whatever is in the text box. A "new note" button will create a note from the text box and timestamp it. Delete will remove the note. What do you think?

zsole2, 2015-10-01:
Since my system is quite stable now, I did not check the operation of the change logs... I will soon. On the other hand, I have only HT sensors in my setup, so I don't know how the columns with other sensors match or not. I can see your problem with the relay change log, though... On this topic, shouldn't a timer change log is needed for completeness?
<OFF> 
I always  wanted to add a T sensor for the ambient temperature, I have the sensor already, but reluctant to add it,  because  in that case it will be the top graph, which is in my case would be data that is nice to have, but the least important. 
So when you will not have absolutely nothing to improve, I'd suggest to find a way to reorder the sensors/graphs. I know that it is probably a lot of work, so just saying as a future feature request. 
</OFF>
...
I just run a quick test on the change log headers to see the differences... I suggest to put a title on top of the table to know which data you see currently, I got lost a few times not knowing which button was clicked last. Just repeating the button name would be enough above the header...

So I copied all headers into a spreadsheet, and apparently there are lots of similar types of columns, just their name is different based on the sensor. With the  combo sensors you can put the specific data first (hum and press) and the temp as a secondary will tail the table. Other than that, the GPIO looks different. So if you  add a 'Type' column, the whole combination looks feasible. Of course, I guess it is again a lot of coding...

As for the text boxes, I'd leave to  Myke974, as he suggested it, but your approach looks sensible.

kizniche, 2015-10-01:
Good idea. I added a change log title above the column titles and added the timer change log in commit https://github.com/kizniche/Mycodo/commit/ff904d84d21614d22509b33edbba0741654917b9

kizniche, 2015-10-01:
@zsole2 You can check the Log checkbox and uncheck the Graph checkbox to have the sensor be recorded but not display the graph. It will also display the data in the header when it's not graphed so you can easily see the current temperature from that sensor and still have your desired graph at the top.

zsole2, 2015-10-02:
I know that I can turn off graphing, though I never tried it, since I have only two sensors currently, so it is not a problem scrolling a bit. 
Agree that  reordering of the graphs/sensors would be difficult, that's why said future request, if nothing else remained to work on. In any case, my idea would be to keep an ordered array of sensors, with their types, i.e. 
0 - HT
1 - T
2 - HT
3 - CO2
4 - T
and so on ...
Then rather than running separate loops for each sensor types, you would have an outside loop, enumerating the sensor-array, and based on the type of any given element, the appropriate sensor would be presented. For the graphs the same array could be used, except if the graphing is not select, it would be omitted.
As for programming it, that is way over my head...

kizniche, 2015-10-07:
By popular demand, note-taking is included in commit https://github.com/kizniche/Mycodo/commit/79ec05f7805b8703f6a42c075f2eedc07e617d51

This is just a basic implementation with a SQLite database, so don't expect much ;)

boomstick8x, 2015-10-08:
@kizniche would you please, when you will able to spend time for it, also add option for edit notes?

kizniche, 2015-10-08:
That will be the next update I make to the note system. Maybe I'll get that working this weekend.

Myke974, 2015-10-08:
WoW : that's great ! thanks !
It took me few days to show what I had in mind : please see attached picture.
[image]

kizniche, 2015-10-08:
Editable notes in https://github.com/kizniche/Mycodo/commit/a014ce6e751627ea6f52265deea270857566a0f1

Myke974, 2015-10-08:
:+1:  AWESOME ! !

boomstick8x, 2015-10-09:
Yeah! :+1:

kizniche, 2015-10-09:
file uploads in https://github.com/kizniche/Mycodo/commit/a0f2e57f79bf52540bb8d0b5bcd198e3398cdb6c

kizniche, 2015-10-10:
After much toiling with layouts and formatting, I believe I have everything with uploads working.

[image]

closed 2015-10-10 by kizniche

Myke974, 2015-10-10:
Excellent ! Thank you ! !

kizniche, 2015-10-10:
Even though I closed this issue, feel free to comment here for any suggestions for changes/features.

Myke974, 2015-10-10:
We need to be sure to never lose thoses notes ! with some auto-save somewhere  perhaps ?

kizniche, 2015-10-10:
How would they get lost?

I was going to add the ability to backup at any time, so that could be what you're looking for.