kizniche/Mycodo #31: Can't update after fresh install

kizniche, 2015-09-29:
@boomstick8x I'm starting this (and deleting the other posts) because it was off-topic for that thread (and there's a lot of chat, so I don't want who is subscribed to that issue to keep getting notified).

Is there an error of any kind?

kizniche, 2015-09-29:
@boomstick8x Copy what you have displayed when you press "Git Commits" button on the Data tab.

boomstick8x, 2015-09-29:
```
Current commit: 5da6d73
 
If a backup exists for a certain commit, a "Restore" button will appear beside the commit ID.
Restoring a backup will create a backup of the current commit.
Restoring a backup will restore the databases from the restored backup.
 
Commit  Description (The most recent commit is at the top)
 
5da6d73 v3.5.79: Add backup restore feature to revert back to a previously-backed up system and databases (found in 'Git Commits' of the Data tab)
4a77d43 Restore feature test (WARNING: DO NOT USE YET)
b81a558 Restore feature test (WARNING: DO NOT USE YET)
ecf8a8d Restore feature test (WARNING: DO NOT USE YET)
dee80b7 Restore feature test (WARNING: DO NOT USE YET)
f9ad8a8 Restore feature test (WARNING: DO NOT USE YET)
36ca435 Restore feature test (WARNING: DO NOT USE YET)
8af423a Restore feature test (WARNING: DO NOT USE YET)
59f6181 Restore feature test (WARNING: DO NOT USE YET)
1784c9c Restore feature test (WARNING: DO NOT USE YET)
8eed639 Restore feature test (WARNING: DO NOT USE YET)
cf749fa Add backup date to Restore button title (mouse hover text)
b43a45e Edit name/value of restore button
9ab609d Add framework for restoring backups (can view non-working button next to commits that have backups, from the 'Git Commits' button in the Data tab)
ca21be2 test of new backup directory
e8b9e51 Use /var/Mycodo-backups instead of home directory (too difficult to determine which user the backups are under when attempting to create restore feature
fab1ea5 include commit in backup directory name
7cff0e3 formatting changes
364e673 fix bug preventing saving of pressure sensor values
98073d8 bug fix for new history feature
c0ae28e critical bug fix
6cb09a7 v3.5.78: Record history of changes to sensor/relay values, reduce number of spaces between values in relay/sensor logs to one (reduce file size)
bc08442 layout tweaks
412dc2a Don't know how this didn't get included in the previous update (I recall adding it)
20bec88 Test update system
9929641 fix update checker
d1c4363 Test for new update notification system
fb75ff1 Add system update check every 24 hours (minimum, determined by how often the page is refreshed) and place persistant notification on settings page
457d84d add ability to view all git commits and descriptions ('Git Commits' in Data tab)
c79b37f reformat sensor/relay log header when printing logs
```

kizniche, 2015-09-29:
I don't think you are doing a fresh clone if this is the output you are receiving.

boomstick8x, 2015-09-29:
I manually hard reset to this commit for test auto `Backup feature` when updating to a new version

kizniche, 2015-09-29:
I'm not sure what that means. I would get a fresh clone by doing...

`sudo mv /home/user/Mycodo /home/user/Mycodo-old`
`sudo git clone https://github.com/kizniche/Mycodo.git /home/user/Mycodo`
`sudo cp /home/user/Mycodo-old/3.5/config/*.db /home/user/Mycodo/3.5/config/`
`sudo cp /home/user/Mycodo-old/3.5/log/*.log /home/user/Mycodo/3.5/log/`
`sudo cp /home/user/Mycodo/3.5/init.d/mycodo /etc/init.d/`
`sudo service mycodo initialize`
`sudo service start`

boomstick8x, 2015-09-29:
I think it`s all about restarting service... One second, I will try...

boomstick8x, 2015-09-29:
Yes! That was it!

kizniche, 2015-09-29:
How did that fix it?

boomstick8x, 2015-09-29:
I really dont know, but after I `service mycodo restart` in terminal I saw `A newer version is available`.

kizniche, 2015-09-29:
Okay... I'm not sure how that could have fixed it, but if it's working, great. I'll close this issue.

closed 2015-09-29 by kizniche

boomstick8x, 2015-09-29:
Ok, thank you!