kizniche/Mycodo #34: Humidity/Temperature sensors : Presets buttons "Save" & "New" Bug

Myke974, 2015-09-30:
when either of thoses 2 buttons are clicked the result is a white page

kizniche, 2015-09-30:
Did you put in a name?

kizniche, 2015-09-30:
Oh,  a white page. I'll try to reproduce this. What type of sensor?

kizniche, 2015-09-30:
Sounds like an SQL or PHP error.

kizniche, 2015-09-30:
[Wed Sep 30 17:08:42 2015] [error] [client 66.87.3.198] PHP Warning:  SQLite3::prepare(): Unable to prepare statement: 1, 49 values for 51 columns in /home/eville/Mycodo/3.5/includes/restricted.php on line 1218
[Wed Sep 30 17:08:42 2015] [error] [client 66.87.3.198] PHP Fatal error:  Call to a member function bindValue() on a non-object in /home/eville/Mycodo/3.5/includes/restricted.php on line 1219

kizniche, 2015-09-30:
Fixed https://github.com/kizniche/Mycodo/commit/f8127c22234256025baf87aa0dd8ba31f619aba0

closed 2015-09-30 by kizniche