kizniche/Mycodo #37: Ability to attach photos to Notes made with Camera Tab

boomstick8x, 2015-10-11:
I got an idea - ability to attach photos from RPi camera to notes(made with 'Camera tab')

kizniche, 2015-10-11:
It seems the easiest way to do this is add a button to the camera tab to create a note with the currently-displayed image. Being able to select a specific note is toublesome. If you need to add to a previously-created note, you can easily save the image, edit the note, and upload it. Also, saving multiple images would be difficult.

boomstick8x, 2015-10-11:
Yes, Im thought the same way about it, just add button to the Camera Tab, and I think its enough for that stage.
p.s. And maybe other options, you wrote about, will be great to see in future of 5.x version of Mycodo Project :)

kizniche, 2015-10-13:
I haven't quite figured out if it's just that I'm getting more proficient with Python/PHP or these coding tasks are getting easier ;)

Added note creation from camera still image in commit https://github.com/kizniche/Mycodo/commit/1c2ec681b3947eb118ae17e28e8d7557866b5b91

closed 2015-10-13 by kizniche

boomstick8x, 2015-10-14:
The one from another! ;) And of course you the boss here! +1:
p.s. Im a little jealous of you about upgrading your skills here, but also Im getting energy of understanding that I must go forward and work hard...
p.p.s sorry for my bad english :)