kizniche/Mycodo #43: AM2302 Sensor Support

hephaestoskin, 2015-11-03:
I have been using the AM2302 sensor on my projects for some time and it has proven very rugged, reliable and easily mounted in real application. Unlike other sensors, it works via i2c with a static address, so multiple sensors would require an i2c multiplexer. Those are widely available with drivers, but they almost all work the same way. It would be great if Mycodo would support these sensors.

kizniche, 2015-11-03:
I agree. However, a quick search didn't turn up anything related to the AM2302 (DHT22) and I2C. Do you have any additional resources beyond the datasheet? I have a few DHT22s, so it wouldn't be that difficult to get a multiplexer and begin testing in the near future.

hephaestoskin, 2015-11-03:
Oh dang, I meant the AM2315, sorry for the confusion. Drivers are part of tentacle_pi (github).

https://github.com/lexruee/tentacle_pi

kizniche, 2015-11-03:
Nice! A little pricey, but it has a few percent better accuracy in the extreme ends of the humidity range than the DHT22. I'll see about picking one up and a multiplexer and giving it a shot.

kizniche, 2015-11-12:
Got the sensor today. I should have the multiplexer soon.

hephaestoskin, 2015-11-12:
That's great, thanks!

Benjamin Morrison

kizniche, 2015-11-13:
Got the multiplexer today. Maybe Saturday I'll have some time to hook them up.

kizniche, 2015-11-13:
Do you have experience using this I2C multiplexer, either with i2c-tools or Python?

kizniche, 2015-11-13:
This looks like it includes a class for the TCA9548A: https://pypi.python.org/pypi/pymlab/

hephaestoskin, 2015-11-13:
No I'm afraid I don not have experience in this. I'm still just learning.

kizniche, 2015-11-14:
Well, the easy part of adding sensor support is done (https://github.com/kizniche/Mycodo/commit/c27e903bf82ba406d2aec7e38ac7aecd29113cc7), now I need to figure out how to use the multiplexer. I spent a good while tonight on it and I think I got to the point where I'm selecting the channel of the multiplexer, but I haven't been able to get a reading though the multiplexer.

kizniche, 2015-11-14:
I'm pretty sure I've figured out changing the channel of the multiplexer with the following code:

```
#!/usr/bin/python

# Change channel of TCA9548A
# Example: sudo ./multiplexer_channel.py 0

import smbus
import time
import sys

I2C_address = 0x70
I2C_bus_number = 1
I2C_ch_0 = 0b00000001
I2C_ch_1 = 0b00000010
I2C_ch_2 = 0b00000100
I2C_ch_3 = 0b00001000
I2C_ch_4 = 0b00010000
I2C_ch_5 = 0b00100000
I2C_ch_6 = 0b01000000
I2C_ch_7 = 0b10000000

def I2C_setup(i2c_channel_setup):
	bus = smbus.SMBus(I2C_bus_number)
	bus.write_byte(I2C_address,i2c_channel_setup)
	time.sleep(0.1)
	print "TCA9548A I2C channel status:", bin(bus.read_byte(I2C_address))

I2C_setup(int(sys.argv[1]))
```

But I still haven't been able to read the AM2315 while connected to the multiplexer.

hephaestoskin, 2015-11-14:
Hm, perhaps we should add a thread to the Adafruit forums so their staff can respond? It is their multiplexer. I wish I knew enough to help myself.

Benjamin Morrison

kizniche, 2015-11-14:
Okay. Creating a thread on Adafruit forums.

For completeness, here's the code I use to read the AM2315, form tentacle_pi drivers:

```
#!/usr/bin/python

import datetime
import time
from tentacle_pi.AM2315 import AM2315
am = AM2315(0x70,"/dev/i2c-1") # Hardcoded address 0x5c

def Timestamp():
    return datetime.datetime.fromtimestamp(time.time()).strftime('[%Y-%m-%d %H:%M:%S]')

for x in range(0,10):
    temperature, humidity, crc_check = am.sense()
    print "temperature: %s" % temperature
    print "humidity: %s" % humidity
    print "crc: %s" % crc_check
    time.sleep(2)

```

kizniche, 2015-11-14:
Posted to the Adafruit Forum: [Using the TCA9548A I2C Multiplexer with a Pi and AM2315 Temp](http://forums.adafruit.com/viewtopic.php?f=19&t=83918)

kizniche, 2015-11-18:
The I2C multiplexer and AM2315 are finally working in harmony! Thanks to adafruit_support_rick on the Adafruit support forum.

kizniche, 2015-11-20:
Up to 8 simultaneously-connected multiplexers supported with commit https://github.com/kizniche/Mycodo/commit/cb7c33c5a469f9348d109ea6dc3ed117f63fa665

closed 2015-11-20 by kizniche

mdobres, 2016-11-16:
Hi, I used your code above to change the channel of the multiplexer. It all seemed to work, However I assume your program just reports the status or number of the channel. How would you get it to report the I2c address of the attached device?