kizniche/Mycodo #50: Odd trigger behavior / things not logging / timer problems

MisterMeeseeks, 2015-11-23:
I'm trying to set up a makeshift timer for light. So what I did was set turn on relay 1 (normal lights) for 64800 seconds (18 hours). Then I set up a trigger so that when relay 1 turns off, relay 4 (night light) is supposed to turn on for 21600 seconds (6 hours). And another trigger so that when relay 4 turns off, relay 1 is supposed to turn on for 64800 seconds again... so that it does a cycle.

So relay 1 works fine and turns off, turning relay 4 on... but after that, it stops triggering the cycle. Relay 4 turns off, but doesn't write to the log and doesn't trigger relay 1 to turn back on. If I start the cycle from the relay 4 starting point, it does the same thing, except that it logs 4 turning off and turns on relay 1, but doesn't turn relay 1 off, write 1 turning off to the log, or trigger relay 4 turning on.

I'm not sure what the best way to do this is. I would love a time based trigger; so I could put "Turn on relay 1 at 8:00am for ___ hours" or "Turn on relay 1 at ___ and off at ___" would be even better.

kizniche, 2015-11-23:
Interesting. I'll look into it.

kizniche, 2015-11-23:
Aha. Found the error. When a relay was turning off, it would only turn the activation relay on, even though it maybe set to turn on for a certain duration. Fixed in https://github.com/kizniche/Mycodo/commit/9a272718fe5a6b2a5d9eb0d849578bdbd56a872e

closed 2015-11-23 by kizniche

MisterMeeseeks, 2015-11-23:
Looks like it works now. I will watch it for the next couple of days to be sure, but early testing (60 second intervals) appears to work perfectly. Thanks!