kizniche/Mycodo #52: What does this code do?

Cabalist, 2015-11-23:
Hello!

I am just curious about the code here: https://github.com/kizniche/Mycodo/blob/7034895f733f30f466ebb64ed77bd8751ad05402/3.5/cgi-bin/mycodo.py#L3305

on my system it seems to produce an array of zeroes.  Something like this:

where
 conditional_t_number_conditional has a length of 10 and 
 conditional_t_number_sensor has a length of 4
```
[[[0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0]],
 [[0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0]],
 [[0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0]],
 [[0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
  [0, 0, 0, 0, 0, 0, 0, 0, 0, 0]]]
```

Could you help me understand this code?

kizniche, 2015-11-23:
Heh... This may not be efficient at all, but it works. I am storing the variables for each conditional statement (may be multiple) for each sensor (may be multiple).

Directly after initializing the array, the arrays are set:
Lines 3116 though 3156.

Then on lines 768 and beyond it will check if the value is not 0 to determine if there is a conditional.

Why there is a range of 10, I'm not sure. This was a tricky section for me to get working and I believe I haven't put much more though into it after I found the first working solution.

kizniche, 2015-11-23:
Initialized to 0s based on how many conditional statements created for each sensor:
```
    cur.execute('SELECT id FROM tsensor')
    for row in cur:
        conditional_t_number_sensor.append(row[0])

    cur.execute('SELECT id FROM tsensorconditional')
    for row in cur:
        conditional_t_number_conditional.append(row[0])

    conditional_t_id = [[[0 for k in xrange(10)] for j in xrange(len(conditional_t_number_conditional))] for i in xrange(len(conditional_t_number_sensor))]
```

Set from 0 to value in SQLite DB:
```
for row in cur:
            conditional_t_id[j][count][0] = row[0]
```

Then check if != 0:
```
        #
        # Check if T conditional statements are true
        #
        for j in range(0, len(conditional_t_number_sensor)):
            for k in range(0, len(conditional_t_number_conditional)):
                if conditional_t_id[j][k][0] != 0 and client_que != 'TerminateServer' and pause_daemon != 1 and PID_change != 1:
```

kizniche, 2015-11-23:
Aha! I think I used 10 because at the time I was unsure how to define the proper length. I figured 10 would be enough conditional statements for each sensor and I would come back and fix this later. I actually am unsure what I was thinking and I'm still a little unsure how the code works, but I managed to get it working.

Cabalist, 2015-11-23:
Gotcha.  Everybody has code like that sometimes.  :laughing: 

I rewrote some of this.  You will need to test to make sure that it creates the same values you are expecting but I don't anticipate that it is any different.

closed 2015-11-23 by Cabalist

kizniche, 2015-11-23:
Thanks for the help. If you can assist me in understanding a little bit... Is what I'm trying to accomplish achieved by changing:
`conditional_t_id = get_3d_array(10, len(conditional_t_number_conditional), len(conditional_t_number_sensor))`
to
`conditional_t_id = get_3d_array(len(conditional_t_number_sensor), len(conditional_t_number_conditional), 1)`
Initial tests seem to work... Still testing, though.

kizniche, 2015-11-23:
I'm fairly sure that did it. Because I'm only setting the first element of the third list, this needs to be 1?
The array is initialized with:
`conditional_t_id = get_3d_array(len(conditional_t_number_sensor), len(conditional_t_number_conditional), 1)`
Then the last element is set with:
`conditional_t_id[j][count][0] = row[0]`

kizniche, 2015-11-23:
Nope. List out of range error when multiple sensors is use.

kizniche, 2015-11-23:
Aha. After some rudimentary testing, I found that I was misinterpreting how I was declaring the 3d array. I was understanding it backwards. The following is the correct way to declare the array:
`conditional_t_id = get_3d_array(1, len(conditional_t_number_conditional), len(conditional_t_number_sensor))`
If I want to read it like:
`conditional_t_id[sensor_number][conditional_number][0]`
Fixed in https://github.com/kizniche/Mycodo/commit/fe583e9cf0b3f288dbb2afa0dc438ff702b9aca2

kizniche, 2015-11-23:
Perhaps x, y, and z of get_3d_array() should be reversed, to enhance readability.

Cabalist, 2015-11-23:
X Y and Z are arbitrary.  I purposefully didn't put any of the sensor logic
into that function.  It might be worthwhile to put some notes before the
blocks of initialization to say what you are using it for there.  :)

kizniche, 2015-11-23:
Good idea. I'll add some comments.

I was meaning with reversing x y, and z for readability, so the order of the function parameters is the same order as the array indexes. For instance, currently this is how the function and array is used:

```
conditional_array = get_3d_array(1, number_conditionals, number_sensors)
conditional_array[number of sensors][number of conditionals][0] = True
```

But if x, y, and z are reversed, it appears as:

```
conditional_array = get_3d_array(number_sensors, number_conditionals, 1)
conditional_array[number of sensors][number of conditionals][0] = True
```