kizniche/Mycodo #55: Update from v3.5.89 to current - daemon will not start

MisterMeeseeks, 2015-11-23:
Trying to update Mycodo from v3.5.89 to current (to add the fix for timers and triggers). The daemon shuts down fine and the update finishes without issue (at least according to logs). But the daemon will not start back up... not from the web interface (which shows up with the daemon off and the last log entry is the update switching off the daemon) and not from the command line (via "sudo service mycodo start"). I tried restarting the Pi as well, but the daemon will not start back up. I've tried a few times now and the same thing happens every time.

(I meant to say before that this software is great. Thank you so much for making and maintaining it, and also for your fast responses.)

kizniche, 2015-11-23:
It's been a pretty chaotic evening of updates. It should be stable right now. Go ahead and update and see if everything works.

kizniche, 2015-11-23:
If after the latest update, the daemon doesn't automatically start, go ahead and issue the command "sudo /var/www/mycodo/cgi-bin/mycodo.py -l d" and copy any errors here.

kizniche, 2015-11-23:
I'm off to sleep for now.

kizniche, 2015-11-23:
It's morning (barely) and another day. Let me know if you get any errors with that command and I'll diagnose the issue.

MisterMeeseeks, 2015-11-23:
I tried the command and it output this:

Traceback (most recent call last):
  File "/var/www/mycodo/cgi-bin/mycodo.py", line 49, in <module>
    import smbus
ImportError: No module named smbus

I'm restoring my backup and will try again when you have a look.

kizniche, 2015-11-23:
Ah, a new prerequisite is python-smbus, so use the following command it should start up.

`sudo apt-get install python-smbus`

MisterMeeseeks, 2015-11-23:
That fixed it perfectly... thank you so much for all your help!

kizniche, 2015-11-23:
You're welcome. I'm glad to help.

closed 2015-11-23 by kizniche

kizniche, 2015-11-23:
The bigger question is whether the timer fix worked ;)

Let me know when you get a chance to test it.

MisterMeeseeks, 2015-11-23:
I tested the timers with 60 seconds on each (i.e. relay 1 on for 60, relay 1 off triggers 4 on for 60 seconds, relay 4 off triggers 1 for 60 seconds... to loop them). It appeared to work that way, including writing the relay information to the logs. I just set them for their real times and I will watch them for the next day or two to make sure, but it looks like you nailed it. Thanks so much!

MisterMeeseeks, 2015-11-30:
Great work... it is working flawlessly. Thank you for your quick solution(s).

kizniche, 2015-11-30:
Glad to hear! Thanks for the followup letting me know it's working.

You actually came up with a good alternative to a time-based timer. However, a power outage would disrupt or offset it. I have an issue created as a reminder to implement a time-based timer in the near future. There are just other priorities ahead of that to take care of.