kizniche/Mycodo #56: int and time.time() and precision

Cabalist, 2015-11-24:
Out of curiosity why is time cast as an ```int``` in your code?  

Examples:

roughly L2009 in mycodo.py
```python
while last_ht_reading > int(time.time()):
        time.sleep(0.25)
```
or

roughly L2038 in mycodo.py
```python 
last_ht_reading = int(time.time()) + 2
```

time.time() returns a ```float``` normally and python will add and compare ```int```s and ```float```s without issue.

It produce inexact results in timers as well.  

Consider ```time.time()``` is 1448345968.99.  Converted to an int that is 1448345968.  It takes one ```time.sleep(0.25)``` to become 1448345969.  

Now consider  ```time.time()``` is 1448345968.01.  Again converted to an int that is 1448345968. But it takes 4 ```time.sleep(0.25)``` to become 1448345969.  

Your readings from the sensors could take ± .75 seconds here.  Floats would alleviate the precision issue.  

Just curious. :)

kizniche, 2015-11-24:
Good question!

I went through all instances and can see no reason to do it. I suppose I may have had a problem in the past with it being a float, then never stopped casting it as an int for some reason.

Cabalist, 2015-11-25:
This seems to be fixed.  Closing my issue

closed 2015-11-25 by Cabalist