kizniche/Mycodo #62: Conversion to SQLAlchemy

Cabalist, 2015-11-26:
Hello again! 

I just made a branch on my fork that converts the initial database setup to SQLAlchemy and replaces optparse with argparse.

I am not submitting it as a pull request yet because it mildly changes some behaviour.   Can you take a look at it and tell me what you think.  

Every piece of SQL was replaced with Python objects.  It is now Backend independent, meaning that it would work just as well on Postgres, MySql and SQLite and more.  

This is my first step to eliminating globals and simplifying the code immensely.  :)  

Check it out here:  https://github.com/Cabalist/Mycodo/tree/Sqlalchemy_conversion

Cabalist, 2015-11-26:
I'm about to go do turkey stuff but if things seem unclear as to what has happened post it here and I'll respond as quick as I can.

kizniche, 2015-11-26:
Very Nice! I understand most of it.

What does `action='store_true'` do?

kizniche, 2015-11-26:
Found more files.. still going over it all.

Cabalist, 2015-11-26:
It just means that if the flag is given the value associated with it is
```True``` rather than a user given one.   This way I can still test for it
and run code.

Cabalist, 2015-11-26:
It's alot.  haha

kizniche, 2015-11-26:
I don't see a method for updating the database (add/delete/change columns/tables based on DB version), unless it's not immediately obvious.
Everything looks great!

Cabalist, 2015-11-26:
Haha!  SO!  SQLAlchemy is amazing.  You can see users being added in update_database.add_user()

Run ```update_database.py -a``` and then look at your SQLite table.  The user should be in there when you are done.  

This is a great intro to SQLAlchemy (if a little outdated):
http://www.blog.pythonlibrary.org/2012/07/01/a-simple-sqlalchemy-0-7-0-8-tutorial/

You can do some amazing thing with SQLAlchemy.  Right now we just have a bunch of unconnected tables but we can integrate functions into the objects (I used Users.set_password as an example.)  This is going to eliminate hundreds and hundreds of lines in mycodo.py

I just don't want to commit this and have it break a bunch of your stuff!  The SMTP table and the lack of updates are the only changes I believe I made but there may be subtle things that I missed.  Without tests I won't know for sure.

kizniche, 2015-11-26:
Right! I was about to say SQLAlchemy is amazing, but I didn't want it to steal your thunder for putting this all together. I'll need some time to go over everything to understand it, but I'm thoroughly impressed. It's so much more readable and easier to edit.

Cabalist, 2015-11-26:
Updates were a casualty because from what I could see it was a manual process of you writing down the changes in a function.  I plan that from here on out we'll integrate [Alembic](https://alembic.readthedocs.org/en/latest/) for updates.  This is a) more automated and b) easy.  

Current users can use the old update-database to get up to the SQLAlchemy version of this code and then you don't need to worry about it anymore.

Cabalist, 2015-11-26:
I realize I am throwing a lot of new packages at you.   These are proposals and this is your project.  I am changing a pretty base level piece.  Take the time to look it over and understand it.  I am in no rush to break your baby.

Cabalist, 2015-11-26:
Oh!  And check out populate_db in mycodo_db.__init__ There was some random stuff that I duplicated there.

kizniche, 2015-11-26:
I'll need at least a cup of coffee and an open day to test it all, but my cup is wearing off and it's near midnight, so I'll continue this another day... Probably this weekend. I cloned your branch and started testing, and received this:
```
kiz@tango:Mycodo-Sqlalchemy_conversion/3.5$ ./update_database.py -a                     11/25/15 11:44 PM
Add user to database
User (a-z, A-Z, 2-64 chars): test
Password:
Password (again):
Email: test@test.com
Failed to create user.  You most likely need to create the DB before trying to create users.
```

kizniche, 2015-11-26:
Created it first... Okay, that's the cue for sleep. I should make a commitment not to code tired. Bad things can happen.

kizniche, 2015-11-27:
All databases work perfectly after adding a missing table, renaming a column name and adding the following to create_dbs():
```
        from databases.mycodo_db import populate_db
        populate_db(MYCODO_DB_PATH)
```

I have your branch cloned with the edits. How should I go about updating it? (I'm not the most familiar with git)

kizniche, 2015-11-27:
I also noticed using `pass` as a column name resulted in errors, which is most likely why you renamed it `passw`

kizniche, 2015-11-27:
Just noticed this comment:
`passw = Column(TEXT)  # TODO:  Note that this is 'passw' and not 'pass'. 'pass' is reserved`

Cabalist, 2015-11-30:
Great! 
 I was mulling this over and while I did add some stuff I effectively removed the update database functionality from the ```update_database.py``` script.   I am going to take a stab at reincorporating that functionality along with the SQLite version PRAGMA.

Probably the easiest way to get me to see the changes is if I gave you write access to this repo and you pushed your changes up as a branch.  You should see those come across in a few minutes.

kizniche, 2015-11-30:
It probably doesn't help that I rearranged some things, but I pushed the changes to https://github.com/Cabalist/Mycodo/tree/Sqlalchemy_conversion_kizniche

One thing that irked me about SQLite is its inability to delete a column without having to rewrite the entire table, which I sometimes wanted to do in a database update. Can SQLAlchemy do this somehow or would you recommend a different backend?

Cabalist, 2015-11-30:
All sensible changes.  I made a pull request.

Yeah.  Alembic would be the system that I would recommend.  I meant to implement that this weekend but ended up working on mycodograph.  Here is a quick blurb about Alembic:

>Alembic is a database migrations tool written by the author of SQLAlchemy. A migrations tool offers the following functionality:

>- Can emit ALTER statements to a database in order to change the structure of tables and other constructs
>- Provides a system whereby “migration scripts” may be constructed; each script indicates a particular series of steps that can “upgrade” a target database to a new version, and optionally a series of steps that can “downgrade” similarly, doing the same steps in reverse.
>- Allows the scripts to execute in some sequential manner.

Which should be very familiar to what was in update-database manually.  Instead of thinking of the DB as tables and entries directly you start thinking of them as Python objects.  If you make a commit that alters the structure of one the SQLAlchemy models you would run Alembic and that would make a small script to alter your table that gets committed to github. Over time you accumulate a directory that is able to move your database up to the current code commit from where ever the current user is.  

There is obviously a little bit of a learning curve here.  First is SQLAlchemy and then Alembic.

kizniche, 2015-11-30:
So just to be clear, I shouldn't merge the pull request until we get the update system working?

I noticed update-mycodo.sh and update-post.sh were not changed (update-post.sh still executes `update-database.py -i update` and checks the DB version to enact some log format modifications).

Cabalist, 2015-11-30:
Yeah.  I am going to put the old update database code back into the updated
file.  That way there is no loss of functionality.  I have some big commits
done and a couple more that I need more time to cook that are dependent on
these.

kizniche, 2015-11-30:
Wow... A lot of stuff to go over.

I noticed that it appears my commit https://github.com/Cabalist/Mycodo/commit/bbe7e851e5278f180fe4028d4a696c26b739ef4f doesn't seem to be reflected in the latest version of your fork, specifically the Uploads table addition to databases/notes_db/models.py. Maybe there are others, but I just noticed that. Perhaps I'm just not understanding git.

kizniche, 2015-11-30:
I see a lot of action happening and the master branch all caught up. I saw you closed the pull request. I suppose you're waiting to create another pull request until the update system is operational.

Cabalist, 2015-11-30:
Yep.  Dug myself in a hole with Git.  I believe I am recovered and all up
to date.  I like doing smaller pull requests but it looks like I may make a
big one...

kizniche, 2015-12-01:
If I wanted to, for instance, change all references of update-database.py to legacy_update-database.py, in your fork, to ensure a future merger doesn't break the system, what's the best way for me to do this? Just push the edits to master?

kizniche, 2015-12-01:
Or just a new branch for you to merge when ready?

kizniche, 2015-12-01:
...Just looking for ways to help.

Cabalist, 2015-12-01:
I'd say make a branch and I'll merge it into master when you are finished.  This is going to get confusing.  Hopefully this gets back into shape soon and it can be merged into the proper project.  I don't have copies of the previous database schema.  

Hmm...  I assumed that the version I made was the latest and didn't need the 'updates'.  Are the updates applied when the DBs are initialized?

Who is update-databases for?

kizniche, 2015-12-01:
There's an option on the Settings Tab of the web interface that allows updating the latest version of git. It runs update-mycodo.sh, which will force pull the latest version (including update-post.sh), then execute update-post.sh. In update-post.sh is where `update-database.py -i update` is run from.

I set it up this way because there may be a circumstance that I need to execute code from within the latest version that's on git. Because the databases are versioned, I can determine where the user is at. So, if I need to manipulate the database, or run a command, it can be performed with these three scripts.

The update I was referring to was for https://github.com/Cabalist/Mycodo/blob/master/3.5/update-post.sh so it points to the correct update-database.py.

For instance, if someone is updating from an earlier version that doesn't have support for email notification, the table for these settings needs to be created, otherwise the web interface will give a warning or a critical PHP error and not render the page.

kizniche, 2015-12-01:
So when a new feature is added, the database(s) need to be updated if there are new settings that need to be stored. You can see that the database update script will check the database version and update accordingly until the database is up-to-date.

kizniche, 2015-12-01:
It does no other checking beyond the database version. So, there is one chance to update the database properly. If it's at say, version 22, and we're currently at version 24, it will apply the updates to being it to version 23, then apply the updates to bring it to version 24, then it will update the version to 24.

kizniche, 2015-12-01:
In the update-database.py script, you may notice it call the ModNullValue() function a lot. This just changes NULL entries to a particular value. I should have kept the AddColumn() function that went with those together in mycodo_database_update(), but I actually placed those additions in mycodo_database_create() every time I updated the script.

kizniche, 2015-12-01:
So it should have looked like:
```
        if current_db_version_mycodo < 15:
            AddColumn(sql_database_mycodo, 'Misc', 'Relay_Stats_Volts', 'INT')
            ModNullValue(sql_database_mycodo, 'Misc', 'Relay_Stats_Volts', 120)
            AddColumn(sql_database_mycodo, 'Misc', 'Relay_Stats_DayofMonth', 'INT')
            ModNullValue(sql_database_mycodo, 'Misc', 'Relay_Stats_DayofMonth', 15)
```

Instead of only the ModNullValue()s in there.

kizniche, 2015-12-01:
I hope that answers your question or clarifies things, and I didn't just tell you stuff you already knew ;)

kizniche, 2015-12-01:
Ah, it makes sense you were referring to:
> I noticed that it appears my commit Cabalist@bbe7e85 doesn't seem to be reflected in the latest version of your fork, specifically the Uploads table addition to databases/notes_db/models.py. Maybe there are others, but I just noticed that. Perhaps I'm just not understanding git.

It appears these updates were applied after all the mergings. The only change I see now is to rename the references for update-database.py to legacy_update-database.py

kizniche, 2015-12-01:
Sorry for so many comments, but I just noticed your commit message:
>  restored legacy update-datebase code to ensure that databases pre-3.5 can be upgraded to 3.5.

To clarify, update-database.py is used to update to specific versions just within v3.5. For instance, when settings are added, a new table, columns, entries need to be created.

Cabalist, 2015-12-02:
Don't know how I missed all these! 

Basically you'll need to replace anywhere it says:
```update-database.py -i update```
with:
```legacy_update-database.py -i update```

That is the only feature I took out of update_database.py.  All the others are still there.

kizniche, 2015-12-02:
Yup, a simple fix. But there's only one place that this command is executed, so it makes it easy. But still crucial that it gets changed, or it will temporarily break the user's database update system, which could allow updating everything but the database (and with little error-handling on the front-end, this could break their ability to perform another update to correct the issue).

closed 2016-07-02 by kizniche