kizniche/Mycodo #63: mycodoGraph.py questions

Cabalist, 2015-11-30:
Hello,

I am spending some time on refactoring this file and thought I would consolidate my notes here.  It seems like this function takes a number of values that come from the DB and generate some text files from them and then run gnuplot on the text files.  Much of this is self-explanatory and I think I have made some serious inroads on this file but I am going to post my questions here.

```sensor_t_graph``` is a list of values from the DB that are in the ```graph``` column.  Could you give me some of these example values? 

I am following along in ```mycodo.py``` and sensor_t_graph looks like this:
```python 
global sensor_t_graph
...
sensor_t_graph = []
...
for row in cur:
    sensor_t_graph.append(row[8])  # 8 is the 'graph' column in the 'tsensors' table

```
and then it is passed to mycodoGraph.  From what I can tell that makes it a list of strings but I don't know their content.

kizniche, 2015-11-30:
`sensor_t_graph` is either 1 or 0 and determines if a graph should be generated for that particular sensor. If it is set to 1, a graph will be generated and displayed on the Graph Tab of the web interface. The index is the sensor number, starting at 0, for that particular sensor class (temperature sensors, in this case).

Cabalist, 2015-11-30:
Sometimes bash is a mystery to me.  What does /var/www/mycodo/cgi-bin/log-parser.sh do?

kizniche, 2015-11-30:
It will produce $OUTFILE with only the last number of $LINES lines of $INFILE:
```
SENSORNUMBER=$1
SENSORTYPE=$2
LINES=$3
INFILE=$4
OUTFILE=$5

if [[ ( "$SENSORTYPE" = "t" ) && ( "$LINES" != "0" ) ]]; then
        awk -v num=$SENSORNUMBER '$3 == num' $INFILE | tail -n -$LINES | tee $OUTFILE 1> /dev/null
elif [[ ( "$SENSORTYPE" = "t" ) && ( "$LINES" = "0" ) ]]; then
        awk -v num=$SENSORNUMBER '$3 == num' $INFILE | tee $OUTFILE 1> /dev/null
fi
```

This is actually a very crude way of displaying a certain amount of time in the past. It caluclates how many lines to go back based on the currently-set Measure Interval, which can be used to calculate roughly how many lines was, for instance, X hours in the past. This is easily skewed if the daemon restarts and changes the cycle or a queue in the daemon (more noticeable when displaying short time periods and if the Measure Interval is set to a long duration).

A better version of this is in cgi-bin/log-parser-chart.sh which actually finds the actual lines with the timestamp that is between the start and end dates.

```
SENSORNUMBER=$1
SENSORTYPE=$2
TIMESTART=$3
TIMEEND=$4
INFILE1=$5
INFILE2=$6
OUTFILE=$7

...
        if [[ ( "$SENSORTYPE" = "t" ) ]]; then
                if [[ ( "$TIMESTART" != "0" ) && ( "$TIMEEND" != "0" ) ]]; then
                        cat $INFILE1 $INFILE2 | awk -v num=$SENSORNUMBER '$3 == num' | awk -v timestart="$TIMESTART" -v timeend="$TIMEEND" -F, '{ if ($1>timestart && $1<timeend) print }' | tail -n -$LINES | tee $OUTFILE 1> /dev/null
                elif [[ ( "$TIMESTART" != "0" ) && ( "$TIMEEND" = "0" ) ]]; then
                        cat $INFILE1 $INFILE2 | awk -v num=$SENSORNUMBER '$3 == num' | awk -v timestart="$TIMESTART" -F, '{ if ($1>timestart) print }' | tee $OUTFILE 1> /dev/null
                fi
        fi
...
```

kizniche, 2015-11-30:
Hmm, interesting to see `tail -n -$LINES` in cgi-bin/log-parser-chart.sh

I think that's in error.

kizniche, 2015-11-30:
The multiple $INFILES refer to the temporary log file on the tmpfs and the permanent log file in the logs directory, and will be concatenated before processing.

Cabalist, 2015-11-30:
Ah.  So the files already exist and when you generate a graph file you concatenate a bunch together.

So the sensor readings are stored in files and not in the DB?

kizniche, 2015-11-30:
Correct. The sensor log values are stored as plain text. Here are a few examples:
```
Y/M/D-H:M:S          CO2   Sensor
2015/11/29-23:13:32  1136  0
2015/11/29-23:15:36  1138  0
2015/11/29-23:17:45  1140  0
```
```
Y/M/D-H:M:S          Temperature(C)  Pressure(kPa)  Altitude(m)  Sensor
2015/11/29-23:13:09  27.5            98240          259.6        0
2015/11/29-23:15:19  27.5            98242          260.2        0
2015/11/29-23:17:30  27.5            98237          260.3        0
2015/11/29-23:19:41  27.5            98230          261.3        0
```
Values are actually only separated by one space. I copied these from the web interface user output that runs it through the linux command `column` first.

kizniche, 2015-11-30:
I had wanted to move the storage of log data to a DB, but I haven't committed to doing so because I was hesitant about which to chose and how to implement it, even though it would make things a lot easier for manipulating data.

Cabalist, 2015-11-30:
Interesting!  Alright.  That makes some of this easier than.  

DB stuff is hard.  I tend to abstract it away with SQLAlchemy and that does a really good job of preventing me from worrying about implementation too much.  I'll chip away at some stuff over here and maybe offer some suggestions for Mycodo 4.0.  :wink:

Cabalist, 2015-11-30:
Is the length of sensor_t_name and the length of sensor_t_graph always going to be the same?  From the code in mycodo this looks like the case.  

If ```sensor_t_name``` is a list of sensor names and ```sensor_t_graph``` is a bool on whether or not to generate a graph for the sensor name that should be the case.

kizniche, 2015-11-30:
There is probably a performance improvement to be gained by moving logging to a DB, but I have so much relying on plain text that it made it feel like a daunting task to convert everything over.

Yes, every index for arrays `sensor_t_variable_name` will be the same and correspond to a particular sensor.

kizniche, 2015-11-30:
I also like the safety with using plain text, as there's little risk of DB corruption, and it makes importing data into statistical software or excel very easy.

Cabalist, 2015-11-30:
Yep.  I hear you on that.  Tonight especially I feel like I am a good distance away from understanding the full architecture of this project.  There are a number of things happening all at once in these files that I need to wrap my head around.  I really appreciate you being so responsive to my questions.

kizniche, 2015-11-30:
Haha, I bet it's a little daunting to go through it all, especially since I lack a full understanding of good technique and I've been developing my style along the project, so one piece of code may make you scratch your head at a completely crazy way to do something and another piece may reaffirm that I may know something about Python.

"Don't know how to do it in Python? No problem, let's just grab the output from executing a PHP (password hash) or shell (log manipulation) script!"

Cabalist, 2015-12-01:
I *think* i just understood something about this whole project that is making it so hard to figure out.  The source files are so large it is hard to see individual actions as they are happening.  I have been trying to understand some of the patterns that I see over and over.  I think I found the a nice succinct version of it while cleaning up small files:

GPIO-initialize.py:
```python
import RPi.GPIO as GPIO
import sqlite3

#### Configure Install Directory ####
install_directory = "/var/www/mycodo"

sql_database = "%s/config/mycodo.db" % install_directory

# GPIO pins (BCM numbering) and name of devices attached to relay
relay_id = []
relay_pin = []
relay_trigger = []
relay_start_state = []

def ReadSQL():
    global relay_pin
    global relay_trigger
    conn = sqlite3.connect(sql_database)
    cur = conn.cursor()
    cur.execute('SELECT Id, Pin, Trigger, Start_State FROM Relays')
    count = 0
    for row in cur:
        relay_id.append(row[0])
        relay_pin.append(row[1])
        relay_trigger.append(row[2])
        relay_start_state.append(row[3])
        count += 1

ReadSQL()

# Turn all relays off
# Set all relays to be turned off at startup
for i in range(0, len(relay_id)):
    if relay_trigger[i] == 0: relay_trigger[i] = 1;
    else: relay_trigger[i] = 0

# Turn specific relays on
# If relay is set to be turned on, reverse the trigger
for i in range(0, len(relay_id)):
    if relay_start_state[i]:
        if relay_trigger[i] == 0: relay_trigger[i] = 1;
        else: relay_trigger[i] = 0

GPIO.setmode(GPIO.BCM)
GPIO.setwarnings(False)

for i in range(0, len(relay_id)):
    if relay_pin[i] != 0:
        GPIO.setup(relay_pin[i], GPIO.OUT)
        GPIO.output(relay_pin[i], relay_trigger[i])
```

I am going to try and step through the code as I see it and make some comments about different bits.  I am not infallible and you may disagree with me (or someone else might as this is public).  Feel free to question my conclusions here.  I'll point out some general stuff as well.  

First, The file name ```GPIO-initialize.py```  should be ```gpio_initialize.py```.  It is difficult to import modules (the name for python source files) if they have hyphens.  See [here](https://www.python.org/dev/peps/pep-0008/#package-and-module-names) and [here](http://stackoverflow.com/questions/7583652/python-module-with-a-dash-or-hyphen-in-its-name)

PEP8 is your best friend and it is very readable.  Keeps your code from looking weird and allows others to jump right in.  

Now for the actual source, 
```python
relay_id = []
relay_pin = []
relay_trigger = []
relay_start_state = []
```
I am going to go straight to the 4 global variables ```relay_id```, ```relay_pin```, ```relay_trigger``` and ```relay_start_state```.   There is some tricky namespace stuff here that wouldn't let this function work if it was imported.   ```relay_pin``` and ```relay_trigger``` are globals but ```relay_id``` and ```relay_start_state``` will not exist outside outside the scope of the functions if this function is imported into a different file.  Not good.  
[... 72 lines cut ...]
# Set all relays to be turned off at startup
for i, each_trigger in enumerate(relay_trigger):
    if each_trigger == 0: 
        relay_trigger[i] = 1;
    else: 
        relay_trigger[i] = 0
```

I still don't like this code because it is changing a global.  It isn't writing back to the DB either.  It is only changing it locally.  

The rest of the code has much the same feedback as this block so I won't repeat myself.   

The updated version of this uses SQLAlchemy.  I'll post that tomorrow.  I'm very sleepy!  

The pseudocode for that is:

```python
for each_relay in relays:
       if each_relay.trigger:
           # do a thing
       else:
          # do a different thing
```

Outside of some imports that is going to be the totality of the code.  For each relay check the status of some things and then do a thing.  No globals.  Specifics soon!

kizniche, 2015-12-01:
All good points and I concur.

In GPIO-initialize.py, ReadSQL() was merely copied from mycodo.py, but I forgot to include the cur.close() at the end. This is also why the local variable names are the same as the globals from mycodo.py.

I'm not sure how I came to use the variable/index method to store the SQL values. Make no mistake though, it was most likely the result of the first solution that worked!

As for everything else, I suppose I was just trying to get things to work and not necessarily writing with the intent of releasing it for others' eyes [to tear from looking at]. I eventually decided to share it and began first with adding comments to try to help understand what things did, then I would go back, as I learned more, and redesigned parts of the code to be more efficient and understandable. Many parts remain in the same crude state as they were when they were conceived, the first creature born that was able to desperately crawl out from the muck, displaying the most basic assemblage of functions to survive, which although happened to work, may never have evolved since, or only slightly evolved to work with other advances elsewhere.

I knew from the beginning that a lot of what I was doing wasn't the best way, and I took the gamble to release it to the world to help others and ultimately learn. I appreciate your help with improving it.

Cabalist, 2015-12-01:
Haha.  Your writing is very funny.  

Well I should ask then because mycodo.py would seem to be the next target after this.  Do you want to spend time 'correcting' Python?  Or are there other things that you'd rather be working on? Features, etc.  

I have some ideas on making inroads to eliminating the globals but it is an all or nothing rewrite.  Is this what you'd like to do?

kizniche, 2015-12-01:
There aren't any major features I'm itching to add. I added support for using the Pi's integrated CPU/GPU temperature sensors last night... And that was just a good idea that sporadically came to me and I sat down for an hour to add it.

I'll go through mycodo.py when I get some in a few hours and work on cleaning it up.

kizniche, 2015-12-01:
It's difficult for me to spearhead major changes when I'm not knowledgeable about the proper technique. I made as many edits as I could in commit https://github.com/kizniche/Mycodo/commit/780437ad226e185200ce8bd596f4cfbe03ba2883 without just doing things experimentally. If you see any other changes I could perform, I could help, otherwise I'm not sure where to go from here.

Cabalist, 2015-12-02:
No worries.

Quick question if you are still up.  I just want to make sure I understand this code correctly:

```python
# Turn all relays off
# Set all relays to be turned off at startup
for i, each_trigger in enumerate(relay_trigger):
    if each_trigger == 0:
        relay_trigger[i] = 1;
    else:
        relay_trigger[i] = 0

# Turn specific relays on
# If relay is set to be turned on, reverse the trigger
for i, each_trigger in enumerate(relay_trigger):
    if relay_start_state[i]:
        if each_trigger == 0:
            relay_trigger[i] = 1;
        else:
            relay_trigger[i] = 0

GPIO.setmode(GPIO.BCM)
GPIO.setwarnings(False)

for i, each_pin in enumerate(relay_pin):
    if each_pin != 0:
        GPIO.setup(each_pin, GPIO.OUT)
        GPIO.output(each_pin, relay_trigger[i])
```

The first ```for``` loop says: for each relay_trigger if it is 0 set it to 1 else set it to 0.  If I understand this correctly this would just flip the values since there is only 0,1 for the relay_trigger.  A list that looks like this:

```
[0,0,1,1,0,1,1,1,1]
```

would turn into this:

```
[1,1,0,0,1,0,0,0,0]
```
But the notes seem to imply something different.  The notes seems to say the code should set all of these to 0/off.

The next step takes that list and check another variable ```relay_start_state```.  If it is 1 then the value would be flipped again.  

The final step says that for each pin EXCEPT the first one we would set each pin to GPIO.OUT and output the relay_trigger on that pin.  Nothing is changed on the HW side until this point.  

Could you, just for my sake as an outsider, outline what you'd like this code to do?

(This is my handicap.  I look at the code so much I can sometimes miss what it is supposed to be doing)

kizniche, 2015-12-02:
So we have two states we are considering here. The state of the GPIO and the state of the relay. The relay can be active either when the GPIO is HIGH or LOW. The trigger indicates if the relay is on when it receives a HIGH signal (trigger = 1) or a low signal (trigger = 0). To save space, at the cost of readability, I reused the trigger variable to store the value that will be passed to the GPIO function.

For example, if we have a relay that activates when there is a LOW signal, trigger is initially 0. If this relay has not been configured to turn ON at boot, only the first function will reverse the value of trigger (0 to 1) and the GPIO function will turn the GPIO HIGH (1), turning the relay off. If it is configured to be turned on, the second function will again reverse trigger (back to 0), and when used in the GPIO function, will set it LOW (0).

If I hadn't reused the trigger variable and had commented about a relay either activating high or low, this might have been clearer.

kizniche, 2015-12-02:
Note: for complete clarity, start_state is set to 0 to be off and 1 to be on. So only if it is 1 does trigger get reversed to permit the relay to turn on.

kizniche, 2015-12-02:
Last note: if the pin is 0, the relay has not been correctly configured (there is no GPIO 0 using BCM numbering), and it should not be attempted to be used with the GPIO function. If it is not 0 (the initial value), we can assume the user has configured the pin.

kizniche, 2015-12-02:
> for each pin EXCEPT the first one

I think you may be misinterpreting the for loop. It is not excluding the first pin, it is going through all values of the pin list and checking if it is not 0 (the default value) and then modifying that GPIO based on the pin value and trigger value.

Cabalist, 2015-12-03:
Oh neat!  I see.  So it flips all the triggers off and then depending on the start_stop_state it flips them back on.  Does that seem right?  

I just made a new branch in my code.  I renamed the file to initialize_relays.py  Could you run it and see if it gives the expected results?

Cabalist, 2015-12-03:
I am beating Git off with a stick these days.  Let this be a lesson to me to keep my pull requests small.  *sigh*

kizniche, 2015-12-03:
Looking at the code, I don't seem to see a part where the relays are initially turned OFF. This step is crucial. Since some GPIOs are HIGH and some are LOW at boot, if all relays aren't switched to their respective OFF state (which could be either HIGH or LOW), then they could be ON and remain ON at startup (very bad depending on what is powered by the relay).

kizniche, 2015-12-03:
I also don't see any checks for if the `relay_pin` is 0, Let me know if I'm misinterpreting the code.

kizniche, 2015-12-03:
To further explain, what I see the code doing is only activating the relay for which `each_relay.start_state` is true. It is ignoring all other relays, which because GPIOs can be either HIGH or LOW at boot, the other relays can either be ON or OFF.

kizniche, 2015-12-03:
Testing now...

kizniche, 2015-12-03:
I receive this extremely long error:
```
Traceback (most recent call last):
  File "./initialize_relays.py", line 42, in <module>
    relays = the_session.query(Relays).all()
  File "/usr/lib/python2.7/dist-packages/sqlalchemy/orm/query.py", line 2320, in all
    return list(self)
  File "/usr/lib/python2.7/dist-packages/sqlalchemy/orm/query.py", line 2438, in __iter__
    return self._execute_and_instances(context)
  File "/usr/lib/python2.7/dist-packages/sqlalchemy/orm/query.py", line 2451, in _execute_and_instances
    close_with_result=True)
  File "/usr/lib/python2.7/dist-packages/sqlalchemy/orm/query.py", line 2442, in _connection_from_session
    **kw)
  File "/usr/lib/python2.7/dist-packages/sqlalchemy/orm/session.py", line 854, in connection
    close_with_result=close_with_result)
  File "/usr/lib/python2.7/dist-packages/sqlalchemy/orm/session.py", line 858, in _connection_for_bind
    return self.transaction._connection_for_bind(engine)
  File "/usr/lib/python2.7/dist-packages/sqlalchemy/orm/session.py", line 322, in _connection_for_bind
    conn = bind.contextual_connect()
  File "/usr/lib/python2.7/dist-packages/sqlalchemy/engine/base.py", line 1798, in contextual_connect
    self.pool.connect(),
  File "/usr/lib/python2.7/dist-packages/sqlalchemy/pool.py", line 338, in connect
    return _ConnectionFairy._checkout(self)
  File "/usr/lib/python2.7/dist-packages/sqlalchemy/pool.py", line 644, in _checkout
    fairy = _ConnectionRecord.checkout(pool)
  File "/usr/lib/python2.7/dist-packages/sqlalchemy/pool.py", line 440, in checkout
    rec = pool._do_get()
  File "/usr/lib/python2.7/dist-packages/sqlalchemy/pool.py", line 1057, in _do_get
    return self._create_connection()
  File "/usr/lib/python2.7/dist-packages/sqlalchemy/pool.py", line 285, in _create_connection
    return _ConnectionRecord(self)
  File "/usr/lib/python2.7/dist-packages/sqlalchemy/pool.py", line 411, in __init__
    self.connection = self.__connect()
  File "/usr/lib/python2.7/dist-packages/sqlalchemy/pool.py", line 538, in __connect
    connection = self.__pool._creator()
  File "/usr/lib/python2.7/dist-packages/sqlalchemy/engine/strategies.py", line 96, in connect
    connection_invalidated=invalidated
  File "/usr/lib/python2.7/dist-packages/sqlalchemy/util/compat.py", line 199, in raise_from_cause
    reraise(type(exception), exception, tb=exc_tb)
  File "/usr/lib/python2.7/dist-packages/sqlalchemy/engine/strategies.py", line 90, in connect
    return dialect.connect(*cargs, **cparams)
  File "/usr/lib/python2.7/dist-packages/sqlalchemy/engine/default.py", line 377, in connect
    return self.dbapi.connect(*cargs, **cparams)
sqlalchemy.exc.OperationalError: (OperationalError) unable to open database file None None
```

Cabalist, 2015-12-03:
>Looking at the code, I don't seem to see a part where the relays are initially turned OFF. This step is crucial. Since some GPIOs are HIGH and some are LOW at boot, if all relays aren't switched to their respective OFF state (which could be either HIGH or LOW), then they could be ON and remain ON at startup (potentially very bad, depending on what is powered by the relay).

From what I can tell from the original code (and there may be more to these data structures from what I am seeing), it doesn't do that either.   

```python
# Turn all relays off
# Set all relays to be turned off at startup
for i in range(0, len(relay_id)):
    if relay_trigger[i] == 0: relay_trigger[i] = 1;
    else: relay_trigger[i] = 0
```

This sets values in the relay_trigger list object but it doesn't communicate anything to the GPIO pins for the relays.  

Could you give a sample SQLite db that has a populated relay table?  I am making a minimal mycodo locally for testing some of these behaviours.

----
As for the error it looks like the database engine isn't getting a value passed to it that it should.  Also fun fact that I just patched locally: with SQLite paths it is 3 slashes for a relative file location, 4 slashes for absolute.  http://docs.sqlalchemy.org/en/rel_1_0/core/engines.html#sqlite

kizniche, 2015-12-03:
Yes, my mistake. It doesn't initially turn them off, but it does execute the GPIO function for each relay (except if the pin is 0). Your code only executes the GPIO function if the start_state is 1. This will exclude setting the GPIO for relays that have a start_state of 0.

For instance, if we have:

`relay_trigger = [0, 0, 0, 1, 1, 1]`

This means relays 1-3 are ON when the GPIOs is LOW, and relays 4-6 are ON when the GPIOs are HIGH.

if the following is true:

`relay_start_state = [1, 0, 0, 1, 0, 0]`

This means that only relays 1 and 4 should be activated at bootup, so lets look at what each of our codes do:

My code:
```python
# Turn all relays off
# Set all relays to be turned off at startup
for i, each_trigger in enumerate(relay_trigger):
    if each_trigger == 0:
        relay_trigger[i] = 1;
    else:
        relay_trigger[i] = 0

# Turn specific relays on
# If relay is set to be turned on, reverse the trigger
for i, each_trigger in enumerate(relay_trigger):
    if relay_start_state[i]:
        if each_trigger == 0:
            relay_trigger[i] = 1;
        else:
            relay_trigger[i] = 0

GPIO.setmode(GPIO.BCM)
GPIO.setwarnings(False)

for i, each_pin in enumerate(relay_pin):
    if each_pin != 0:
        GPIO.setup(each_pin, GPIO.OUT)
        GPIO.output(each_pin, relay_trigger[i])
```

After the first for loop:

`relay_trigger = [1, 1, 1, 0, 0, 0]`

After the second loop:

`relay_trigger = [0, 1, 1, 1, 0, 0]`

Then, if each GPIO is set (pins irrelevant as long as they aren't set to 0), the final loop:

Relay 1 ON
Relay 2 OFF
Relay 3 OFF
Relay 4 ON
Relay 5 OFF
Relay 6 OFF

Now, your code:

```python
    for each_relay in relays:
        #Setup all the pins as an output
        GPIO.setup(each_relay.pin, GPIO.OUT)
        if each_relay.start_state:
            # Only push the trigger to the relay with the start_state
            GPIO.output(each_relay.pin, each_relay.trigger)
```

The following will happen:

Relay 1 ON
Relay 2 Floating
Relay 3 Floating
Relay 4 ON
Relay 5 Floating
Relay 6 Floating

kizniche, 2015-12-03:
About the database, I'll send one over in a moment.

kizniche, 2015-12-03:
I believe this should fix the missing part:

```python
    for each_relay in relays:
        #Setup all the pins as an output
        GPIO.setup(each_relay.pin, GPIO.OUT)
        if each_relay.start_state:
            # Only push the trigger to the relay with the start_state
            GPIO.output(each_relay.pin, each_relay.trigger)
        else:
            # Turn relay off
            opposite_trigger = 1 - each_relay.trigger
            GPIO.output(each_relay.pin, opposite_trigger)
```

kizniche, 2015-12-03:
By "Floating" above, I don't mean the pins are actually floating, but like I said before, some GPIOs are set HIGH and some GPIOs are set LOW at bootup, and depending on what pins were chosen to be connected to the relays, this means that it can either be HIGH or LOW at bootup, so I'm calling this floating.

Cabalist, 2015-12-03:
AH!  This is awesome.  Thank you for walking me through this.

I love this file because it is a microcosm of the bigger one. Ok SO.  

GPIO.output take a boolean as it's second value.  This can be 0 or 1, True or False, GPIO.LOW or GPIO.HIGH.  This means we could just ```not``` the trigger.

For readability and succinctness we could possibly do:

```python 

    for each_relay in relays:
        #Setup all the pins as an output
        GPIO.setup(each_relay.pin, GPIO.OUT)

        # First turn the relay off
        GPIO.output(each_relay.pin, not each_relay.trigger)

        # Turn on relays that are set to start on
        if each_relay.start_state:
            # Only push the trigger to the relay with the start_state
            GPIO.output(each_relay.pin, each_relay.trigger)
```

kizniche, 2015-12-03:
The only reason I would recommend against this way is because of one potential scenario:
If the normal bootup state of a pin is ON, then it will turn it OFF and then ON in rapid succession. For some devices connected to the relay, this rapid switching could be bad. But if we use the else I suggested, the GPIO is only modified once to be wither ON or OFF. There may be a more readable way to do it besides the else, though.

Cabalist, 2015-12-03:
Would a sleep make sense?   ```time.sleep(2)```

My changes are ENTIRELY based around the readability of the code.  You (as far as I know) are the only one who knows HOW these should operate.  I have a big box of components coming in the mail which will minimize my ignorance of this.

kizniche, 2015-12-03:
How about this compromise?
```python
    for each_relay in relays:
        #Setup all the pins as an output
        GPIO.setup(each_relay.pin, GPIO.OUT)

        # First turn the relay off
        if not each_relay.start_state:
            GPIO.output(each_relay.pin, not each_relay.trigger)

        # Turn on relays that are set to start on
        if each_relay.start_state:
            # Only push the trigger to the relay with the start_state
            GPIO.output(each_relay.pin, each_relay.trigger)
```

Cabalist, 2015-12-03:
Very readable.  :)

Cabalist, 2015-12-04:
Alright so I now understand the globals and why they are there.  We've also replaced them with a Python object.  This simplifies the code and increases its flexibility.  

I am worried that as we make some big changes to mycodo.py that we won't be able to make them all at once and keep your project moving forward at the same time.  I've run into this already with the database_update.  The changes I make have to pull it down to base level and then add features to bring it back up to compatible.  I do not plan to change the database structure or the end results from mycodo.py just how that data is processed.  

Would you be open to making a brand new branch or even a new Github project for the Python code alone?  We would turn it into a module that would get called from your cgi-bin directory.  I think that this would be the cleanest way to avoid my branch drifting further and further away from mainline.  We aren't talking a diff.  This is going to be new directories and functions and things moving about.  

Thoughts?

kizniche, 2015-12-04:
That all sounds good. How would the module be installed with Mycodo? Clone Mycodo then clone the module in the cgi-bin directory?

kizniche, 2015-12-04:
Ah, Git has submodules that can take care of the linking

Cabalist, 2015-12-04:
Bingo!  :)

kizniche, 2015-12-04:
I'm all aboard! What should I name the new project?

Cabalist, 2015-12-04:
It's up to you!  I still haven't gotten the whole thing in my head yet.

I know it generates graphs.   Are the logs that mycodo generates the data that mycodo_graph reads?

I will say that a brief summary in a Readme will help me keep track of what the end goals should be.

kizniche, 2015-12-04:
Yes, mycodo_graph reads the data produced by mycodo.py.

So maybe something generic like mycodo_py or mycodo_python?

Cabalist, 2015-12-04:
That works!  Just so I know the boundaries:  all the PHP does is create the website correct?  No system level stuff?

Cabalist, 2015-12-04:
Like if the mycodo_graph went to a folder you could host the website on a differnet machine and just point to that folder?  

Saying that out loud makes me thing that the PHP has hooks in it to control relays and such.

kizniche, 2015-12-04:
I think so. What do you mean by system level?

Nothing else restructures the databases, or modifies log files

includes/restricted.php modifies database values

kizniche, 2015-12-04:
There is mycodo-client.py that the website uses to communicate with mycodo.py. For instance, to turn a relay on or off.

Cabalist, 2015-12-04:
System level meaning that the php could run on a different machine.  It doesn't do anything to the raspberry pi itself.

Cabalist, 2015-12-04:
mycodo_client!  I rewrote that!  I should have known!

kizniche, 2015-12-04:
includes/restricted.php creates lock files for the camera, modifies database values (most of what this file does)
includes/functions.php has a function to delete generated graphs if there are more than 40, creates thumbnails,

Cabalist, 2015-12-04:
includes/functions.php sounds appropriate.

includes/restricted.php sounds interesting.  A good target to convert to python.

kizniche, 2015-12-04:
includes/graph.php does a lot of manipulation of logs (using cgi-bin/log-parser-chart.sh) and makes tmp files

kizniche, 2015-12-04:
includes/restricted.php is essentially what is used by administrators to modify the database from the web interface, and more. It is what checks almost all button-presses and if the user is not a guest, executes the action for that button. Save settings, take photo, start video stream, turn PID controller on/off, turn timer on/off, add sensor, delete sensor, etc.

Cabalist, 2015-12-04:
ah.  I am ... not a fan... of PHP.  It sounds like that will remain as is for the moment.  We'll tidy up this first for sure.

kizniche, 2015-12-04:
lol

The border between PHP and Python is something I've always been hazy about. Maybe a complete separation is the healthiest. Though I don't think it's going to be quick and painless.

[9 more replies]

closed 2016-07-02 by kizniche