kizniche/Mycodo #66: PID Regulation Question

harrywykman, 2015-12-05:
Hi.  After your suggestion, Kyle, that I move to 3.5 from 3 I have done so.  I have everything up and running except I have a problem / question with the PID Regulation.

When I turn on a thread with, for example:

## temperature
 - Set Point = 26
 - Regulate = UP
 - Measure Interval = 90
 - Up Relay = 1
 - Kp = 1

the temperature thread will turn on, it is visible in the daemon and the green light is on, but all the values in the interface return to the defaults (eg. set point 25).  The default set point also appears in the header display of sensor values under the SET heading.

Is there some install step that I messed up or something that I am missing?

Thanks,

Harry

kizniche, 2015-12-05:
Just as a starting question to make sure, did you first save the configuration with the "Save" button?

kizniche, 2015-12-05:
Ah, if the relay is going on, then that must be the case. Could you provide me a login to see what's going on with your system? You can contact me at http://kylegabriel.com/contact

harrywykman, 2015-12-05:
Nope, you're right, I was just turning it on without saving. I didn't realise the top save applied to that setting.  Thanks.

kizniche, 2015-12-05:
Yeah, that's an area I've been meaning to work on to make it more clear. There's the ability to save presets and the "default" preset is the current configuration that must be saved over to take effect.

harrywykman, 2015-12-05:
Thanks.  It's working great now.  Just got to tune everything.

kizniche, 2015-12-05:
Now the fun begins ;)
Cheers

closed 2015-12-05 by kizniche