kizniche/Mycodo #68: dead link in Software Install section of 3.5 README.md?

boomstick8x, 2015-12-14:
I found that link in README.md not working anymore (anymore means from today:))
'http://sourceforge.net/code-snapshots/svn/m/mj/mjpg-streamer/code/mjpg-streamer-code-182.zip'
If you confirm dead link try by yourself this variant:
svn co https://svn.code.sf.net/p/mjpg-streamer/code/mjpg-streamer/ mjpg-streamer
cd mjpg-streamer
It works for me.

kizniche, 2015-12-14:
Interesting. I actually just reinstalled an entire system from a blank memory card yesterday and had no problem with any of the install commands. I have seen sourceforge give 404 errors for this file before. 

Have you tested whether that version of mjpg-streamer works? I recall using version 182 because there was a problem of some kind with the other versions.

kizniche, 2015-12-14:
Ah, I see the link you provided is version 182. I'll update the README. Thanks for catching that!

kizniche, 2015-12-14:
Fixed in https://github.com/kizniche/Mycodo/commit/5c0e2540527513e48e68815fc3af7c17a6ae291e

closed 2015-12-14 by kizniche

boomstick8x, 2015-12-14:
You are welcome!
p.s. Yesterday all was right(I install Jessie on my Pi), but today, when Im installing system on second Pi I catch this error.

boomstick8x, 2015-12-14:
Dont want to open new issue, will ask here.
Im trying to run Mycodo on Minibian(Lighter version of Raspbian), but cant get mycodo service running.
'service mycodo status'
'● mycodo.service - LSB: Start|Stop mycodo.py daemon
   Loaded: loaded (/etc/init.d/mycodo)
   Active: active (exited) since Tue 2015-12-15 01:16:27 MSK; 2min 39s ago
  Process: 1412 ExecStop=/etc/init.d/mycodo stop (code=exited, status=0/SUCCESS)
  Process: 1448 ExecStart=/etc/init.d/mycodo start (code=exited, status=0/SUCCESS)
Dec 15 01:16:27 project13 systemd[1]: Started LSB: Start|Stop mycodo.py daemon.'
May be you can imagine what is wrong?

kizniche, 2015-12-14:
I'm not sure. It looks like a systemd issue. What happens when you try to execute just the script?

```sudo /etc/init.d/mycodo start```

boomstick8x, 2015-12-14:
nothing!

boomstick8x, 2015-12-14:
And when run with debug option
___
sudo /etc/init.d/mycodo debug
Traceback (most recent call last):
  File "/var/www/mycodo/cgi-bin/mycodo_client.py", line 264, in <module>
    menu()
  File "/var/www/mycodo/cgi-bin/mycodo_client.py", line 87, in menu
    c = rpyc.connect("localhost", 18812)
  File "/usr/local/lib/python2.7/dist-packages/rpyc/utils/factory.py", line 89, in connect
    s = SocketStream.connect(host, port, ipv6 = ipv6, keepalive = keepalive)
  File "/usr/local/lib/python2.7/dist-packages/rpyc/core/stream.py", line 132, in connect
    return cls(cls._connect(host, port, **kwargs))
  File "/usr/local/lib/python2.7/dist-packages/rpyc/core/stream.py", line 102, in _connect
    s.connect(sockaddr)
  File "/usr/lib/python2.7/socket.py", line 224, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 111] Connection refused
Checking if daemon has stopped (please wait 10 seconds)........... No daemon running.
___

kizniche, 2015-12-14:
Those errors are expected if the daemon isn't already running. It's basically saying mycodo-client.py cannot connect to the daemon to tell it to shut down.

What about initialization?

```sudo /etc/init.d/mycodo initialize```

This should set cgi-bin/mycodo-wrapper suid root.

If initialization works, then attempt to start the daemon manually with:

```sudo /var/www/mycodo/cgi-bin/mycodo.py -l d```

Which should start it in debug mode.

kizniche, 2015-12-14:
Also, using ```-v``` will output to the terminal in addition to the log.

```sudo /var/www/mycodo/cgi-bin/mycodo.py -l d -v```

kizniche, 2015-12-14:
Also there is a Raspian Jessie Lite vrsion on the [download page](https://www.raspberrypi.org/downloads/raspbian/). This is what I used for my latest system install.

boomstick8x, 2015-12-14:
'sudo /var/www/mycodo/cgi-bin/mycodo.py -l d'
___
Traceback (most recent call last):
  File "/var/www/mycodo/cgi-bin/mycodo.py", line 42, in <module>
    import RPi.GPIO as GPIO
ImportError: No module named RPi.GPIO
___

kizniche, 2015-12-14:
AH. Compile wiringPi. From the README:

```git clone git://git.drogon.net/wiringPi ~/WiringPi && cd ~/WiringPi && sudo ./build```

boomstick8x, 2015-12-14:
Compiled already. I tried now again and even reboot, but got the same error.

kizniche, 2015-12-14:
Not sure what to say. That's where the module comes from.

Minibian Jessie is 200 MB, Rasobian Jessie Lite is 350 MB. Maybe try out the Lite version if you want Raspbian Jessie and don't mind another 150 MB. It's better than the 1.4 GB (is that correct) for the full version with the GUI installed.

kizniche, 2015-12-14:
There's something different in Minibian. I don't know enough about Python modules to help much more in the investigation.

kizniche, 2015-12-14:
Hmmm.. http://git.drogon.net/wiringPi

```The requested URL /wiringPi was not found on this server.```

kizniche, 2015-12-14:
How about:

```sudo pip install RPi.GPIO```

or

```sudo pip install --upgrade RPi.GPIO```

kizniche, 2015-12-14:
What do you get back from:

```pydoc RPi.GPIO```

boomstick8x, 2015-12-14:
You are right! RPi.GPIO was not installed. And after reboot all working now! Thank you man!

kizniche, 2015-12-14:
Great! Glad to hear. You're welcome.