kizniche/Mycodo #69: Timelapse not working like I'm expected)

boomstick8x, 2015-12-15:
Hi Kyle!
Something wrong with timelaps function or I misunderstood it behavior.
1. Even if Duration 60(min) camera led always! On after I submit 'Start timelaps' button.
2. And all file names have identical timestamps:
Timelapse20151215234309-00001.jpg
Timelapse20151215234309-00002.jpg
Timelapse20151215234309-00003.jpg

kizniche, 2015-12-15:
That is to be expected. Under "Camera: Time-lapse" on the Settings Tab, there are the options "Photo filename prefix" and "Time-lapse start time in filename" that you can set. The filename is where the photos are distinguished as a set, so they need both static (start data/time) and dynamic (incrementing number) identifiers in the filename to be able to determine the set and order within the set. You should be able to determine the time of each photo by the creation time, the exif data, or by multiplying the duration by the photo number and add that to the time it started.

Have you determined if your files are indeed different photos?

About the red light always being on, this may be how raspistill operated the camera. In any case, there may be a setting about this. You can put extra parameters in "Extra parameters for camera (raspistill)" under "Camera: Time-lapse"

boomstick8x, 2015-12-15:
Aha) Now Im understand. Thanks for explanations.
About red light... The reason of my asking not about just red light, but about continuous work of camera module even if duration is 5minutes...

kizniche, 2015-12-15:
When raspistill is creating a timelapse, the program continually runs until the end, so that could be why it requires a constant connection to the camera.

boomstick8x, 2015-12-15:
Yep, I understand that. Please close issue.

kizniche, 2015-12-15:
I've had the idea to build in the functionality of being able to use the camera and stream function while in the middle of a timelapse. If this gets done, there will need to be a change in how the timelapse functionality works. Instead of allowing raspistill to run int he timelapse mode, a timer somewhere will merely take a photo every period.

closed 2015-12-15 by kizniche

boomstick8x, 2015-12-15:
easiest way - crontab with bash script

kizniche, 2015-12-15:
Probably Python for the potential complex signalling required for pausing the timelapse to stream or take a still shot.

kizniche, 2015-12-15:
A PiCamera module for the Mycodo daemon, perhaps.