kizniche/Mycodo #70: Ajax/JQuery form not working as expected

kizniche, 2015-12-18:
Moving this discussion to a new issue, from [here](https://github.com/kizniche/mycodo_python/issues/4).

I put a test page up at https://fungi.kylegabriel.com/ajax-forms-jquery/ with the code from https://github.com/scotch-io/ajax-forms-jquery

Using the test code above works fine when I have only one form on the page. When I attempt to use this code, for instance, with one form on the Settings Tab, It works for that form, but all the other non-ajax forms cease to work (refresh the page). What happens is, when I click any other button, it will execute the javascript code and output the success message. This will cause the normal action of the forms to not work. I have experimented with adding id/class to the forms (as answered in several stackexchange posts) and changing the javascript to:

$('form#ajax-form').submit(function(event) {

or

$('.ajax-form').submit(function(event) {

But every other form continues to execute this code as though I submitted the ajax form. I'll admit I know very little javascript and will continue to work on this, but if anyone reading this sees the issue, please let me know how to fix it.

etiology, 2015-12-18:
So my guess is in your jQuery at: 

> $('form#ajax-form).submit(function(event) {

I think it should be: 

> $('#ajax_form).submit(function(event) {

-----------

##### A Flask Demo of ajax form submission

I don't use php much anymore but I wanted to post a demonstration of the ajax form submission in python.  Here is a flask app that I wrote to show how this might be done.  [Here](http://blog.miguelgrinberg.com/post/the-flask-mega-tutorial-part-i-hello-world) is a great guide to flask for anyone who is interested.  This replaces apache & php (and their nasty setup & config junk) in just a few lines of python.

Setup steps:
 - Copy the code below into a file, lets call it *ajax_demo.py*
 - Create a new python virtual environment
```bash
virtualenv flask_demo
```

 - Activate that virtual environment
```bash
source flask_demo/bin/activate
```

 - Install Flask

```bash
pip install flask
```

 - Run the code

``` bash
python ajax_demo.py
```

-----------------

##### Behold the entire web server & app... (mostly html)

```python
# coding=utf-8
"""
    Create a flask app with a form that is submitted using jQuery and ajax.
"""
from flask import Flask, request

# Create a simple flask app
app = Flask(__name__)

# Pretend we have some database to store our form submissions
db = {'value_submitted': None}

@app.route('/', methods=['GET', 'POST'])
def index():
    """ Simple web endpoint with a form """

    # This part is all of the jQuery needed for this app to work.  Notice that we use e.preventDefault()
    # to stop the browser from following the form submission to it's submission endpoint.
    ajax_submit_script = """
                            <!-- found at: http://hayageek.com/jquery-ajax-form-submit -->
                            <script>
                                $("#my_form_id").submit(function(e)
                                {
                                    var postData = $(this).serialize();
                                    var formURL = $(this).attr("action");
                                    $.ajax(
                                    {
[... 87 lines cut ...]

@app.route('/my_endpoint_for_post', methods=['POST'])
def endpoint_of_post():
    """
    Ajax form submission endpoint

    This endpoint accepts POST data and stores it in our dictionary db (defined at top).  The data in this dict
    is displayed on the index page ('/').  This requires a refresh since the index page only shows what was
    in the db at the time the template was rendered.
    
    This endpoint returns either 'success' or None.  I suspect returning None is not the right and might cause errors,
    but my goal was to show that data from one page can be posted to another
    endpoint using jQuery and ajax.
    
    """
    if request.method == 'POST':
        if "input_name_attribute" in request.form:
            db['value_submitted'] = request.form['input_name_attribute']

        return 'success'
    return None

if __name__ == '__main__':
    app.run()
```

boomstick8x, 2015-12-18:
@kizniche Can you provide full js code here?

etiology, 2015-12-18:
It's this part:

```javascript
<!-- found at: http://hayageek.com/jquery-ajax-form-submit -->
                            <script>
                                $("#my_form_id").submit(function(e)
                                {
                                    var postData = $(this).serialize();
                                    var formURL = $(this).attr("action");
                                    $.ajax(
                                    {
                                        url : formURL,
                                        type : "POST",
                                        data : postData,
                                        success : function(data, textStatus, jqXHR)
                                        {
                                            alert('Data was submitted');
                                        },
                                        error : function(jqXHR, textStatus, errorThrown)
                                        {
                                            alert('Data submission FAILED. Error was: ' + errorThrown);
                                        }
                                    });
                                    e.preventDefault();  //STOP default action
                                    e.unbind(); //unbind.
                                })
                            </script>
```

kizniche, 2015-12-18:
I just had a chance to test the Flask code. I'm not on my home network, so I could only use lynx, but I'm impressed. I'll play around with it later when I have more time.

closed 2016-07-12 by kizniche