kizniche/Mycodo #71: Force SSL/HTTPS with Flask

kizniche, 2015-12-19:
I've succesfully enabled SSL with the following edit to the code from [here](https://github.com/kizniche/Mycodo/issues/70#issuecomment-165681859):

```
if __name__ == '__main__':
    context = ('/etc/ssl/localcerts/apache.crt', '/etc/ssl/localcerts/apache.key')
    app.run(host='192.168.0.10', port=443, ssl_context=context, threaded=True, debug=False)
```

However, I am unable to both enable HTTP (port 80) and force a redirect to HTTPS (port 443) to require SSL use. I've attempted to use the [Flask-SSLify module](https://github.com/kennethreitz/flask-sslify) without success. There isn't much discussion about this, so I'll just continue my search/learning about Flask unless someone comes along that's familiar with how to do this.

closed 2016-07-02 by kizniche

wayne-david, 2023-11-25:
Hello @kizniche was this ever fixed on Mycodo.I get not secure but ssl certificate is already issued by mycodo and its not expired.

kizniche, 2023-11-25:
This is a 8 year old issue. This is not something you should be responding to about a question how something works. No, this is no longer an issue. If you have a specific question or can articulate a specific issue with adequately documented evidence, open a new issue.