kizniche/Mycodo #72: Add support to view Graphs tab with the last chosen settings

boomstick8x, 2016-01-07:
Hi Kyle!
When Im refreshing page with "Refresh Page" button - graph settings are going to their defaults(or when browsing page next time).
How about add an option(checkbox?) for save last chosen settings on Graph tab?
Or may be not an option, but fix for that(save last "graph generating" settings in database)?
What do you think about this enhancement?

kizniche, 2016-01-11:
The graph settings should be saved as a cookie and not be overwritten by using the "Refresh Page" button, from any tab. I just tested, and my graph settings don't get overwritten.

kizniche, 2016-01-11:
I'll add, there are only cookies saved for the static graph options and not the dynamic graph options. Were you referring to the dynamic graph options, or both static and dynamic?

boomstick8x, 2016-01-12:
Yes, Im talking about both options saving(dynamic and static) and also! about dynamic or static graph to shown on page by default(simply last choosen?)

closed 2016-07-02 by kizniche