kizniche/Mycodo #73: fresh install issue

boomstick8x, 2016-01-08:
Hi, Kyle!
I reinstalled system once again with instructions in README.md and got strange behavior of Relays GPIOs. I able to switch it on, but after that cant do anything with it. The button is 'Turn on' always 'Turn on' and led always red, but my relay is on! In daemon-tmp.log I get only messages 'Changing Relay 1 (Relay) to HIGH'.
I tried this part of code, that you removed in d73a44ac1e8aabc596827648370c78c72b6ac8d4 commit:
`git clone git://git.drogon.net/wiringPi ~/WiringPi && cd ~/WiringPi && sudo ./build`
and got working Relays.
Yep, this is my story :)

kizniche, 2016-01-08:
Did you also do the following (from the same commit)?

```
sudo pip install --upgrade lockfile rpyc pyserial tentacle_pi RPi.GPIO
```

boomstick8x, 2016-01-08:
I think I tried, its unbelievable to miss this string two times... Аlso tried `sudo pip install lockfile rpyc pyserial tentacle_pi` but no luck. I can try one more time fresh install to another sdcard for sure?

boomstick8x, 2016-01-08:
Ah! Sorry, dont saw '(from the same commit)', yes, I think I tried, but now will make one more install for shure...

boomstick8x, 2016-01-08:
Yes, I confirm the same story!
'sudo pip install --upgrade lockfile rpyc pyserial tentacle_pi RPi.GPIO' not work like expected.

boomstick8x, 2016-01-08:
'''
/var/log/mycodo/daemon-tmp.log                                                                                                                                     1492/1492              100%
2016-01-08 06:00:08,331 [INFO] [GPIO Initialize] Set GPIO mode to BCM numbering, all set GPIOs as output
2016-01-08 06:00:08,335 [INFO] [GPIO Initialize] Turning off all relays
2016-01-08 06:00:08,337 [INFO] [GPIO Initialize] Turning on all relays set to on at startup
2016-01-08 06:00:08,338 [INFO] [Daemon] Starting daemon
2016-01-08 06:00:08,340 [INFO] [Daemon] Starting communication server
2016-01-08 06:00:08,350 [INFO] server started on [0.0.0.0]:18812
2016-01-08 06:01:00,958 [INFO] accepted 127.0.0.1:55612
2016-01-08 06:01:00,967 [INFO] welcome [127.0.0.1]:55612
2016-01-08 06:01:01,001 [INFO] [Client command] Changing Relay 1 (Incub_Heater) to HIGH
2016-01-08 06:01:01,025 [INFO] goodbye [127.0.0.1]:55612
2016-01-08 06:01:11,148 [INFO] accepted 127.0.0.1:55613
2016-01-08 06:01:11,156 [INFO] welcome [127.0.0.1]:55613
2016-01-08 06:01:11,190 [INFO] [Client command] Changing Relay 1 (Incub_Heater) to HIGH
2016-01-08 06:01:11,202 [INFO] goodbye [127.0.0.1]:55613
2016-01-08 06:01:14,880 [INFO] accepted 127.0.0.1:55614
2016-01-08 06:01:14,888 [INFO] welcome [127.0.0.1]:55614
2016-01-08 06:01:14,923 [INFO] [Client command] Changing Relay 1 (Incub_Heater) to HIGH
2016-01-08 06:01:14,941 [INFO] goodbye [127.0.0.1]:55614
2016-01-08 06:03:53,988 [INFO] accepted 127.0.0.1:55619
2016-01-08 06:03:53,996 [INFO] welcome [127.0.0.1]:55619
2016-01-08 06:03:54,031 [INFO] [Client command] Changing Relay 1 (Incub_Heater) to HIGH
2016-01-08 06:03:54,042 [INFO] goodbye [127.0.0.1]:55619
'''

boomstick8x, 2016-01-10:
Aha, now I know whats going on...
in mycodo.php, line 663:
`if ((shell_exec($read) == 1 && $relay_trigger[$i] == 0) || (shell_exec($read) == 0 && $relay_trigger[$i] == 1)) {`
and: `$read = "$gpio_path -g read $relay_pin[$i]";` and `$gpio_path` is WiringPi GPIO utility.

boomstick8x, 2016-01-10:
And also  want to ask, when I adding new relay to some GPIO, I must restart Raspberry before it will work, is it possible to initialize it without restart?

kizniche, 2016-01-10:
I'm on the last day of a vacation since Dec 31st... Just letting you know I'll take a look at this soon, when I have the capability and time ;). From what I remember, the GPIO should be initialized when added. However, if that isn't working, instead of restarting the Pi, you should only have to restart the daemon.

boomstick8x, 2016-01-10:
Okay).
I will continue my investigation :).
The next thing that I found is about after adding new Relay in /var/log/mycodo/daemon-tmp.log
`
2016-01-10 19:37:29,015 [INFO] [Client command] Relay 7 GPIO pin changed to 0, initialize and turn off
2016-01-10 19:37:29,017 [INFO] [GPIO Initialize] Set GPIO mode to BCM numbering, GPIO 0 as output
2016-01-10 19:37:29,020 [INFO] [Client command] Reload SQLite database (Note: May cause interruption of sensor readings)
2016-01-10 19:37:29,426 [WARNING] [Read T Sensor-2] Could not read second Temp measurement!
2016-01-10 19:37:29,428 [WARNING] [PID T-Temperature-2] Could not read Temp sensor, not updating PID
2016-01-10 19:37:30,419 [INFO] goodbye [127.0.0.1]:34400
2016-01-10 19:38:07,895 [INFO] accepted 127.0.0.1:34402
2016-01-10 19:38:07,903 [INFO] welcome [127.0.0.1]:34402
2016-01-10 19:38:07,938 [INFO] [Client command] Changing Relay 8 (Relay) to HIGH
2016-01-10 19:38:08,014 [INFO] goodbye [127.0.0.1]:34402
`
Relay 7 and then Changing Relay 8?
Aha...
We need to add ` + 1` for correct message.
`Set GPIO mode to BCM numbering, GPIO 0`?
Aha...
We need move initialization function after get data from db.
p.s. Tested, now all working right. Will make pull request.

boomstick8x, 2016-01-10:
And in continue of this issue topic, do you plan to go away from php code at all and also in this part of mycodo.php?:
`if ((shell_exec($read) == 1 && $relay_trigger[$i] == 0) || (shell_exec($read) == 0 && $relay_trigger[$i] == 1))`

Do you already have part to read state of relay in new mycodo_client.py?(I didn't found it...)

kizniche, 2016-01-11:
There's some major changes to the framework being undertaken by @Cabalist so at the moment I'm keeping updates to only critical issues (no new features) until we get the new system running. Thanks for the work tracking down this issue. I've merged it in commit https://github.com/kizniche/Mycodo/commit/d42983dbb1499a434176663a7701ff088ac72ac5

closed 2016-01-11 by kizniche

boomstick8x, 2016-01-12:
You are welcome.
p.s. Okay then, I will try to understand and test Python RPyc and threading for that moment. Also really want to ask you about commenting code in future work and when refactoring old code to new system.