kizniche/Mycodo #75: Unable to set any configuration on website

lordzweigar, 2016-01-19:
I hate to have to ask, but i've spent far too many hours trying to fix this.

sys info:
mycodo 3.0 (latest as of (jan 20)
raspbian, kernel 4.1.13+
...
everything is identical as the readme (which is needs a bit of an update tbh)
except for one line in /etc/fstab:
/var/run/.... - this had to be disabled because it broke systemd, and would compeltely break the system.
anyways, my issue is that I cannot change any of the settings from the website, only with manually with vim ./config/mycodo.cnf

the file was chmod'd with 777, to no avail. there is nothing in the logs, i have no idea what to do.
And sorry if this poorley written, i really should sleep.

kizniche, 2016-01-20:
Mycodo 3.0 was built on Raspbian Wheezy, so if you're using Jessie, you'll need to figure out the issues that arise. I would recommend using Mycodo 3.5 if you're on Jessie, as it was built to work on that version. Currently, development on Mycodo 3.5 is at a standstill until the new framework is completed (https://github.com/kizniche/mycodo_python).

lordzweigar, 2016-01-21:
ty

closed 2016-02-05 by kizniche