kizniche/Mycodo #76: Special symbols in passwords are causing problems at log-in

MountAthos, 2016-01-28:
After setting the admin password via installation step:
sudo /var/www/mycodo/update-database.py -i
example of password: #pR*uDjbLK@Jb!!NElkJSpx73j
then due to special symbols are meaningful for the script itself, so it does process those wrong.
Please confirm also that database can keep this type of passwords, since it may interfere with the encryption of password script too.
Error will appear on log-in "wrong cookie data" 

TEMPORARY SOLUTION: Use passwords Alpha-Numeric only and if got this error already,
then re-run database initialization again and set new alpha-numeric password.|
In case of you set in browser to memorize the password, the browser have to be reset (empty cookies, and passwords for past hours), otherwise on page load the error will appear again.

kizniche, 2016-01-28:
Yes, passwords must be alphanumeric. This is not explicitly stated, but it is so.

This should be mentioned either in the readme or the script to create the password, or both.

MountAthos, 2016-01-29:
I believe that for good security all passwords shall be always mixed case , special symbols etc. especially if server is exposed to the public. IMHO complex , mixed case passwords are standard when it's about security.

kizniche, 2016-01-29:
I agree, but I needed an initial implementation that excluded special characters, to simplify development (and reduce risk of SQL injection). 

If you know of a solution, you're welcome to submit a pull request updating the database install and login scripts, so long as they follow standard security practices.

MountAthos, 2016-01-29:
Dear Kyle,
please tell me , the table in DB called "HTSensorConditional"
In which cases you are using to keep data in this one?
I need to understand better the idea of your structure.
I am implementing sensor according to the way you have made for other
sensor, with exception of multiplexer, because this sensor will mostly not
work with multiplexer.

I am in programming right now , so hope to get from you soon answer, just
don't want to remake later too much :)

Since i don't know all of your thoughts :) Nor long enough this project, i
am sure i will make not that perfect implementation, but i am trying to
understand it ...

Yours Seraphim.

kizniche, 2016-01-29:
This is specifically for a humidity and Temperature sensor (humidity and temperature on the same chip/board), that stores the variables for the conditional statements that can be created for a sensor. For instance, "Every **200** seconds, if the **temperature** is **above** **23** degrees, turn relay **2** **on** for **30** seconds"

MountAthos, 2016-01-29:
Ahh ok , i understood :)
So its actually correct to use HT tables for new sensor , since this one is
combined one too..
Just was not sure about the purpose for the conditional tables.
But now i understood.. thank you :)

commit 29349c3a (2017-01-05): Permit symbols to be used in user passwords (fixes #76).
closed 2017-01-05 by kizniche