kizniche/Mycodo #79: Strange behaviour with timers

lordzweigar, 2016-02-09:
Raspberry Pi 1B.
jessie, kernel 4.1.13+

I have been experiencing strange behaviour with the timers,
If i set a timer for one second on, and 1 off. my relays will be on for about a second, but will be off for 7.5
For most this might no be an issue, but as I am using this as a controller for HPA Aeroponics, sub-second timing would be preferable, and having such a great delay that doesn't seem to be consistent poses a serious issue. One other thing: it would be really nice to have a save all button on the sensors page, so that it doesn't have to reload after changing every little thing.
- regards LordZweigar

kizniche, 2016-02-09:
At the moment, there is a cycle in the daemon that requires the sensors to be read before timers are checked if they need to be turned on. So, the more sensors you have reading, the longer it will take to get to the timer check. This is something that is planned to be fixed in the next major code revision ([mycodo_python](https://github.com/kizniche/mycodo_python)).

As for the save all, this is difficult with the current implementation of the code, as it's unknown what has been changed. Each sensor has two parts, the sensor configuration and conditional statements. Any setting can be changed in the configuration part of each sensor and the Save button will save all changes (multiple).

kizniche, 2016-05-18:
@lordzweigar The next version of mycodo, my repository mycodo_python, has matured fairly well that I'd like to recommend you try it out if you're still looking for sub-second timing. I believe the timer functionality has an accuracy of ~0.001 seconds.

kizniche, 2016-07-02:
Fixed with update to [Mycodo version 4.0](https://github.com/kizniche/Mycodo/tree/master/4.0) (commit 1bf4971)

closed 2016-07-02 by kizniche