kizniche/Mycodo #80: I2C LCD support

kizniche, 2016-02-27:
I2C-interfaced LCDs are cheap these days, such as this [16x2 line LCD](http://www.sainsmart.com/new-sainsmart-iic-i2c-twi-1602-serial-lcd-module-display-for-arduino-uno-mega-r3.html) and this [20x4 line LCD](http://www.sainsmart.com/sainsmart-iic-i2c-twi-serial-2004-20x4-lcd-module-shield-for-arduino-uno-mega-r3.html).

The I2C address may or may not be changeable, however, the [TCA9548A I2C multiplexer](https://learn.adafruit.com/adafruit-tca9548a-1-to-8-i2c-multiplexer-breakout/overview), which is already [integrated into Mycodo, for I2C sensors](https://github.com/kizniche/Mycodo/tree/master/3.5#i2c-multiplexer), could be easily integrated and allow LCDs to be used out of the box to connect up to 64 LCDs on the same address (if you don't want to go through the trouble of figuring out how to change I2C addresses, if it's even possible).

What I envision as a base configuration is a selection of how many LCDs are connected, the I2C address and whether a multiplexer is used (and it's I2C address) for each, a selection of available measurement or measurements, the cycle style, and how fast to cycle. The cycle style for each measurement (or set) could be simply a static display from one to another (e.g. display sensor1 temperature and humidity for 5 seconds, display sensor2 temperature and humidity for 5 seconds, repeat), scroll all measurements left, or scroll all measurements up.

kizniche, 2016-02-28:
I found for these boards, [changing the I2C address](https://arduino-info.wikispaces.com/LCD-Blue-I2C#toc14) can be accomplished by changing three jumpers, allowing up to 8 unique addresses. So, if you only require up to 8 LCDs, a multiplexer shouldn't be necessary.

Myke974, 2016-02-28:
Just for you to know that I'm totally supporting thisenvision : been abble
to see the sensors reading(and more) on 1 (or more) small LCD !
Way to go ! ;) :D

kizniche, 2016-02-29:
Thanks to a generous donor, I'll have a few of these LCDs very soon to play with.

kizniche, 2016-03-11:
First working implementation with commit https://github.com/kizniche/Mycodo/commit/6066018b5c387a60c7326c5b78f13d496799700f

Only 16x2 character I2C LCDs supported, and only able to display temperature sensor data, with up to two measurements (different sensors), one on each line.

kizniche, 2016-03-12:
Full support for all sensors/measurements with commit https://github.com/kizniche/Mycodo/commit/f6dd0c9e9712784ce2494fb88d6804a443aaff8d

More fancy options (like custom text and more than two measurements) coming in the future.

closed 2016-03-12 by kizniche