kizniche/Mycodo #81: Can't create file /home/pi/Mycodo/3.5/public_html/../log/auth.log

zsole2, 2016-03-12:
Kyle,
Freshly installed v3.5.96. 
After login, I get the error message on the page, nothing else. That path looks certainly incorrect. I guess it is a bug from the directory restructuring a few weeks ago.
Then if I refresh the page, it loads regularly, but nothing is saved, if I try to add a sensor, relay, anything.

kizniche, 2016-03-12:
The path is correct, but the file was unable to be created through the PHP command fopen(). I've moved the creation to the init.d script, which runs with root access, then sets the permissions of all files. Commit https://github.com/kizniche/Mycodo/commit/8abf01952248dc6da16ebf3d14708bea0179920f should fix this issue. Let me know if a system update works.

zsole2, 2016-03-12:
Thanks for the new commit, but the error is still there.
The file auth.log is created in the log directory, though. It is owner by root, as expected.

kizniche, 2016-03-12:
Are you having mycodo start by "sudo /etc/init.d/mycodo start"?

This is how it should at startup. Included in 'start' is the initialize command, that ensures files, directories, and symlinks are created and then sets all the appropriate permissions. If you're not using the init.d script, I'd recommend running "sudo /etc/init.d/mycodo initialize" at least once.

If you've done all this, and the error is still there, I'm stumped and will have to investigate when I get to a computer.

zsole2, 2016-03-12:
Actually I started from scratch: deleted Mycodo directory, re-downloaded from github, and followed the steps from the Read.me

kizniche, 2016-03-12:
Give it a try now. I'm not sure if what I did to the init.d script will even make a difference. The script is creating auth.log for me.

zsole2, 2016-03-13:
Thanks!
I can conform that this fixed the problem.
Zsolt

kizniche, 2016-03-13:
Great!

closed 2016-03-13 by kizniche