kizniche/Mycodo #82: add ability to read signal from analog sensors

itzik16, 2016-05-28:
Hello, I dig you script man its really great, though I think it would be great if it could monitor values from analog sensors, like light sensor or from a switch to know if the door is open and such, or even for an soil moisture, I will really appreciate it if you could explain me which files do i need to modify and add in order to do this, and even more appreciate it if you will add it to the code, I will even donate you 100$ if will do so :) thanks in advance

kizniche, 2016-05-28:
Hi @itzik16 Thanks for the support. 

I've actually started working with an I2C analog to digital converter. The MCP3424, a 4-channel 18-bit, I2C ADC. [Working MCP3424 test script](https://github.com/kizniche/mycodo_python/blob/master/mycodo/tests/Test_I2C_MCP3424.py).

I still need to get some analog sensors for testing before I can integrate their support in Mycodo, but I have the hard part out of the way (reading the ADC). I'm no longer developing Mycodo 3.0 (except for bugs), and I'm focusing most of my attention on the next version, currently called 3.6 over at [mycodo_python](https://github.com/kizniche/mycodo_python).

I'll order some sensors this weekend. I've wanted to get an indoor plant nursery set up, so that will be my test bed (no pun intended). As you've suggested, lux, reed, and moisture sensors are all great sensors to start with. Do you have any recommendations for specific devices or any other sensors that I should pick up to test with?

itzik16, 2016-05-28:
Hi @kizniche Thanks for the fast reply, Im really happy to know that you are working on that :)
and in fact I have a recommendations for specific sensors, simple ones but important :)
https://www.sparkfun.com/products/13322 - Soil Moisture Sensor
And a water level float switch
those two sensors are all i need to monitor, thanks for sharing this great code, Keep on rocking man!

kizniche, 2016-05-29:
I just completed an order for a simple reed float sensor and the following.

[image]

jj2473, 2016-05-29:
I would have provided magnetic door switches. I have them from when I got started with Mycodo and it was on my wish list, lol. I will help test as well.

kizniche, 2016-07-02:
Full support with update to [Mycodo version 4.0](https://github.com/kizniche/Mycodo/tree/master/4.0) (commit 1bf4971)

closed 2016-07-02 by kizniche

itzik16, 2016-07-02:
You are the best mate :+1: