kizniche/Mycodo #84: wsgi:error TypeError: 'NoneType' object is not iterable when creating "graph"

hamster007, 2016-06-24:
[Fri Jun 24 10:44:50.025570 2016] [wsgi:error] [pid 19346:tid 1962841136] [remote 192.168.1.70:31606] [2016-06-24 10:44:50,006] ERROR in app: Exception on /graph [POST]

[Fri Jun 24 10:44:50.027493 2016] [wsgi:error] [pid 19346:tid 1962841136] [remote 192.168.1.70:31606] Traceback (most recent call last):

[Fri Jun 24 10:44:50.027614 2016] [wsgi:error] [pid 19346:tid 1962841136] [remote 192.168.1.70:31606]   File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1988, in wsgi_app

[Fri Jun 24 10:44:50.027674 2016] [wsgi:error] [pid 19346:tid 1962841136] [remote 192.168.1.70:31606]     response = self.full_dispatch_request()

[Fri Jun 24 10:44:50.027712 2016] [wsgi:error] [pid 19346:tid 1962841136] [remote 192.168.1.70:31606]   File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1641, in full_dispatch_request

[Fri Jun 24 10:44:50.027749 2016] [wsgi:error] [pid 19346:tid 1962841136] [remote 192.168.1.70:31606]     rv = self.handle_user_exception(e)

[Fri Jun 24 10:44:50.027801 2016] [wsgi:error] [pid 19346:tid 1962841136] [remote 192.168.1.70:31606]   File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1544, in handle_user_exception

[Fri Jun 24 10:44:50.027828 2016] [wsgi:error] [pid 19346:tid 1962841136] [remote 192.168.1.70:31606]     reraise(exc_type, exc_value, tb)

[Fri Jun 24 10:44:50.027870 2016] [wsgi:error] [pid 19346:tid 1962841136] [remote 192.168.1.70:31606]   File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1639, in full_dispatch_request

[Fri Jun 24 10:44:50.027896 2016] [wsgi:error] [pid 19346:tid 1962841136] [remote 192.168.1.70:31606]     rv = self.dispatch_request()

[Fri Jun 24 10:44:50.027940 2016] [wsgi:error] [pid 19346:tid 1962841136] [remote 192.168.1.70:31606]   File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1625, in dispatch_request

[Fri Jun 24 10:44:50.027967 2016] [wsgi:error] [pid 19346:tid 1962841136] [remote 192.168.1.70:31606]     return self.view_functions[rule.endpoint](**req.view_args)

[Fri Jun 24 10:44:50.028006 2016] [wsgi:error] [pid 19346:tid 1962841136] [remote 192.168.1.70:31606]   File "/var/www/mycodo/mycodo/mycodo_flask.py", line 195, in page

[Fri Jun 24 10:44:50.028043 2016] [wsgi:error] [pid 19346:tid 1962841136] [remote 192.168.1.70:31606]     flaskutils.graph_add(formAddGraph, display_order)

[Fri Jun 24 10:44:50.028079 2016] [wsgi:error] [pid 19346:tid 1962841136] [remote 192.168.1.70:31606]   File "/var/www/mycodo/mycodo/flaskutils.py", line 473, in graph_add

[Fri Jun 24 10:44:50.028115 2016] [wsgi:error] [pid 19346:tid 1962841136] [remote 192.168.1.70:31606]     elif formAddGraph.validate():

[Fri Jun 24 10:44:50.028152 2016] [wsgi:error] [pid 19346:tid 1962841136] [remote 192.168.1.70:31606]   File "/usr/local/lib/python2.7/dist-packages/wtforms/form.py", line 310, in validate

[Fri Jun 24 10:44:50.028201 2016] [wsgi:error] [pid 19346:tid 1962841136] [remote 192.168.1.70:31606]     return super(Form, self).validate(extra)

[Fri Jun 24 10:44:50.028276 2016] [wsgi:error] [pid 19346:tid 1962841136] [remote 192.168.1.70:31606]   File "/usr/local/lib/python2.7/dist-packages/wtforms/form.py", line 152, in validate

[Fri Jun 24 10:44:50.028301 2016] [wsgi:error] [pid 19346:tid 1962841136] [remote 192.168.1.70:31606]     if not field.validate(self, extra):

[Fri Jun 24 10:44:50.028323 2016] [wsgi:error] [pid 19346:tid 1962841136] [remote 192.168.1.70:31606]   File "/usr/local/lib/python2.7/dist-packages/wtforms/fields/core.py", line 193, in validate

[Fri Jun 24 10:44:50.028373 2016] [wsgi:error] [pid 19346:tid 1962841136] [remote 192.168.1.70:31606]     self.pre_validate(form)

[Fri Jun 24 10:44:50.028396 2016] [wsgi:error] [pid 19346:tid 1962841136] [remote 192.168.1.70:31606]   File "/usr/local/lib/python2.7/dist-packages/wtforms/fields/core.py", line 502, in pre_validate

[Fri Jun 24 10:44:50.028419 2016] [wsgi:error] [pid 19346:tid 1962841136] [remote 192.168.1.70:31606]     values = list(c[0] for c in self.choices)

[Fri Jun 24 10:44:50.028442 2016] [wsgi:error] [pid 19346:tid 1962841136] [remote 192.168.1.70:31606] TypeError: 'NoneType' object is not iterable

kizniche, 2016-06-24:
Do you have it updated to the latest version (Admin->Update)?

hamster007, 2016-06-24:
Brilliant, that did the trick! Thanks very much

kizniche, 2016-06-24:
Great! I had thought I fixed that bug last night ;)

closed 2016-06-24 by kizniche