kizniche/Mycodo #85: Questions :P

itzik16, 2016-07-04:
Hello Kyle,
I've got some questions that are might sound very basic.
I've seen the addition of Wireless device support (z-wave) on you'r todo list, what devices you are testing with, could you specify the model and vendor that you bought it from I would like to add this feature on my setup with the CO2 sensor maybe to do some trail and error to find where is right to place it. 
I've got another question that might sound silly, does the i2c multiplexer allows me to connect few DHT22 sensors to it or is it for other type of sensors, either way i want to know what is the optimal way to connect the DHT22 sensors Ive got 4 of them and I like the option of adding more ;)

I Really appreciate your help and kindness thanks a lot mate, Keep up the good work.

kizniche, 2016-07-05:
I don't currently have any z-wave devices, so at the moment it's just an idea. I'm not sure what you're trying to do with your CO2 sensor, but if you elaborate I may be able to help.

The I2C multiplexer helps when you have multiple sensors with the same I2C address (like the AM2315 temperature/humidity sensor). Normally, you would only be able to use only one of these sensors (because there is only one address), however, if you connect them to the multiplexer, you can read all of them. The multiplexer has a changeable address (8 addresses) and 8 channels per multiplexer, so you could theoretically have up to 64 sensors with the same I2C address in use at the same time.

The DHT22 communicates directly with the GPIO, so all you need to do is make sure you connect it to a free GPIO pin. I'm sure there are ways to expand the number of free GPIOs but I'm not that familiar with it. I doubt you'll be maxing out your free GPIO pins, so you don't really have to worry about any extra hardware to support more DHT22 sensors. Below is a diagram for connecting the DHT22.

[image]

itzik16, 2016-07-06:
Thanks a lot for explaining me this mate, looking forward to see what features you will add, this script is just awesome and very useful, great job!! you are the best

kizniche, 2016-07-06:
I may be putting Mycodo development on hold for a while. Things just got a little complicated...  https://www.reddit.com/r/motorcycles/comments/4rj1kq/stolen_ktm_690_enduro_2012_in_atlanta_ga_please/

I'll still respond to messages, but I may be preoccupied with sorting transportation out for a while. But who knows, I may end up channeling my frustration and anger into coding some kickass features.

closed 2016-07-19 by kizniche