kizniche/Mycodo #86: new install - requirement.txt problem

zsole2, 2016-07-13:
I did a fresh install both yesterday and today (don't ask, long story...)

It worked flawlessly yesterday, today I had a lot of problems.

The most evident is the following: 
 running ' sudo pip install -r ~/Mycodo/requirements.txt --upgrade --force-reinstall" gives the error:

Collecting adafruit-dht (from -r /home/pi/Mycodo/requirements.txt (line 21))
  Could not find a version that satisfies the requirement adafruit-dht (from -r /home/pi/Mycodo/requirements.txt (line 21)) (from versions: )
No matching distribution found for adafruit-dht (from -r /home/pi/Mycodo/requirements.txt (line 21))
 
It must be because of  your changes to requirements.txt   

******************

On the other hand, the daemon did not start, even with this reinstall.
Finally I ended up doing 'sudo pip install' for each items, until it worked. It took a little time...

If you see something evident from this info, that's good. Otherwise I'll have to check my install tomorrow to see what could happened... (Nothing deters me after burning myself with a smoking microSD card  trying to rescue my Pi3 ... it succeeded)

kizniche, 2016-07-13:
I'm not sure what's going on with requirements.txt. I Thought these were the same repositories. I restored the git locations instead of relying on the pip packages. This should fix it.

Smoking microSD card?? I gotta hear this.

closed 2016-07-13 by kizniche

zsole2, 2016-07-14:
I reinstalled on top of the previous, and can confirm it works now.
One thing: when trying to log in, I got the internal server error. Run flask in debug mode, and turned out that it was the cookie error. In the old version, this situation was immediately evident, instead of this cryptic error.

So, the smoking card.... It was from Ebay, nicely "rebranded" from 8GB to 128GB. Still tried to use it, 8GB should be plenty, right? 
This was the system that you have access a few weeks back, and already starting misbehaving that time. Now, it completely stopped. I went to the basement to check it, unplug-replug the power a few times, nothing happened. Then, the Pi blinked a few times, and I thought that it was same contact problem. 
It was not, the smoke came, and I pulled the plug. The mistake was to try to remove the card to soon, it was burning hot still, the Pi enclosure too. Fortunately, the Pi still seems to be working. 
Never seen/heard this before!

commit f16e65c6 (2016-07-13): Update requirements.txt to fix #86