kizniche/Mycodo #87: Raspberry Pi 3 fresh install jessie

hamster007, 2016-07-16:
pi@raspberrypi:/var/log $ ~/Mycodo/init_databases.py -i all
  Traceback (most recent call last):
  File "/home/pi/Mycodo/init_databases.py", line 33, in <module>
  from mycodo.databases.users_db.models import Users
  File "/home/pi/Mycodo/mycodo/databases/users_db/__init__.py", line 28, in <module>
  from .models import Base
  File "/home/pi/Mycodo/mycodo/databases/users_db/models.py", line 27, in <module>
  import bcrypt
  File "/usr/local/lib/python2.7/dist-packages/bcrypt/__init__.py", line 24, in <module>
  from bcrypt import _bcrypt
  ImportError: No module named _cffi_backend

hamster007, 2016-07-16:
I found this in the syslog if it helps..

Jul 16 14:14:56 raspberrypi systemd[1]: Starting Mycodo server...
Jul 16 14:14:57 raspberrypi mycodo_daemon.py[1784]: Traceback (most recent call last):
Jul 16 14:14:57 raspberrypi mycodo_daemon.py[1784]: File "/var/www/mycodo/mycodo/mycodo_daemon.py", line 21, in <module>
Jul 16 14:14:57 raspberrypi mycodo_daemon.py[1784]: import daemonutils
Jul 16 14:14:57 raspberrypi mycodo_daemon.py[1784]: File "/home/pi/Mycodo/mycodo/daemonutils.py", line 6, in <module>
Jul 16 14:14:57 raspberrypi mycodo_daemon.py[1784]: import geocoder
Jul 16 14:14:57 raspberrypi mycodo_daemon.py[1784]: File "/usr/local/lib/python2.7/dist-packages/geocoder/__init__.py", line 36, in <module>
Jul 16 14:14:57 raspberrypi mycodo_daemon.py[1784]: from geocoder.api import get, yahoo, bing, geonames, mapquest, google, mapbox  # noqa
Jul 16 14:14:57 raspberrypi mycodo_daemon.py[1784]: File "/usr/local/lib/python2.7/dist-packages/geocoder/api.py", line 6, in <module>
Jul 16 14:14:57 raspberrypi mycodo_daemon.py[1784]: from geocoder.osm import Osm
Jul 16 14:14:57 raspberrypi mycodo_daemon.py[1784]: File "/usr/local/lib/python2.7/dist-packages/geocoder/osm.py", line 5, in <module>
Jul 16 14:14:57 raspberrypi mycodo_daemon.py[1784]: import ratelim
Jul 16 14:14:57 raspberrypi mycodo_daemon.py[1784]: ImportError: No module named ratelim
Jul 16 14:14:57 raspberrypi systemd[1]: mycodo.service: control process exited, code=exited status=1
Jul 16 14:14:57 raspberrypi systemd[1]: Failed to start Mycodo server.
Jul 16 14:14:57 raspberrypi systemd[1]: Unit mycodo.service entered failed state.

Thanks,
Mike

kizniche, 2016-07-16:
Make sure you didn't skip the install of libffi-dev.

```sudo apt-get install libffi-dev```

If that doesn't fix it, then try:

```sudo pip install -r ~/Mycodo/requirements.txt --upgrade --force-reinstall --no-deps```

hamster007, 2016-07-17:
Good afternoon,

sudo apt-get install libffi-dev
Then
sudo pip install -r ~/Mycodo/requirements.txt --upgrade --force-reinstall --no-deps

Still no change giving same errors as above.

Please find attached the log i created for output of the above.

Its complaining about MCP432x module but renamed .py to suite requirements.txt, file definitely there but install script cannot see it.
[Mycodo-reinstall-log.pdf](https://github.com/kizniche/Mycodo/files/367647/Mycodo-reinstall-log.pdf)

Thanks,
Mike

kizniche, 2016-07-17:
Try deleting your Mycodo directory and cloning again and following the new set of install instructions I just finished pushing to github. I think I managed to create a script that will entirely install Mycodo.

hamster007, 2016-07-17:
Hello, i did what you suggested, please see attached:
Same error.

Thanks,
Mike
[2ndMycodo-reinstall-log-setup-sh.pdf](https://github.com/kizniche/Mycodo/files/367656/2ndMycodo-reinstall-log-setup-sh.pdf)

kizniche, 2016-07-17:
Can you try to install Mycodo again after a fresh install of Raspbian?

hamster007, 2016-07-17:
Will do, ill post results shortly.

Thanks,
Mike.

kizniche, 2016-07-17:
I have a RPi 1, 2, and 3 that I test on, and it's working on all, so probably not because of that.

hamster007, 2016-07-17:
Hello,

Just did a complete reinstall as suggested, please see log at the end new error.

[fresh-install-setup-sh.pdf](https://github.com/kizniche/Mycodo/files/367668/fresh-install-setup-sh.pdf)

Thanks,
Mike.

kizniche, 2016-07-17:
An error is stopping proper install of the script. I changed how it's executed so an error will compeletely stop the script and be obvious. Try this:

```
cd ~/Mycodo
git fetch
git reset --hard origin/master
```

Then

```
sudo ./setup.sh
```
cd ~/Mycodo

hamster007, 2016-07-17:
Hello,

Yes it aborted setup:

[setup-sh-aborted.pdf](https://github.com/kizniche/Mycodo/files/367673/setup-sh-aborted.pdf)

Thanks.

kizniche, 2016-07-17:
And you started with a completely fresh install of raspbian? A freshly-burned image?

hamster007, 2016-07-17:
Yes, i even used my Macbook using dd instead of Windows 7 disk imaging util, latest Rasbian Jessie image.

Do you have a already setup/configured install of Mycodo as an image? i could dd this to scared turn on start setting up sensors/relays etc...

Thanks,
Mike

kizniche, 2016-07-17:
I'm going to start with a fresh install of raspbian myself on my Pi 3 and see if I run into any errors. I've started fresh at least 5 times today and worked the last few times.

kizniche, 2016-07-17:
And you're starting with Raspbian Jessie Light (2016-05-27)?

hamster007, 2016-07-17:
Yes, i will re download image and try again aswell

kizniche, 2016-07-17:
I just had a flawwless install on a fresh raspbian jessie lite image.

hamster007, 2016-07-17:
[Untitled 4.pdf](https://github.com/kizniche/Mycodo/files/367700/Untitled.4.pdf)

I will try the lite version of jessie.

kizniche, 2016-07-17:
I don't know why but it looks like you're installing an old version of pip. From your log:

```
Searching for pip
Best match: pip 1.5.6
Adding pip 1.5.6 to easy-install.pth file
Installing pip script to /usr/local/bin
Installing pip2.7 script to /usr/local/bin
Installing pip2 script to /usr/local/bin
```

My log excerpt (full: [setup.log.zip](https://github.com/kizniche/Mycodo/files/367701/setup.log.zip)):

```
Searching for pip
Reading https://pypi.python.org/simple/pip/
Best match: pip 8.1.2
Downloading https://pypi.python.org/packages/e7/a8/7556133689add8d1a54c0b14aeff0acb03c64707ce100ecd53934da1aa13/pip-8.1.2.tar.gz#md5=87083c0b9867963b29f7aba3613e8f4a
Processing pip-8.1.2.tar.gz
Writing /tmp/easy_install-U1XpKu/pip-8.1.2/setup.cfg
Running pip-8.1.2/setup.py -q bdist_egg --dist-dir /tmp/easy_install-U1XpKu/pip-8.1.2/egg-dist-tmp-uv74Bs
warning: no previously-included files found matching '.coveragerc'
warning: no previously-included files found matching '.mailmap'
warning: no previously-included files found matching '.travis.yml'
warning: no previously-included files found matching '.landscape.yml'
warning: no previously-included files found matching 'pip/_vendor/Makefile'
warning: no previously-included files found matching 'tox.ini'
warning: no previously-included files found matching 'dev-requirements.txt'
warning: no previously-included files found matching 'appveyor.yml'
no previously-included directories found matching '.github'
no previously-included directories found matching '.travis'
no previously-included directories found matching 'docs/_build'
no previously-included directories found matching 'contrib'
no previously-included directories found matching 'tasks'
no previously-included directories found matching 'tests'
Adding pip 8.1.2 to easy-install.pth file
Installing pip script to /usr/local/bin
Installing pip2.7 script to /usr/local/bin
Installing pip2 script to /usr/local/bin
```

kizniche, 2016-07-17:
You're running this first, right?

```
sudo apt-get update && sudo apt-get upgrade -y && sudo apt-get install -y git
```

kizniche, 2016-07-17:
Perhaps you can update pip with:

```sudo pip install -U pip```

hamster007, 2016-07-17:
Ok worked straight out of the box, just like you said. I think there is a problem with installation on full size distro of jessie vs lite version.

Thanks for your assistance,

Mike.

kizniche, 2016-07-17:
Well, great! But that's something that's annoying. I'll have to go and install the large image and mess with it. At least I think I narrowed it down to the pip version.

closed 2016-07-17 by kizniche