kizniche/Mycodo #88: RF Power?

berserktron3k, 2016-07-16:
I'm relatively new to the Pi (intermediate at Ubuntu), and new to the sensors, etc., however I pick up quick.

I stumbled upon your project, and a sincere kudos to you. This is amazing. Sorry to hear of your recent loss of your bike. That sucks big time.

I think my question is simple, so sorry for being a noob, but I bought a SMAKN Rf transmitter and receiver to work with some Etekcity Rf controlled power outlets (poor mans automation). Is there support already in Mycodo for something like that? I think I may just not have the right terminology and not be looking in the right place.

Again, great work with this. It's pretty amazing.

kizniche, 2016-07-16:
Hey. Thanks for the support. You don't have to be shy about asking questions here. One of the reasons I do this (make everything open source) is to help people learn about how things work. I wouldn't be half as knowledgeable as I am with this and Mycodo may not even have been possible if there weren't people answering seemingly simple stuff that I had no idea about.

I haven't heard of the wireless you mentioned. I looked it up but will have to read more about it. However, I have been wanting to integrate wireless support for controlling devices and measuring sensors from afar. I've been looking at z-wave but am open to other considerations. 

My main selection criteria for choosing which wireless to support has been (in no particular order):
1) The amount of devices already on the market (the more the better)
2) Security (prevent hackers from reading/mimicking signals)
3) Ease of integration/implementation
4) Cost
5) Signal distance/spectrum/noise (basically the best band and power combination)

Have you played with your transceiver to communicate with your outlets yet? What I would be interested in to be able to integrate into Mycodo is Python code specific for the Raspberry Pi.

kizniche, 2016-07-16:
Does [this tutorial](https://www.raspberrypi.org/forums/viewtopic.php?f=37&t=66946) use the transmitter/receiver you're talking about?

kizniche, 2016-07-16:
I just bought a few testing supplies :)

[image]

berserktron3k, 2016-07-16:
I had to laugh when I read this. Within 17 minutes you had found exactly
what I was referencing and purchased it already. Pretty amazing. I figured
maybe I just didn't know what terminology to use.

I haven't yet tested the RF transmitter/receiver with my Pi. I realized I
forgot to buy the wires, or a breadboard, so I'm waiting on those to come
in.

I'm struggling with getting the camera to work in Mycodo as well. Would you
have time to give me some pointers for that (or should that be a separate
issue)? I know the camera is good. I can take a still using picamera via
the command line (using a headless install of Jessie), however in the
Mycodo web interface, I keep getting (wiped the Pi twice and followed to a
t (I'm pretty sure) your install instructions):
Internal Server Error

The server encountered an internal error or misconfiguration and was unable
to complete your request.

Please contact the server administrator at [no address given] to inform
them of the time this error occurred, and the actions you performed just
before this error.

More information about this error may be available in the server error log.
------------------------------
Apache/2.4.10 (Raspbian) Server at 10.10.0.114 Port 80
Thanks again for all you work on this project. Simply amazing.

Brady Hummel

berserktron3k, 2016-07-19:
Got my breadboard and wires today. Will test with RF tomorrow.

kizniche, 2016-07-19:
Awesome! If you use any python code to get it working, please post it here.

berserktron3k, 2016-07-19:
Had a few minutes to play with this. Dead simple from CLI (using this: https://www.samkear.com/hardware/control-power-outlets-wirelessly-raspberry-pi).

I don't know the slightest bit about Python, unfortunately. I'm an infrastructure guy in IT, not a programmer (Powershell is about all I know). Found this (http://www.instructables.com/id/Super-Simple-Raspberry-Pi-433MHz-Home-Automation/?ALLSTEPS) that may be of some help?

Looks like you bought exactly the same equipment as I did. Each outlet has a label on the back that the last digit corresponds to the number on the remote, as an FYI. I went ahead and labeled the top of my outlets. Got a decent range (as far as my needs go) without having to add an antenna. Shot the signal 5 feet no problem. Considering my Pi will be less than a few feet away from these, these things will do quite nicely.

Would be awesome if a day/time trigger could fire an on/off on these, as well as triggers on humidity, or even CO2. Again, thanks for putting this project together. Very awesome design.

berserktron3k, 2016-08-10:
So, I finally decided to do what I did to learn PowerShell, and ripped off examples from the web... here's what I've come up with so far:
1. cd
2. git clone git://github.com/timleland/rfoutlet.git
3. sudo chown root.root /home/pi/rfoutlet/codesend
4. sudo chmod 4755 /home/pi/rfoutlet/codesend
5. sudo /var/www/rfoutlet/RFSniffer to find the remote codes
6. Wrote the following simple python script:
`#!/usr/bin/python
import sys
import Adafruit_DHT
import subprocess
humidity, temperature = Adafruit_DHT.read_retry(22, 4)
if humidity < 80:
        subprocess.call(["/home/pi/rfoutlet/codesend", "1111111"])
        sys.exit(1)
else:
        subprocess.call(["/home/pi/rfoutlet/codesend", "1111112"])
        sys.exit(1)`
7. Then put it in cron to run every 5 mins.

It's not very intelligent. It simply checks the humidity every 5 mins, then either sends the code to turn it on, or send the code to turn it off based on the reading.

Also, happy to hear the camera was a simple fix! Just updated, and it works great. Streaming is very laggy, but not much of a concern, in my opinion.

berserktron3k, 2016-08-10:
The formatting got a bit screwy on the post above for the code, sorry about that. Still new to this stuff.

berserktron3k, 2016-08-10:
Wow... it's even simpler than that. I simply added a conditional in the sensor to run the external command.

kizniche, 2016-08-10:
I was actually thinking today how handy it would be for the ability to pass the sensor value to the command execution, for instance "/home/user/script.sh --temperature {{value}}" and it replace {{value}} before executing the command.

kizniche, 2016-08-10:
I updated your comment ;)
Thanks for the code and steps! The next thing I want to work on is the wireless. I've had my power outlets and transceiver here for the past few weeks, but I've been out of town and then busy since getting back.

kizniche, 2016-08-10:
This is the Python library I've been looking to use: https://github.com/lexruee/pi-switch-python
I'm going to finish some soldering, then get this hooked up.

berserktron3k, 2016-08-10:
Well... I think it may be even simpler. I just realized I could put 2 conditionals on the sensor so that if humidity is below x% for 10 mins, then execute the command to send the code to turn on the humidifier. Then a second conditional that says if humidity is above x% for 1 minute, execute the command to turn off the humidifier.

It still relies on an external command, but it works.

I think if you're going to spend time on this, then maybe add these as relays? It'd be nice to be able to kick on the relay for durations of time, as well as based on conditional values.

Anyway, again, GREAT work with this. It's truly an amazing work you're doing.

kizniche, 2016-08-10:
The plan was to have a relay option for a 433MHz transmitter, with the ability to set the on/off codes, and a way to reprogram the outlet with a new code.

kizniche, 2016-08-10:
Alright, using pi-switch, I was able to map the codes from my remote with this code:
```Python
#!/usr/bin/python
# coding=utf-8
from pi_switch import RCSwitchReceiver
receiver = RCSwitchReceiver()
receiver.enableReceive(2)
num = 0
while True:
    if receiver.available():
        received_value = receiver.getReceivedValue()
        if received_value:
            num += 1
            print("Received[%s]:" % num)
            print(received_value)
            print("%s / %s bit" % (received_value, receiver.getReceivedBitlength()))
            print("Protocol: %s" % receiver.getReceivedProtocol())
            print("")
        receiver.resetAvailable()
```
Which gave this type of output when I pressed a button on my remote:
```
Received[86]:
29955
29955 / 24 bit
Protocol: 1

Received[87]:
29964
29964 / 24 bit
Protocol: 1
```
Leading to a full table for all 5 outlets:
```
	On		Off
1	21811	21820
2	21955	21964
3	22275	22284
4	23811	23822
5	29955	29964
```
I'm using the following code to transmit:
```Python
#!/usr/bin/python
# coding=utf-8
import pi_switch
sender = pi_switch.RCSwitchSender()
sender.enableTransmit(2)
sender.setProtocol(1)
sender.sendDecimal(29955, 24) # switch on
```
However, my outlet isn't turning on. I've soldered a 15cm antenna, so I doubt the 3 feet from the outlet is an issue.

berserktron3k, 2016-08-10:
Sorry for simple questions, but have you tried another pin to rule that out?

kizniche, 2016-08-10:
I've moved it to different pins, tried both 3.3v and 5v. I think I might have a bad transmitter. I've also tried ```codesend```, without success.

berserktron3k, 2016-08-10:
That's what I was getting at. I've heard rumors of them being bad out of the gate. I ordered a second one I can send you when it arrives, if you'd like.

kizniche, 2016-08-10:
Funny that the first US seller I found on eBay is only 20 minutes north of where I live. Thanks for the offer, but they're so cheap, I just went ahead and bought two more sets. If they both end up working, I'll play with wireless transmission between Pis.

berserktron3k, 2016-08-10:
I tried the code you listed. Can't seem to get it to send the signal properly either. But mine does work using codesend.

kizniche, 2016-08-10:
Well, damn... I tried rfoutlet's codesend and it worked!

berserktron3k, 2016-08-10:
In your code, it looks like you chose Type B as the switch type. How'd you come to select that?

kizniche, 2016-08-10:
Ok, now the original codesend is working, but as long as I _don't_ select the pin:
```
sudo ./codesend 29955 -l 189
```
This doesn't seem to work:
```
sudo ./codesend 29955 -p 2 -l 189
```

The Python code doesn't work, but I used the example code from https://github.com/lexruee/pi-switch-python

kizniche, 2016-08-10:
Got the Python code working!
```Python
#!/usr/bin/python
# coding=utf-8

from pi_switch import RCSwitchSender

sender = RCSwitchSender()
sender.setProtocol(1)
sender.setPulseLength(189)
sender.enableTransmit(0)
sender.sendDecimal(29955, 24) # switch on
```

berserktron3k, 2016-08-10:
Yup, that did it. Mine works with that as well.

kizniche, 2016-08-10:
Success! The Python code was what I wanted all along. I should be able to easily integrate this now that I have receiving and transmitting working in Python. I also have two more sets coming in the mail :)

kizniche, 2016-08-10:
Updated receiver code that includes the delay length:
```Python
#!/usr/bin/python
# coding=utf-8

from pi_switch import RCSwitchReceiver

receiver = RCSwitchReceiver()
receiver.enableReceive(2)

num = 0

while True:
    if receiver.available():
        received_value = receiver.getReceivedValue()
        if received_value:
            num += 1
            print("Received[{}]: {}".format(num, received_value))
            print("Bit Length: {} bit".format(receiver.getReceivedBitlength()))
            print("Delay: {}".format(receiver.getReceivedDelay()))
            print("Protocol: {}\n".format(receiver.getReceivedProtocol()))
        receiver.resetAvailable()
```

kizniche, 2016-08-10:
One potential issue I foresee with this will come from the inability to query the state of the relay (as can be done with a GPIO). There should probably be an option to repeat the command x number of times.

berserktron3k, 2016-08-10:
I agree. That occurred to me last night as well. Again, it's a poor man's
automation, for sure. I imagined using conditionals to simply fire off the
command. The problem is you can't validate whether it's functioning
properly or not. It'd be firing blind. In my scenario, it shouldn't be that
big of a deal, but for others it certainly wouldn't be ideal.

Brady Hummel

mycol, 2017-05-02:
Hi Kizniche,
I'm also an admirer of mycodo, it's beautiful, powerful, and still easy enough for a novice's novice like me to setup. I was able to get the DHT22 and 433Mhz Transmitter/Receivers connected and tested on my pi. Coincidentally I had purchased the exact same material as you depict and I was able to get the codes and I've tested 'codesend'. I don't know much about python so I got a little confused on how you resolved controlling the RF outlets. I'd like to incorporate in mycodo if possible. What do you suggest as the optimal method to control humidity like this... if humidity <88 turn on RF outlet, if >93 turn off RF outlet (controlling cool mist humidifier).

berserktron3k, 2017-05-02:
mycol, I agree with your last statement. It would be best if it could be used as a relay in Mycodo. I came up with a workaround by giving conditionals to the sensors, and having Mycodo execute a command line when certain conditions are met. Been working pretty reliably since last August.

kizniche, 2017-05-02:
I've had these transceivers for a while now but haven't gotten around to integrating it into Mycodo. I still think it would be a nice addition. I have half this week off, so maybe I'll sit down one day and get a nice test environment going to allow this to be used with relays.

mycol, 2017-05-02:
berserktron3k - I noticed the 'command' option and was going to tinker with it, great suggestion!

kizniche - That would be huge, I know a lot of people are opting for RF over the power relays so they can avoid working with mains power. It could be a very helpful option. :)

Solid program! Thanks for releasing to the world.

closed 2017-10-24 by kizniche