kizniche/Mycodo #89: Camera Capture Crashes Apache

kizniche, 2016-07-17:
Using the Python module picamera, a simple image capture is executed from the Camera page, and causes an error (500) with a cryptic and unhelpful error. I've isolated the issue to the camera code in daemonutils.py's camera_record() function, which is essentially:

```
now = time.time()
timestamp = datetime.datetime.fromtimestamp(now).strftime('%Y-%m-%d_%H-%M-%S')
path_stills = '{}/camera-stills'.format(INSTALL_DIRECTORY)
still_filename = 'Still-{}.jpg'.format(timestamp)
output_filepath = '{}/{}'.format(path_stills, still_filename)
assure_path_exists(path_stills)
with picamera.PiCamera() as camera:
    camera.resolution = (1296, 972)
    camera.hflip = True
    camera.vflip = True
    camera.start_preview()
    time.sleep(2)  # Camera warm-up time
    camera.capture(output_filepath, use_video_port=True)
```

Apache Log:
```
[Sun Jul 17 13:33:35.536307 2016] [wsgi:error] [pid 19033] [client 192.168.0.4:55927] Truncated or oversized response headers received from daemon process 'mycodo_flask': /var/www/mycodo/mycodo_flask.wsgi, referer: https://192.168.0.7/camera
[Sun Jul 17 13:33:36.244687 2016] [wsgi:info] [pid 21228] mod_wsgi (pid=21228): Attach interpreter ''.
```

Interestingly the error doesn't occur when it's running as a Flask app (and the image is captured and saved), but it does when running in production mode.

To run as an app in debug mode:
```sudo ~/Mycodo/mycodo/mycodo_flask.py -d```

To run in production mode:
```sudo /etc/init.d/apache2 start```

berserktron3k, 2016-07-17:
I've confirmed on mine that debug mode works properly as well. My setup is: RPi 3 B, Raspian Jessie Lite (5-27-2016) fully updated, and a RPi NoIR Camera V2. Mycodo 4.0.5 (sunning your setup script).

berserktron3k, 2016-08-01:
Hey, I know you're busy working on a bunch of stuff, but I wanted to know a few things. I'm trying to research the problem, and in looking through the code, this issue may be bigger in scope than I thought (or, my Pi is having issues).

When a user clicks on Tools -> Camera, what should the user see? Also, what should a user see when he clicks Admin -> Configuration -> Camera?

I see a button to take a still, and what should be the previous still taken (when taken in debug mode). I was reviewing the code, to see if I could identify the issue, and it raised a few questions.

In the code, there's a section for time lapse. I can see that that is under development. Then I see a live stream, however nothing in my installation shows an option for a live stream (in the gui). Also, the previous JPG taken does not display, and I believe that's an issue in the code. I'm pretty sure the code to display the image is trying to pull it from a different location than where it is being saved.

I know you're busy with the PID's, etc., so not looking to rush you. Just wanting to know what features should be expected by end users, that way I can research a little better.

kizniche, 2016-08-01:
You're correct the timelapse is disabled, because I've yet to test it. And I've disabled the streaming because of the issue I've had with it. If you want to enable the stream buttons to test, you have to uncomment the  on [Lines 18-29 of camera.html](https://github.com/kizniche/Mycodo/blob/master/mycodo/frontend/templates/pages/camera.html#L18-L29)

kizniche, 2016-08-01:
The still image capture should be working (it is for me). It may be a permissions issue. What are the permissions of your ~/Mycodo/camera-stills/ directory? The following command should correct any permissions issues:

```
sudo ~/Mycodo/mycodo/scripts/update_mycodo.sh initialize
```

FYI, this initialization is executed during the install script and after every upgrade from the web interface, so it should have already been executed at least once to set up the proper permissions.

berserktron3k, 2016-08-01:
Here are the permissions:
drwxrwxr-x 2 mycodo mycodo   4096 Jul 27 19:38 .
drwxrwxr-x 8 mycodo mycodo   4096 Jul 30 16:07 ..
-rw-rw-r-- 1 mycodo mycodo 597920 Jul 27 19:38 Still-2016-07-27_19-38-07.jpg
What should the permissions be?

I've ran the update from the web console multiple times (everytime you've pushed an update), and still see this issue. I figured it was related.

Thanks for clarifying what we should see. That will help me narrow my troubleshooting.

kizniche, 2016-08-01:
/camera/still is a Flask route that will direct to the photo. This is the only way authentication can occur, otherwise anyone can load that web address and view the photo. What happens when you try to go to that address? Is there an error?

kizniche, 2016-08-01:
Here's the pertinent [function to display the still image](https://github.com/kizniche/Mycodo/blob/master/mycodo/mycodo_flask.py#L858-L885)

berserktron3k, 2016-08-02:
When I replace /camera/still/{stillname...} with the camera-stills path, it doesn't work (as expected, your explanation makes absolute sense).

Can you clarify your comment earlier "The still image capture should be working (it is for me)."? Do you mean it's displaying your stills, or that it's capturing them properly? If it's capturing them properly, I wonder if it's unique to my rig, since it crashes when I try to capture, unless in debug mode (as you pointed out earlier)?

Side note, how was NOLA? My wife and I just got home last night from there.

kizniche, 2016-08-02:
My install is both capturing and displaying them properly.

When I said can you go to the address, go to ```http://yourIPaddress/camera/still/Still-2016-07-27 19-38-07.jpg``` and see if the image loads. If not, we can restart the web server in debug mode and get the full error.

You're getting a crash when not in debug mode?? That's news to me. I've only experienced that with streaming video, not capturing stills.

NOLA was great! Reminded me a lot of my home state of Florida. The conference was a success. I won an award for presenting my research (and $500!).

kizniche, 2016-08-02:
Oops, my mistake. Yes, my issue has been with capturing stills. I had forgot I was running the HTTP server in debug mode when I tested and it worked. I can confirm I can only capture stills while in debug mode. Sorry for the confusion. It seems I can only capture AND view images while running it in debug mode.

berserktron3k, 2016-08-02:
Phew... I'm not crazy then.

kizniche, 2016-08-09:
After spending some time on this, I believe the issue is the web server user ```mycodo``` is not a member of the group ```video```. My initial tests have fixed the error. I'll push an update soon if all my tests work.

kizniche, 2016-08-09:
You may need to reboot for the user addition to take effect. I thought just logging out and back in would work, but it didn't.

kizniche, 2016-08-09:
Well, I guess I closed this issue by fixing the bug _and_ getting timelapse to work!

closed 2016-08-09 by kizniche