kizniche/Mycodo #90: support for SHT75 , 71, 21, 25 sensors

wangine, 2016-07-17:
Hello kizniche , first of all you do a great work . I worked so many years under Michochip family and was everything fine , now i found "in mistake " your project and i by some days ago a Raspberry pi 3 board for that . My primary sensors are SHT75 &SHT25 , very acurrate used for my  tobaco fermenting chamber where 2% of RH change radicaly the result , so . After days trying to understand linux and servers i finaly install your project , but for some reasons SHT sersors cant be assigned. All sensors work on any GPIO pin , tested under terminal trouth "sht" driver // as root command
 pi@raspberrypi:~ $ sudo sht -t -r -d 19 26
25.63
55.824185
16.147750813
If i try to add SHT sensor in Mycodo server i get the return "Could not activate Sensor controller with ID r4H1K6Y0: SHT is not a valid device type." ,, so i try many ways to assign the PIN trouth sensor ,with GPIO.24 or 24 or BCM name pin 19 , and on Clock pin also .
I try to look in /var/www/mycodo/mycodo on controller_sensor.py and i see your note about SHT sersors, "line 31" and begin with line "168-192" i didn't find a sht device .I try to add "elif self.device_type == 'SHT':
self.measure_sensor = SHT(self.location)"
But i receive some result . I understand C code , i write many drivers over the time but in python is hard . For same sensor i write a driver for PIC device [https://www.ccsinfo.com/forum/viewtopic.php?t=54354](url)
Aso to be able to install Mycode today  i was forced to comment the #MCP342x in requirements.txt file , otherwise i get the same error on python 2.7 module . Two days ago the git page was with other explication , i install without any error but today i change the SD card and instructions for install was different , i assume you update the git page yesterday.
Thank you

kizniche, 2016-07-18:
My problem is I don't have the SHT sensor.

I actually just pushed an update that may fix your SHT issues. Update your system, and if the SHT sensor is still not working, if you could do some testing for me, I can try to get it working

First, edit the script at ~/Mycodo/mycodo/sensors/sht.py, line 59:

```
sht = SHT(16, 17, '3.5V')
```
Change 16 to be the GPIO pin, 17 to be the clock pin, and the voltage going into the sensor (as a string). Then run the SHT test script with:

```
python ~/Mycodo/mycodo/sensors/sht.py
```

Let me know of any errors.

wangine, 2016-07-18:
Yes , after update the database error was fixed , now can add any SHT sensor without any error when i activate the sensor . I dont need to edit the script , you already do that in update , i just check now and it is `sht = SHT(16, 17, '3.5V')` 
If i run `sudo python ~/Mycodo/mycodo/sensors/sht.py` i get a error in terminal 
`pi@raspberrypi:~ $ sudo python ~/Mycodo/mycodo/sensors/sht.py
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/sensors/sht.py", line 62, in <module>
    print("Temperature: {}".format(measure['temperature']))
TypeError: 'NoneType' object has no attribute '__getitem__'
..and if i run the command according with my pin assigment i get normal return
``pi@raspberrypi:~ $ sudo sht -t -r -d 19 26
23.4
56.6488708605
14.2983740425
`
Now my question , i dont understant the line `SHT(16, 17);` ,i need to put my sensor on 16 pin as DAT and on 17 as CLK  on board ? My sensor work on any GPIO pin , i've testet on all "curiosity" and work as well. Now it is connected as CLK "BCM 19 (GPIO 24)" and DAT "BCM 26 (GPIO 25)" . If i check my IO table (gpio readall) , i see 19 and 26 as OUT , but if i stop the daemon service and i run "sudo sht -t 19 26 " SCK (19) remain OUT and DAT (26) become IN.

kizniche, 2016-07-18:
Ok. Lets back up. Your title mentions the SHT75, SHT71, SHT25, and SHT21. Which of these do you have. The SHT python module in Mycodo only supports SHT1x and SHT7x. If you're using a SHT2x we'll need more testing with possibly a new module.

Now, for the testing. When I wrote:

> First, edit the script at ~/Mycodo/mycodo/sensors/sht.py, line 59:

> sht = SHT(16, 17, '3.5V')

> Change 16 to be the GPIO pin, 17 to be the clock pin, and the voltage going into the sensor (as a string). Then run the SHT test script with:

I am asking you to change "16" to whatever GPIO your SHT sensor data pin connected to, change "17" to whatever your SHT sensor clock pin is connected to, and change "V3.5" to whatever the voltage going into your SHT sensor is.

Than run the script and see if it returns measurements. You need to change these values because it cannot communicate with your sensor otherwise.

I just updated the script on github, so be sure to update Mycodo before making the edits and running it.

kizniche, 2016-07-18:
Alternatively, if you give me ssh access to your Raspberry Pi with your SHT sensor connected, I could probably find a solution quicker.

wangine, 2016-07-18:
Ok , my sensor are SHT75 , i just say (SHT 75, 71, 25, 21) is basicaly the same , In my PIC controller i use same driver for all those sensors. & yes i update Mycodo . The best way i think is to give ypu ssh acces , but you need to explain me how to do that . I already i have configured my RPi to have ROOT acces over SSH. I will change now the pin's in sht.py according with my sensor.

kizniche, 2016-07-18:
To enable remote ssh access, you will need to forward port 22 (SSH) in your router to your Raspberry Pi's IP address. Then, try to connect from a remote location to verify it works. There are a bunch of tutorials online.

After you make the edits to the pins in sht.py, run it with:

`sudo python ./sht.py`

wangine, 2016-07-18:
I think is a sintax problem somewhere , does't matter what pin i put in sht.py , the result are the same in ssh terminal 
_pi@raspberrypi:~ $ sudo python ~/Mycodo/mycodo/sensors/sht.py                   Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/sensors/sht.py", line 67, in <module>
    print("Temperature: {}".format(measurement['temperature']))
TypeError: 'NoneType' object has no attribute '__getitem__'
__

I will try not to give you acces to ssh , 1h im bussy , after i have all time to.
Now the error has move to line 67 in sht.py . I think you add some code to sht.py still terminal return error on same line 
_print("Temperature: {}".format(measurement['temperature']))_
If i comment that line , the error will move to next line , so i think on those lines is something wrong /?

kizniche, 2016-07-18:
That particular error occurs because it's not getting the actual readings. I won't be able to ssh in to look for another few hours anyway.

MountAthos, 2016-07-18:
Will try to check the denser functionality. In Istanbul , where I am located now had military coup attempt. I have to try with latest repository. Can't promise.

kizniche, 2016-07-18:
Thanks, @MountAthos.

@wangine, @MountAthos was the person who did all the testing to integrate the SHT sensors into the previous version of Mycodo. Since the update to the Mycodo framework (to the version 4.0), the sensor code had to be changed. I didn't have a SHT sensor to test the new code. That's about the state of where we're at now.

kizniche, 2016-07-18:
You can also try creating a script, such as test_sht.py with the following code:

```
from sht_sensor import Sht
sht = Sht(21, 17)
print 'Temperature', sht.read_t()
print 'Relative Humidity', sht.read_rh()
```

Change 21 to your data GPIO and 17 to your clock GPIO, save it, then run it with:

```sudo python ./test_sht.py```

kizniche, 2016-07-18:
Also, here's the SHT python module being used: https://github.com/mk-fg/sht-sensor

wangine, 2016-07-18:
Wow , now something became strange for me . Some test :
If i run in ssh terminal _pi@raspberrypi:~ $ sudo sht -v -trd 19 26_
i will get correct readings
_temperature: 24.56
rh: 51.516471808
dew_point: 13.9073395651_
If i made a _test.py_ with content "in same Mycodo sensor folder"
 _from sht_sensor import Sht
sht = Sht(26, 19)
print 'Temperature', sht.read_t()
print 'Relative Humidity', sht.read_rh()_
I will get that error like sensor don't exist on that port
**pi@raspberrypi:~ $ sudo python ~/Mycodo/mycodo/sensors/test.py
Temperature
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/sensors/test.py", line 3, in <module>
    print 'Temperature', sht.read_t()
  File "/usr/local/lib/python2.7/dist-packages/sht_sensor/sensor.py", line 214, in read_t
    t_raw = self._get_meas_result(self.cmd.t)
  File "/usr/local/lib/python2.7/dist-packages/sht_sensor/sensor.py", line 154, in _get_meas_result
    self._send(cmd)
  File "/usr/local/lib/python2.7/dist-packages/sht_sensor/sensor.py", line 117, in _send
    raise ShtCommFailure('Command ACK failed on step-2')
sht_sensor.sensor.ShtCommFailure: Command ACK failed on step-2**
So , now i think the problem is in python module

wangine, 2016-07-18:
Aso i just look in your modified sht.py code , the new line added ,it's write  error or just to dont make variable conflict ?
**class SHT(object):**
    _def __init__(self, pin, clock_pin, voltage):
        self._temperature = 0
        self._humidity = 0
        self.pin = pin
        **self.click_pin** = clock_pin
        self.voltage = voltage
        self.running = True_
Its a write mistake or should be 
**self.clock_pin = clock_pin** ????
I dont know your links in code , otherwise i'm fare away by python , i just try to understand .
Also i found a NOTE about sht-sensor who redirect on that page  
https://pypi.python.org/pypi/virtualenv

kizniche, 2016-07-18:
Well, your sensor responds from the terminal application, so we know your sensor is good. It's strange the python code would be giving errors.

wangine, 2016-07-18:
yes , i just made a account here for ssh acces 
https://www.dataplicity.com
My email account "wangine.tam@gmail.com"
passwd is              "inteligenta"

Myke974, 2016-07-18:
@wangine : we all see thoses informations ! :D
Danger ! Danger ! !

wangine, 2016-07-18:
@Myke974  // so.... is nothing to see, just a simple raspberry with a fresh intall of Raspbian Jessie Lite,
Thanks anyway

kizniche, 2016-07-18:
@wangine, I edited your post to remove your login info. I saved it for later. I should be able to look at it in a few hours.

wangine, 2016-07-18:
@kizniche  OK , i just update my login info, i send you a private email via your website. Thank you

kizniche, 2016-07-18:
I think I figured out the error. About to push an update.

wangine, 2016-07-18:
yes i see the modification , i update Mycodo , but is unable to start daemon server now, 
I type in terminal 
**sudo ~/Mycodo/mycodo/mycodo_daemon.py**
and i get 
**pi@raspberrypi:~ $ sudo ~/Mycodo/mycodo/mycodo_daemon.py
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/mycodo_daemon.py", line 38, in <module>
    from controller_sensor import SensorController
  File "/home/pi/Mycodo/mycodo/controller_sensor.py", line 33, in <module>
    from sensors.sht import SHT
ImportError: cannot import name SHT**

kizniche, 2016-07-18:
I haven't pushed it. I'm still testing (Your Pi just disconnected me). I said I was *about* to push it ;)

kizniche, 2016-07-18:
I did find the problem. I was just working on how to get it all working.

wangine, 2016-07-18:
ok ,, i'm sorry , i just reboot my RPi , was my fault . Ok i let you to test , i dont touch my RPi

kizniche, 2016-07-18:
No problem. I think I figured it out. After your update, check that the proper voltage is set for the sensor, then activate it.

wangine, 2016-07-18:
yes its working , thank you , great job and very fast aslo . I just will made some move on RPi pinout later , just for you to be sure everything are fine with that sensor . Also i will try tomorow the rest of sensors SHT25 .
If SHT25 will work its everything perfect , because SHT71&75 are identical, just less accuracy (SHT71) , 
Can say me the basic dependencies of "sht.py" ? I will try to add the last measurement between temp & RH , the condensing point (dew_point)

kizniche, 2016-07-18:
Great! I just added the dew point to the SHT sensor measurements and pushed to github. Give me a few more minutes to add the measurement to the rest of the system.

wangine, 2016-07-18:
Yes , i see , does't show in live view , anyway  i think you still work on .
In python dont need to init variables like in C ? Dont should be >>
_**class SHT_read(object):**
    def __init__(self, pin, clock_pin, voltage):
        self._temperature = 0
        self._humidity = 0
		**self._dew_point = 0**
        self.pin = pin
        self.clock_pin = clock_pin
        self.voltage = "{}V".format(voltage)
        self.running = True_
I just ask

kizniche, 2016-07-18:
Since the dew point is not an actual measurement from the sensor (it's calculated from the temperature and humidity), I'm not including it in the same way as a measurement. Instead, I'm going to calculate it when it's needed, using the temperature and humidity variables. The code above was temporary. I have a big git commit I'm getting ready to push that will put Dew Point on all web pages. I'm just making sure everything works...

kizniche, 2016-07-18:
Well, what I said was partly true. For only the SHT sensor will I treat dewpoint as a measurement (storing it in the class), only because it has it's own dew point calculation that it uses. For all the other temperature/humidity sensors, I added a dew point calculating function (copied from http://www.meteo-blog.net/2012-05/dewpoint-calculation-script-in-python/).

Dew point should now be stored in the database when using any humidity/temperature sensor, and it should now appear on the Live, Graph, Sensor, and PID pages.

kizniche, 2016-07-18:
Thanks for the help. It's so much easier when I have access to a Pi with the sensor connected than to create a bunch of test scripts for someone to run themselves. It can save a lot of time!

wangine, 2016-07-18:
No no ,me i need to say thanks for help . Because i buy the RPi , for your project , it's still the best project on web, BrwPi is the same but i dont like arduino , because my code depends by others code. Many years my powerful dsPIC's was do a great job for all my automatizations, but now i say to move on . 
I did't realy like python , to many issues ,, i love C and i will learn in the feature to coding in C the RPi board Thanks again , Tomorrow if you want i will connect the SHT25 sensor

kizniche, 2016-07-18:
I also like C. I'm not really sure how it came to be, but Mycodo has become my Python learning project. I knew virtually no Python about a year ago. I'm actually quite impressed with how capable it's proven to be. Even the web server is written in Flask, a Python micro framework.

Yes, please try the SHT25 and let me know if it works. If not, we can go from there ;)

wangine, 2016-07-18:
Unfortunately after last update on 4.0.6 ,adding the dew_point calculation , sensor dont work again ,

kizniche, 2016-07-18:
The latest update installs numpy. It may be updating. It lasted an incredibly long time on my systems. Do this before confirming it's not working. In Mycodo, go to Tools->View Logs, then click Update Log. If it says:
```
    Running setup.py install for numpy: still running...
```
Then the update process hasn't yet finished.

kizniche, 2016-07-18:
In fact, my Raspberry Pi 1 is still updating numpy, for the past 15 minutes! Perhaps I should find a way to calculate dewpoint without numpy.

wangine, 2016-07-19:
Yes , has take 5-6 minutes to update numpy , the process was finished long time ago . I still try to understand your code , still i dont understand here ;
        _response = {
            'humidity': self.humidity,
            'temperature': self.temperature,
            'dewpoint': self.**dew_point**   #linie jos
        }
        return response_
I find just **dewpoint** not **dew_point**

kizniche, 2016-07-19:
Sorry about that. It was a typo!

wangine, 2016-07-19:
Maybe help , here is my working C function to calculate dewpoint on dsPIC33FJxx read from same SHT75 sensor. 
**//--------------------------------------------------------------------
// Based on the input of humidity and temperature to calculate dew point
//--------------------------------------------------------------------
float calc_dew_point(float h,float t)
{ 
    float logex, dew_point;
    logex = 0.66077 + 7.5 * t / (237.3 + t) + (log10(h) - 2);
    dew_point = (logex - 0.66077) * 237.3 / (0.66077 + 7.5 - logex);
    return dew_point;
}**
Next week i willl be free to compile C code under RPi , i still learning

kizniche, 2016-07-19:
Numpy is used for the [log function](https://github.com/kizniche/Mycodo/blob/master/mycodo/sensors/sensorutils.py#L20).

It's actually already installed, it's just a long update process. I'll keep using it. I'd rather it be updated than a old version used. It will be useful when I want to add features that calculate statistics and other mathematics.

wangine, 2016-07-19:
I confused , the **sht** function have his own dewpoint calculation code . I am right ? because ;
**pi@raspberrypi:~ $ sudo sht -v -trd 19 26
temperature: 24.44
rh: 55.275449248
dew_point: 14.8864743383**
Thats mean dont need extra calculation . I'm just confused

kizniche, 2016-07-19:
You are correct. The SHT module does, but not the DHT11, DHT22, and AM2315.

kizniche, 2016-07-19:
I'm looking at the SHT dew point calculation now and trying to figure it out:
```
tn = dict(water=243.12, ice=272.62) # Table 9
m = dict(water=17.62, ice=22.46) # Table 9
```
...
```
	def read_dew_point(self, t=None, rh=None):
		'With t and rh provided, does not access the hardware.'
		if t is None: t, rh = self.read_t(), None
		if rh is None: rh = self.read_rh(t)
		t_range = 'water' if t >= 0 else 'ice'
		tn, m = self.c.tn[t_range], self.c.m[t_range]
		return ( # ch 4.4
			tn * (math.log(rh / 100.0) + (m * t) / (tn + t))
			/ (m - math.log(rh / 100.0) - m * t / (tn + t)) )
```

kizniche, 2016-07-19:
I could rely on the SHT sensor module to calculate the dew point, but I would like it to be a function inside Mycodo, in case something happens to the SHT module source code, dew point will still be able to be calculated for the other temperature/humidity sensors.

wangine, 2016-07-19:
You are double right 
First you need log function for that and second, is good to have your own code , 
That why i dont like arduino, depends to much on others code and work. Is more easy to write "import X function" but very unsafe if the source dissapear. Still the sensor dont work now , but daemon is started now , I was forced to update manualy 
**sudo pip install -r ~/Mycodo/requirements.txt --upgrade --force-reinstall --no-deps**
, via web i was feeling the update has stuck

kizniche, 2016-07-19:
So the SHT sensor is now not working, after the last updates?

wangine, 2016-07-19:
Yes , i just need to desactivate and activate again the sensor . dew point work also . Still i dont understand why Mycodo work only on **lite** RPi version ? Because some days ago i did't read very carefuly the git page and i try to install several times on full image of RPi , and after some modification daemon was started and Mycodo has work also .

kizniche, 2016-07-19:
I think it has to do with the version of pip. Raspbian Jessie Lite doesn't come with pip, so it is installed (version 1.8) with east_install during the Mycodo install process. Raspbian Jessie Full already comes with pip, but it's an old version (1.5). I think smbus has an issue being installed with the lower version, and that's what stops Mycodo from functioning.

This is my theory, as I've recently seen a log of the a Jessie Full install going through the Mycodo install process and it not being able to properly upgrade pip to 1.8.

This is why I recently added a [line to uninstall pip](https://github.com/kizniche/Mycodo/blob/master/setup.sh#L40) before trying to install it with easy_install. I'm not sure if this works, but I'll get around to trying installing Mycodo over a Raspbian Jessie Full some day.

kizniche, 2016-07-19:
Found the error in the sensor code. Pushing an update in a min.

wangine, 2016-07-19:
I succesfuly instaled two times 3 days ago Mycode on Jessie Full , i just folow your instructions on the old git readme page . I keep it save as complete web page in my PC (just in case :d), and because i did't pay attention on your note about Jessie Lite, i just try and try , without know actualy was SHT sensor isue , not the Jessie version . Tomorow i will made some test on other SD card with Jessie Full and also i will add SHT25

kizniche, 2016-07-19:
I wonder what differs between the install procedures.

Can you confirm the SHT sensor now works?

wangine, 2016-07-19:
wait , it still update ,

wangine, 2016-07-19:
yes ,sensor work , I write now a new fresh full jessie on other SD card . I have keep my instruction to install Mycodo on Full Jessie . Tomorow i will made some test , Now realy i need to sleep , after 4h i go to work .
Good night and thanks for the great job.

kizniche, 2016-07-19:
G`night

closed 2016-07-19 by kizniche

wangine, 2016-07-19:
Anyway fell free to make any test you want ower ssh on my RPi , is on all day long . The login and account i will keep for you .