kizniche/Mycodo #91: support for SHT21, 25 sensors

wangine, 2016-07-19:
Hello . we can start here new one. I research a little bit the web , and i also try , the sht library will dont work on SHT25, 21 sensors , the reason is not timing or other hardware issue , is just the CRC check of the sht library and of course will dont pass . So i found a good and simple code , without CRC , but work only on I2C port , i connect on GPIO 2 and 3. I dont think thats a problem . I just download the library , i put in sensor folder the SHT25.py and i run it . Succesfuly return all values on , SHT25 and SHT21 also. Still on web i find different driver for SHT21 , hes more complicated and many includes. The sensors are identicaly , just less accuracy on sht21 (humidity)
Require **smbus** , but is not necesarry , is included in Mycode already.
The library is here , has been tested and fell free to test on my RPi over ssh . 
https://github.com/ControlEverythingCommunity/SHT25
I will be back some h later

kizniche, 2016-07-19:
That's great! I just integrated the SHT25 module into Mycodo with the latest update. Give it a try! I2C is nice because you can connect multiple SHT2x sensors on the same 2 wires.

wangine, 2016-07-19:
Yes i know about I2C in you ,code, You dont have ssh acces right now , i change my SD card with a Full Jessie . I need to make acces for. I have just some days over linux & is not bad , i think i will like it.

kizniche, 2016-07-19:
Linux is a great to work in. I've mainly used my Raspberry Pis headless until I got an LCD for my RPi 3, for the latest build I'm working on (https://hackaday.io/project/11997-mycodo-environmental-regulation-system).

Let me know when you test the SHT2x sensor if it works with the latest update, and I'll close this issue. Thanks.

wangine, 2016-07-19:
oo, just wait , i will test right now

wangine, 2016-07-19:
I get the error when i try to save 
**Sensor settings were not able to be modified: could not convert string to float:**
and when i try to activate 
On top on a green line is ; **Sensor controller activated in SQL database.**
and below on orange green ; **Could not activate Sensor controller with ID VQmKHDUF: global name 'GPIO' is not defined**

wangine, 2016-07-19:
You have acces on ssh on second device . https://www.dataplicity.com/devices/
Is the same login
You can test if you want , is 32G SD device

wangine, 2016-07-19:
Yes , now can save also can activate , but sensor did't respond , So i test SHT25.py in terminal , hes work
**/var/www/mycodo/mycodo/sensors$ python SHT25.py
Relative Humidity is : 51.04 %
Temperature in Celsius is : 24.97 C
Temperature in Fahrenheit is : 76.94 F**

kizniche, 2016-07-19:
The module I created is Mycodo/mycodo/sensors/sht2x.py... not SHT25.py... I think that's a script you created. It's not a part of Mycodo.

I figured out the errors by testing on your system. Pushing a fix to git shortly.

kizniche, 2016-07-19:
Oh, I see what you were saying, that the sensor was working. I found some typos and another error in the module itself. Now you can execute ```python ./sht2x.py``` and it works.

wangine, 2016-07-19:
For me dont work , i use WinSCP as root login over ssh . When i type **python sht2x.py**  in sensors dirrectory i get kicked out from sesion 
But if i type python SHT25.py it works /
I do some mistake ?

kizniche, 2016-07-19:
They are two different files. It works for me. I'm updating your system now and will test adding the sensor in a sec.

wangine, 2016-07-19:
Ok , you notice is Jessie Full on it .

kizniche, 2016-07-19:
Heh... I can't really tell just being in the terminal. But if so, that's good. I removed the line in the README instructing only to use the Lite version and added your purge command to the setup.sh.

kizniche, 2016-07-19:
Looks like it works!

wangine, 2016-07-19:
yes it works . Nice job, and very fast also . So i did't intend to use sht25 , i prefer sht75, my tobacco is very sensitive on fermenting stage  :d some wrong humid procents in couple hours will destroy my tobacco . I do it for your support like i promise .

kizniche, 2016-07-19:
Thanks. I appreciate it! Do you have a website that details how you prepare your tobacco? It sounds really interesting (I'm a microbiologist)!

kizniche, 2016-07-19:
Also, dataplicity has been really handy. It even allows me to access the web UI. I'll have to keep that handy for future Mycodo diagnostic work.

wangine, 2016-07-19:
No problem about dataplicity , i told you , i keep the account untouched for you. Now about tobacco ,,,, is hard and interesting , like your mushroom also . I;m a electronic engineer , so tobacco and code is just (ammore of art) . I have 2 domains , not a host for my website , is expensive in my country , but i will try in the feature to make a server in my router , my connection is over 1Gb some times. So to made a simple ciggar from a green leaf of tobacco , will take 2 years of work that why all automatisations, I can teach you everything about tobaco , you will teach me about mushroom .

kizniche, 2016-07-19:
I was meaning more like I'll keep it handy in case anyone needs assistance with a Mycodo issue and isn't savvy enough to port forward for me to connect (or just doesn't want to expose their Pi to the entire internet).

Hah! Alright, that's a fair trade... I've always wanted to document and publish cultivating a mushroom flush, from start to finish.

closed 2016-07-20 by kizniche