kizniche/Mycodo #92: Install on Jessie Full

wangine, 2016-07-19:
Is not a issue but i dont know where to write 
The single problem of Jessie Full is the **PIP** module , **pip 1.5.6** , that's to old , and new update will dont remove and will cause problems. 

1; check after a fresh install the pip version 
**:~ $ sudo pip --version**
will show like this 
_pip 1.5.6 from /usr/lib/python2.7/dist-packages (python 2.7)_

2; instal a new version of pip
**:~ $ sudo easy_install pip**

3; check again that 
**:~ $ sudo pip --version**
shold be 
_pip 8.1.2 from /usr/local/lib/python2.7/dist-packages/pip-8.1.2-py2.7.egg (python 2.7)_

4; install updates "will take some time"
**sudo apt-get update -y && sudo apt-get upgrade -y && sudo apt-get autoremove -y**

5; Follow the Kyle instructions from here 
**sudo raspi-config**
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~/
Everything will work like a charm 
The ideea comes last night from @kizniche

kizniche, 2016-07-19:
Well, I already have in the setup script now

```
sudo apt-get remove -y python-pip
sudo apt-get install -y easy_install
sudo easy_install pip
```

I haven't tested this series of commands, and they differ slightly from what you have above. Did you uninstall pip between your steps 1 and 2? I found that the original install script that had `sudo easy_install pip` did not upgrade it.

wangine, 2016-07-19:
I'm sorry , i just fogget one step on beginning ,
That command make good job without error in Mycodo
**sudo apt-get purge python-pip**
I already try with 
**sudo apt-get remove -y python-pip**
Will succed but after try i get error mesage 
**pi@raspberrypi:~ $ sudo pip --version
Traceback (most recent call last):
  File "/usr/local/bin/pip", line 5, in <module>
    from pkg_resources import load_entry_point
ImportError: No module named pkg_resources**

pookNast, 2016-09-08:
Encountered issues running the setup script (09/08/16 US) under the full version of Jessie. Attempted remedies suggested to no avail. Reverted to the lite version and had 100% success. FYI.

kizniche, 2016-09-08:
What was the error you encountered?

kizniche, 2016-09-10:
@pookNast, Can provide any information about what the error was?

pookNast, 2016-09-10:
Hi,
I thought I had sent you the screen shot via email. whoops. Here ya go pal.

[Screenshot](https://s16.postimg.org/4ase1g12t/20160908_120111.jpg)

I got that error on a fresh install on an rPi3, 32 GB sd Card using Jessie Full. Something to do with the ntp I believe? I reflashed the SD card with Jessie Lite and everything ran flawlessly afterwards. 

I am currently up and running and waiting for parts to come in, to put the actual controller together. I'll keep you updated.

Cheers.

kizniche, 2016-09-10:
Hmm. It looks like there was some sort of dpkg error with the ntp package. I think this may have been due to a failed apt upgrade or a bad Raspbian image write.

kizniche, 2016-09-10:
To expand on the last comment, the first thing I see in your terminal is a segmentation fault. That's indicative of a larger issue with the package management system (apt). I can't tell where this is occurring in the setup script, but I'm assuming it's at the ```apt-get upgrade``` command. In any case, that segmentation fault is the cause and is not normal.

pookNast, 2016-09-10:
Hmmm. I would put a low probability on a bad image write. The sd card was brand new specifically for this project, and I ran it on a second card before diverting to an image re write to an image of Jessie lite. I ran apt-get update multiple times, and followed your instructions meticulously, every reboot and all. If there is anything else I can do, just let me know.

kizniche, 2016-09-10:
I have a new Pi 3 that I need to put mycodo on. I'll try to get to it this weekend and let you know how it goes.

kizniche, 2016-09-23:
I wasn't able to replicate the error. I installed from Jessie full to a RPi 3 with no issues. From the look of your error, you had an issue with a package, which was most likely the result of 1) a corrupted install package that was downloaded, 2) a corrupt Raspbian image that was installed (either during the download or writing process).

If you can replicate the issue and send the entire setup log (~/Mycodo/setup.log), I'll reopen this issue.

closed 2016-09-23 by kizniche