kizniche/Mycodo #93: Can't rename sensor or add GPIO

berserktron3k, 2016-07-19:
When trying to add RPi CPU Temp or Load sensors and rename them, or add and rename/assign GPIO to DHT22, I get the following in the web interface:

Sensor settings were not able to be modified: could not convert string to float:

kizniche, 2016-07-19:
Ah. Update your system and it should work again. I pushed a fix for that error less than an hour ago. ;)

berserktron3k, 2016-07-19:
Weird. I just did a clean install (about the time you must have made the
change). Thought it was related to the work you were doing on the other
issue, so also just ran an update from within the web console... same
issue. Forced a reboot too, and issue resides.

Let me try re-cloning from GIT.

Brady Hummel

kizniche, 2016-07-19:
Oh, you're right. I didn't make a full fix for that bug. Give me a minute.

berserktron3k, 2016-07-19:
Np. Take your time.

Brady Hummel

kizniche, 2016-07-19:
I just pushed a fix. Let me know if the bug is still there after you upgrade.

commit 8f22da1b (2016-07-19): Fix #93: Inability to save sensor settings.
closed 2016-07-19 by kizniche