kizniche/Mycodo #94: Add a graphic to Mycodo

wangine, 2016-07-19:
Just a least thing . It is possible to made a graphic , dont matter how in Mycodo GUI , in exel , or just simply in txt , with some instructions to Mycodo follow ? Just a example : 
**01.01.2016 -> 04.01.2016 
02.01.2016 -> 05.30pm -> 04.00pm ->
setpoint 33*C  -> 41*C , exp // increase exp to reach 41*C
03.-1.2016 -> 02.10pm -> 06.00am ->
 setpoint _last_st -> 26*C , lin // increase linear to reach 26*C**

Can be possible something like that or you can make with rules already in Mycodo ?

kizniche, 2016-07-19:
Yes! This is half possible. On the PID settings, you can create a changing setpoint, however it can only be a daily program (it will restart the next day). I want to add the ability to also specify a day of the year (and maybe year), which will allow for what you want to do.

I don't think it will be that much more difficult to implement the day or year (famous last words...).

wangine, 2016-07-19:
The title of issue is a little wrong . Should be >> **Import a graphic in Mycodo** 
Anyway now i notice , you add the current consuming state on each relay and can make calculation on the end process? thats impressive

kizniche, 2016-07-19:
Yeah, that's really handy to see how much you're spending on energy costs. Adding a current clamp could make that more accurate. I've been wanting to add a current sensor (inductive-clamp around wire) to Mycodo to be able to be able to track my home energy usage. I should have added one to the last set of sensors I just bought.

wangine, 2016-07-19:
Personaly i dont realy like the PID , it,s verry slow (exclude large rooms) , and soon the room size is changed the parameters are unusefull need to calibrate again . Here it's a piece of my code to choose right parameters for PID. is in C , you can easy ported on any PID controller. 
https://www.ccsinfo.com/forum/viewtopic.php?t=54563&highlight=

Also i we used from years a complementary filter , thats very easy , simple and powertfull filter can controll all applications , even with less than 10us response , that running on ordinary 80MHz processor, I did't test on 1GHz + . 
Hes response is realy fast .

https://www.ccsinfo.com/forum/viewtopic.php?t=46829&highlight=

kizniche, 2016-07-19:
I think I figured out a way to do this. It will be a little different than how I made the current iteration of the changing PID setpoint. For one, it will include an actual graph to visualize the change over the whole period of time that's set. Building it this way will also take off a major thing I wanted to add, which was time-rate method development, for making things like reflow ovens and thermal cyclers.

kizniche, 2016-07-19:
Nice alternatives to PID. So what you're suggesting is I rename the entire "PID" section to "Regulation" and make PID an option, in addition to your controllers ported to Python? ;)

wangine, 2016-07-20:
No , that PID autotune should add on each cycle of your PID > Thats it , no any modifications .Can make a simple line who call that function cycle by cycle . I dont think its a problem for a Raspberry processor.  I explain in that forum how to use . That simple function work for me years, should look on that video on end of the forum post. 
http://www.youtube.com/watch?v=Y6e-i7dk23w 
I dont know how to do, it's you choice . I did't understand how you compile so fast and also update Mycodo. 
I need some months to understand completly python and linux.
Just ignore first function from here 
https://www.ccsinfo.com/forum/viewtopic.php?t=54563&highlight= 
 was a mess , work some days ok , and others give a incredible overshots, I did't understand why.

kizniche, 2016-07-20:
One thing to keep in mind there's a pretty difference between microcontrollers like PIC/ATMega and the Raspberry Pi. If you need to do extremely time-sensitive stuff, you will probably want to go with the MCU. Because you're working in Linux and with a device without even a real-time clock, there will often be latency issues when trying to measure minute time frames. But for the majority of applications, the RPi will be sufficient. If not, then you could easily interface a MCU to regulate a specific process or maybe even interface with Mycodo in a future version.

kizniche, 2016-07-20:
I got it was a PID tuner, but I've also wanted to implement the ability to use different PID methods. There are so many PID algorithms, that I thought it might be useful to expand the one I started Mycodo with.

wangine, 2016-07-20:
Yes that was made here, http://www.brewpi.com/ , but i dont realy like the ideea .
Your project even was made alone , by you self , its much better .
They force you to connect with arduino , and again i dont like those boards , its all about copy and paste , 
Now , if you realy like to add a powerful controller should look again here 
https://www.ccsinfo.com/forum/viewtopic.php?t=54542
It's adaptive , predictive , autotune , everything you want but the math is complex there.

wangine, 2016-07-20:
Can see here the filter in action , of course my son also 
http://www.220.ro/documentare/Self-Balancing-Robot/umHTJRCLns/
And i understand you point about timming , but for humid , temp timing dont matter to much

kizniche, 2016-07-20:
[Untested](https://github.com/kizniche/Mycodo/blob/master/mycodo/controller_pid.py#L442-L474), but I added it to the PID controller ;)

wangine, 2016-07-20:
Realy ? I still dont get it . I told you are fast

wangine, 2016-07-20:
OO i see , now i get it how github works. 
**# enouth(temp, -0.8, 0.8)  # What is this function?**
If the terms will become to high should be limited , That function limit on what value you decide to be max. The limit function is this : 
**#define enough(_abs, _min, _max) (_abs = ((_abs > _max) ? _max : ((_abs < _min) ? _min : _abs)))** 
and is similar with this function but more on point and faster.
**void SetOutputLimits(float Min, float Max)
{
   if(Min > Max) return;
   outMin = Min;
   outMax = Max;
 
   if(iesire_t > outMax) iesire_t = outMax;
   else if(iesire_t < outMin) iesire_t = outMin;
 
   if(ITerm > outMax) ITerm= outMax;
   else if(ITerm < outMin) ITerm= outMin;
}**
 The two functions are similar ,

wangine, 2016-07-20:
I can send you my entire code to understand better . It take some times to understand others codes ,

[auto_PID.c.txt](https://github.com/kizniche/Mycodo/files/372696/auto_PID.c.txt)

kizniche, 2016-07-20:
Ah. That's really helpful.

wangine, 2016-07-20:
Just delete the txt extension , i cant upload as C , not alowed , can understand beter in C , colors are awesome

kizniche, 2016-07-20:
[I ported your code into two Python classes](https://github.com/kizniche/Mycodo/blob/abc8b723c659b088cbeb0893b4391f425b5147f0/mycodo/controller_pid.py).

I haven't tested it, but thought it would be a good exercise to see if I could port it. It should work. I still need to go back and try to understand it all. But for now, it's there. I'm sure we can find a way to make it work with Mycodo if it can improve regulation.

kizniche, 2016-07-20:
Also, I tried playing the video of the balancing robot, but it wouldn't play in mt browsers.

wangine, 2016-07-20:
yea , i see autotune implement , looks good , you modify the "restrict" function , So i assume **self** is like **#define**  ,, i wonder if is correct in that form in python ; _def enough(_abs, _min, _max) (_abs = ((_abs > _max) ? _max : ((_abs < _min) ? _min : _abs)))_
I know , any language have their specific typo, but i suppose the logical signs are identical, then that  function should work , im not sure if is **self** or **def**

kizniche, 2016-07-20:
It should work. I didn't want to use Python's ternary conditional operator like you did in C because Python's is confusing and not as intuitive as C's. I'll make some test scripts later when I get home, that use the classes, and see if they work.

wangine, 2016-07-20:
ok, i just find Geany and looks nice . I install BCM library and first program after i will turn on a LED , like test , is to made a PWM function bassed on internal timer , if i can of course.

kizniche, 2016-07-20:
Awesome robot! How many times does it run the PID function per second?

kizniche, 2016-07-20:
I created a test script that uses the simplest of the three classes. It simulates a PID running for 20 cycles with the error decreasing 40% each time. Take a look and let me know if that's the expected behavior.

```
~/Mycodo/mycodo/pid_filters/autotune_example_01.py
```

wangine, 2016-07-20:
Thank you 
Does't have PID , i told you,  i hate PID, is good only for slow applications like temp , on last video is running on **differential evolution algorithm** , was a lot of work ,36000 lines of code 
https://www.researchgate.net/profile/Ponnuthurai_Suganthan/publication/224330344_Differential_Evolution_Algorithm_With_Strategy_Adaptation_for_Global_Numerical_Optimization/links/0c960525d39935a20c000000.pdf 
The strategy was on last 2048 moves , data stored in external flash  SPI W25Q32BV . i look now in my old code and if **float   dt   = 0.0000368f;** was running on each 36us ,so thats is 2777 times / second. This was first start, i cant tell you exactly because the algorithm itself adjust the time after turn on and depends on my analog read , was made with active feedback on each read.
First attempt was complementary filter , not very stable, the second was kalman filter and last this one. 
If you look carefuly in video , the height dont matter , the weel looks like stay still you can see on LED all moves cant see on weels,  that because the algorithm predict the next move long time ago , in my mind i expect on 600 moves below.

wangine, 2016-07-20:
I cant test right now , i just finish my relay board , next days i will replace my old controll with this one , now i just test all functions or your **great** project . 
If you want i can send you entire project on email , not here , is not for share , there cand find alot of controll test, is under C but thats not a problem for you.
So , back now on RPi , i just play a little bit with Geany compiler , are nice and i can say can made any function you like to controll IO. Basicaly you can do everything you want on that board, still i cant find the **interrups** and **timers** , if they exist.

kizniche, 2016-07-20:
Ahh. I understand it all now. It was throwing me off because your variables were named as P, I, and D gains, yet their purpose is different.

I still like the idea of optional pre and post filters. Pre filters would smooth out the input signal, and post filters would smooth out the output signal.

I've also wanted to attempt to create a PID auto-tuner that could calculate the best gains, based on a few automated experimental runs to determine how fast the environment reacts to different power outputs (relay modulation). Still, auto-tuning would be a big project. The most questions I get from Mycodo users is about tuning. It's an art, really. I've tuned a few systems now and can figure out the gains fairly quickly. They're a pain if they're not working right, but when properly tuned, it's an amazing thing to watch.

[image]

wangine, 2016-07-20:
That why i hate PID. Became annoing some time trying to tune and tune , dont matter waht PID you use , exist to more types , the problems will start with **tunning** . If you want good filter , you can find in my project , or i will upload here , low pass and high pass , Look in email , the file name "**complementary_f.c**"
And again you right , not all users are full of math to know and understand PID , for a normal user should push RUN and expect to run.

wangine, 2016-07-20:
For Mycodo is necesary just **lp_filter( )**  function . Once you set **float fill_rate_lp =  0.9689f;** on right number according with your PID call/second , dont need any adjustment . Just will do the job.
The function have just 4 line of code 
I compiled my first program under RPi , of course the epic blinking LED and its work , If you want i can compile filter for you ,

wangine, 2016-07-21:
I just look in your code , still that PID was made for PWM output controll , like that can be adjusted very precise, if you look in last line 
    _def run_PID_t(self):
        self.intrare_t = final_temp
        self.lastTime_t = get_ticks()-self.SampleTime
        gap = float(abs(self.Setpoint_t-self.intrare_t))  # Distance from setpoint
        if gap > 2 or gap < -2:
            auto_tune_t()
        Compute()

        SetOutputLimits(0, 500)
        **set_pwm1_duty(int(self.iesire_t))**_
That function call all PID modules , and here **set_pwm1_duty(int(self.iesire_t))** need to call **your relay function**  , only like that will work , otherwise the result will be null. And that function **SetOutputLimits(0, 500)** just limit the end of PWM module in my controller, over 500 my PWM will owerflow. Sorry for miss understanding , that code was 3 years old , now i remember because i change all my controlers for Mycodo. . I did't finish the hardware , but will be soon.

kizniche, 2016-07-21:
I'd like to have all code for Mycodo in the same language (except for mycodo_wrapper.c), so if something is going to be integrated into Mycodo, I'd prefer it be in Python, unless there's a compelling reason not to. I haven't had a chance to look at the code you sent, and I don't know how busy I'll be the next two days, but I'll be traveling to New Orleans for the annual SIMB science conference, so my git activity will probably slow or stop during that time.

wangine, 2016-07-21:
Ok , no problem , take you time

wangine, 2016-07-27:
About first post . i've try to understand python in last week to find a way to add days instead of hours or minutes in PID, so i was not realy nice impressed about python . First was the annoing rule **IndentationError: unexpected indent** , and many others but i solved and understand all. The second thing python is not very _smart_ , i just add some random **love storry** lines , without sintax mistakes , and pythom didn't find them wrong, and in last,  i write so many codes in last years in C and C++ and i NEVER use debugger , in python is imposible without.
Back to subject , i found some weird code in **controller_pid.py** file on line 390 
_(start_hour == end_hour and start_minute > end_minute)):_ but i cant find **start_minute** reference , 
i assume should be **start_min** and **end_min** .
Correct me if i am wrong .
Also i cant resolve reference in other file **mycodo_flask.py** on line 882 the code for camera 
**resp = flask.make_response(open(timelapse_path+filename).read())** , if i change **flask to Flask** the error i think will gone .i assume **flask** is the file and **Flask** the class. I'm not realy sure , i need more studdy.
I try to help because realy i want to add different setpoints between days .

kizniche, 2016-07-29:
I have a system in planning that doesn't suffer from the current limitations of a one day loop.

```start_min``` is on line 409, but you are correct about the other variables being incorrectly named.

The line 882 is also a mistake, but time-lapse hasn't been implemented yet so not really an issue.

kizniche, 2016-07-30:
I managed to build the frontend of the new method builder. It allows for date/time or duration method building. It also will allow relays to be manipulated during the execution of the method. Here are a few screenshots of the progress. It currently only verifies the input as being valid (end time/date must come after start time/date, must be integers between x and x, etc).

[image]

[image]

wangine, 2016-07-30:
I assume that is still in progress .This new method requires that you owe more beers .I suppose the new form will be included under  PID page , right ? . Congratulations, do you really interest in the project

kizniche, 2016-07-30:
Yes, very much work in progress. I've only had one beer today, so I think it's time to take a break for another. Yes, there will be an option to select what kind of method to create from the PID screen (see below). I finally got the database sorted so I can begin adding time spand and setpoint changes (below).

[image]

[image]

kizniche, 2016-07-31:
Fairly good progress today. There's now a central area to view, create, and modify methods (below). I'd like to eventually use HighCharts (what's used on the Graph page) to display a graphic to be able to easily visualize each method.

[image]

wangine, 2016-07-31:
Yes , is realy alot of work .even is your project and you know all the links between files, is hard enouth . With that tool you can do everything you want on setpoint . Still i've seen a little bug under PID section , i assume you use you own validator on integers field. If you have set one PID (device dont matter , can be RPi sensor) and you try to modify the data of PID (setpoint for example) and in typo mistake you set a wrong value (ex. 31,00 instead of 31.00) the validator will return the error , thats a good thing but after you correct the PID will stop forrewer. Single way to make it work again is to delete and add again under other PID name , values dont matter if are the same like the first one but if the new PID will have the same name like a deleted PID ,again will dont work . I think is  remain some data in databases or values remain with last error.

kizniche, 2016-07-31:
I can't replicate your issue. Could you create another issue and go through the exact steps to bring about the issue?

wangine, 2016-07-31:
Is not a big problem , realy , only if you miss input a field in a typo mistake . Take ypu time , i will make more tests. Right now i have all my hardware finished and i have to test some days to be sure what i say. I just notice yesterday in my typo mistake ,if the daemon return a error the PID will stop . I will try now

wangine, 2016-07-31:
Sorry about that . The problem was in my system (i think) or maybe on jessie full , i switch SD card with jessie lite and no any issue . If that will happen again i will let you know.

kizniche, 2016-08-01:
A little more progress today. I Have a completely working method builder with options for wither date/time or durations. I still need to build in the ability to add a repeat command so methods can repeat at a defined time. Not exactly sure how to do that most efficiently, but I'll think about it tomorrow. Here's a few screenshots of two test methods, one time/date and the other duration.

[image]

[image]

wangine, 2016-08-01:
I am extremely excited to see the first update . It is one of the most powerful tools in Mycodo

kizniche, 2016-08-01:
I'm hesitant to push it just yet because I may change some of the database variables (which could mess up users' databases if they already go ahead and start building methods). I haven't coded anything on the backend (daemon) yet, but I think I have everything working on the frontend (except the option to repeat). I'd like to tackle the repetition part before pushing to github for people to start testing for bugs.

I think coding the backend will be much easier and quicker than the frontend, so I'm estimating pushing the frontend update early this week and the backend later this week or this weekend. Who knows, though.

kizniche, 2016-08-02:
Things are getting pretty:

[image]

kizniche, 2016-08-02:
Alright. I think I have it at a stable place to push an update! What an update (dfa50ee)... 1,582 additions and 21 deletions!

wangine, 2016-08-02:
Wow , realy a nice job, the graph included in that method is awesome ideea. Right now i work to calibrate my humidifier (is home made based un ultrasonic humidifier and device point for des humid https://en.wikipedia.org/wiki/Thermoelectric_cooling ), so right now i keep made tests on PID page .

wangine, 2016-08-03:
Yea , look great , still , the "year" can miss, i dont think some one will build a method over one year. However that method can gain all possibities , will do the best. 
I've seen now on first graphic page , you choose gains very low , below 0.00 /.? I dont get it , i always start with 1 as proportion , to find the best P gain for my room, after will follow the rest. 
And with Peltier , yeea is the best solution to condense the humidity with less resources , a small fan and a peltier , in 1/2 h can take out 60% humid on 20m/2 .

kizniche, 2016-08-03:
The year can be useful, for instance, if it's December and you want a method that will go into January of the next year.
P values aren't that important when you use I and D, because those will change the output. If just using P, then it's crucial to have a good value. The photo of the PID you're referring, if you look up to my previous photo of the graph output, you'll see that the relay on duration is just below 1 second every period (30 seconds). If the error is 1 (1 degree from the setpoint), then the output will be 0.8, so it is pretty well-tuned at that P value. The values will depend on your environment... Mine just happened to be a small test environment that was insulated and using a powerful heater, so I didn't need a lot of power to heat it.

wangine, 2016-08-03:
ok ok , i get it , i assuming is hudge room for mushrooms , so that why my miss understanding. For humidity i did't use derivative term yet , and still you controller do a good job , i need time to try , because i used on my old system all with PWM(is a little engineering trick to use PWM on alternating current sources, if you want i can make quick a schematic to send you) , and now with relays is different.
Return on method PID page , i've seen you included under PID page submenu , I just asking ,, is more easy to debug than an independent page like sensor or LCD page ?
And with year , yes you have perfectly right , i dont know why i miss that

kizniche, 2016-08-03:
I'm not sure what you mean by this:
Return on method PID page , i've seen you included under PID page submenu , I just asking ,, is more easy to debug than an independent page like sensor or LCD page ?

wangine, 2016-08-03:
No , i want to say ,, "you created a form method in PID page instead to make a new page (like Timer page or Live page or Log page)".

kizniche, 2016-08-03:
The menu is easy to change. I don't want to add too many things to the menu because it will overextend the viewable area on smaller displays. I will most likely rearrange the menu soon.

kizniche, 2016-08-04:
Oh, I see what you're saying. I actually did link it on the menu.

Go to Tools->Method Management

kizniche, 2016-08-04:
That was a good idea. I just updated the menu.

wangine, 2016-08-04:
yes . look very nice , anyway today i made the first FULL test on all systems. The both direction on PID controller do a great job , in 40 min reach setpoint from 48%RH to 92%RH , and 1h.20min to decrease again to 46%RH from 92%. So controller work great , my condenser is to small for that room . 
You want to keep Mycodo only on Python , so i research and test some filters for you in Python.
Can take a look .Require NUMPY
https://pypi.python.org/pypi/padasip
http://matousc89.github.io/padasip/tutorials/tutorial1.html
http://matousc89.github.io/padasip/tutorials/tutorial2.html
http://matousc89.github.io/padasip/tutorials/tutorial3.html
I believe the NLMS filter is more than enouth to made a pre and post filter .

kizniche, 2016-08-04:
Good to hear! What's the size of the space in cubic meters?

The filter idea is good, but how could it be applied to Mycodo?

wangine, 2016-08-04:
my space / ? wow is hudge , now on pre-fermentation stage , when the leaf need to transform from green to yelow and fix that colour is  9x6x2.8 arround 150m/3(procces take 6 days). The final fermenting chamber is a small one , 12 m/3 (the process take 3 months and many RH and temp stages ). I was realy surprised how good job do the relays. Based on PWM all time for me was a chalenge . For humid i use 8 pieces of (do amazing job)
http://www.aliexpress.com/item/New-Ultrasonic-Mist-Nozzle-Fogger-Fog-Water-Fountain-Pond-Atomizer-Air-Humidifier-Garden-Sprayer-23525/1684133084.html?spm=2114.13010308.0.133.v80hkP
and for dez humidifier a regular peltier based on devpoint.
I send you an email with a FTP link to my documentation off tobacco . You downloaded ?

kizniche, 2016-08-04:
That is huge. I've used those ultrasonic humidifiers before. What's nice about those is you can use multiple relays to control them all, and if one relay happens to stop working, it won't affect the rest. The same is not true if you hook all the humidifiers up to the same relay. Also what's nice is the PID will compensate if 1 humidifier or relay stops working, and will increase the power (time on) to the other relays to maintain the humidity level.

I think I missed the window you said before your IP changes, so I wasn't able to download it.

[10 more replies]

closed 2016-09-02 by kizniche