kizniche/Mycodo #95: I2C Multiplexer 9545a

KnewWX, 2016-07-21:
Hi!

I'd like to start off by thanking you for posting your hard work on here.  I'm just starting out and going through examples of what people have done and reverse-engineering them have been extremely helpful to my understanding of Python and the hardware.  With that being said, I have run across a few issues and I was curious if you'd be able to shed some light on them based on your experiences.

I recently purchased a raspberry pi 3, DS1307 real-time chip, (2) AM2315 sensors and an I2C-Mux-9545a (with grove connectors - this board has 4 expansion slots, not 8).  I am having difficulty getting the pi to recognize/read the sensors that are hooked up via the Mux board.  They are recognized on the breadboard as stand-alone sensors and are reading out just fine with the likes of tentacle_pi, but not through the Mux and the testing files.  I can test the mux and it does recognize the 4 expansion slots through a .py test file, but not the sensors connected to the mux card.

Here's the rundown of everything with questions pertaining to the specific problems:

1)  I've been trying to install the proper files for the Mux from here:

https://github.com/camrex/i2c-mux-pca9545a

I've noticed that your .py files are for the 9548a.  How were you able to copy the files to the /boot/overlays/ folder?  I've tried sudo chmod commands, git clone, right-clicking on the folder...everything under the sun and I still am not allowed permissions to copy anything into that folder location.  Is there a different set of files or a work around?  I want to believe that if I can just copy the file to, or even amend the existing '9548' file in the folder, this will solve all my problems, but I am not sure.

2)  I have added the proper command line (dtoverlay=i2c-mux-pca9545a) to the config.txt.  Are there any other files/command lines that need to be amended for the 9545a Mux?

3) SMBUS needs to be used in conjunction with the Mux cards.  Is there another open source that would work instead? Tentacle_Pi? Quick2wire.I2C?

4)  Once I am able to get the AM2315 sensors working through the Mux, do you have a sample .py file that is easily amendable for the I2C-Mux-PCA9545a card?

Thank you so much in advance for your insight and help!!

wangine, 2016-07-21:
@KnewWX  First you need root account , that is't by default. Open terminal and type/
**sudo passwd root**  
and type you root password
type **su** in terminal to login as root 
and type you passw , now you are root.
If you want to alow ssh as root type/
**sudo nano /etc/ssh/sshd_config** 
Search for PermitRootLogin and change it to yes , if dont have that field just add 
_PermitRootLogin yes_
reboot ......
If you want visual edit , change ,move , delete,  permisions over ssh , i use https://winscp.net/eng/docs/guide_install
SCP protocol will do fine

kizniche, 2016-07-21:
Hi @KnewWX,

First, if you install Mycodo, you will not need to install anything else to get the TCA9548A working. Mycodo installs all the necessary modules.

Second, if you have or plan to have your Pi exposed to the internet for remote access, I would recommend against allowing root login on SSH. This is a security risk, as the majority of attempted hacks via SSH are directed at the root user. Disabling root login will prevent this. You can easily access root privileges once logged in as a user. Since a hacker has no idea what users are on your system, it's exponentially more difficult for them to gain access by trying to guess what your user is. "root" , on the other hand, is universal on linux systems, and by allowing this user login via SSH, you're giving them a free account to try to brute force attack. You can easily issue "sudo" in the terminal to execute commands as root. This is in fact the preferred way to work, as it's risky to be logged in as root... It's far safer to remain logged in as your restricted user and issue sudo at will when needed.

A quick anecdote: Back in the early 2000s, I learned the hard way about being logged in as root. Back when the "rm" command had no checks for erasing your entire hard drive, I issued the command:

```
rm -rf /
```

when I meant to issue the command:

```
rm -rf ./
```

The second command would have been fine, but the command I actually typed, without the period, started erasing my entire hard drive. I caught what was happening as I saw my home folders scroll up the screen, as my home directory was being erased. I canceled (Ctrl+c), but it was too late. I lost all my users data and after a reboot, my system would not work, as expected.

If I was logged in as my user instead of root, I would have received a "permission denied" error, and saved myself a bunch of anguish.

kizniche, 2016-07-21:
To use the multiplexer function in Mycodo, merely select what I2C address your multiplexer is assigned and select what channel your AM2315 sensor is connected to, save, then activate the sensor. It should be that easy.

KnewWX, 2016-07-25:
Many thanks for the assistance on the overlays folder,the workaround in the terminal did the trick.

Now though I cannot get my mux to work properly, let alone test properly.  Here are the two installation routes I have traveled for the mux-9545a.

1) The first was from where I purchased the mux - switchdoc labs. When I use i2cdetect, I get a ’73’ reading, showing it is functional, however, when I do a list of the i2c from the /dev folder, I only get i2c-1. It does not recognize the individual buses.

2) There is another set of setup instructions floating around that I have followed, whereby I setup the command ‘dtoverlay=i2c-mux-pca9545a,addr=0x73’ in the config.txt file. When I do this, then run an ‘ls’ from the/dev folder, I get ‘i2c-1, i2c-3, i2c-4, i2c-5, & i2c-6’, essentially recognizing all my buses on the mux. However, when I run the i2cdetect, I get ‘UU’, which is a definite error readout.

I have backtracked on option #2 and commented out (#) the line in the config.txt since UU isn't good.  I can run a test file from github for the mux itself and it acknowledges that it is working.  I've gone through your Mycodo library and found the Test_I2C_Multiplexer.py script in the Mycodo/mycodo/tests/ directory.  When I run it, I get a 'too few arguments' error.  I have just (1) AM2315 sensor attached on bus 3 (from 0-3 (the 4th grove)).  I've opened up the .py script and seen where, I think, to make alterations based on the mux you have:

def I2C_setup (multiplexer_i2c_address, i2c_channel_setup) 
      I2C_address = 0x70 + multiplexer_i2c_address %10

So, I would change:
     0x70 -> 0x73

Here are my questions:

1) Am I correct in the alternations that need to be made to your .py file in order to properly test all the buses on the mux board (4 of them)?
2) Do you think that I could be missing some sort of software installation in order to make the mux work/be recognized properly in the scripts?
3) Once the mux does get working, is there a script in your library that is editable/will recognize the AM2315 sensors on the mux itself?  

Many Thanks!

kizniche, 2016-07-25:
What you're seeing- /dev/i2c-1 is normal. You don't need to touch the config.txt file for I2C to be enabled. It should only be enabled with 'sudo raspi-config'. Issuing 'sudo i2cdetect -y 1' should show it is detected (as you confirm).

Next, look at the source of the multiplexer test script. There's usage instructions at the top with an example command. Should be 'sudo ./Test_I2C_Multiplexer.py 73 1'

Give that a try and let me know what the response is.

KnewWX, 2016-07-25:
sudo: ./Test_I2C_Multiplexer.py: command not found

kizniche, 2016-07-25:
You have to cd to the Mycodo/mycodo/tests/ folder first.

KnewWX, 2016-07-25:
File "./Test_I2C_Multiplexer.py", line 38, in <module>
     menu()
File "./Test_I2C_Multiplexer.py", line 35, in menu
     I2C_setup(args.address, args.channel)
File "./Test_I2C_Multiplexer.py", line 24, in I2C_setup
     bus.write_byte(I2C_address, i2c_channel setup)
IOError: [Errno 5] Input/Output error

kizniche, 2016-07-25:
And you can confirm i2cdetect properly detects your multiplexer at 0x73 before running the test script?

KnewWX, 2016-07-25:
yes, it is there.  shows up in the crosshairs of 70 & 3, reading 73

kizniche, 2016-07-25:
Hmm. I may need to connect to your Pi to perform some tests, if that's alright. I'm not sure what's wrong. Can you give me a link to where you purchased it so I can see if the hardware differs from my breakout board?

KnewWX, 2016-07-25:
http://store.switchdoc.com/i2c-4-channel-mux-extender-expander-board-grove-pin-headers-for-arduino-and-raspberry-pi/

kizniche, 2016-07-25:
The most likely case is this board requires different code than what's in Mycodo to operate. I'm on my phone right now, so I won't be able to compare their code with mine until sometime later. Rest assured, I'll try to integrate their code into Mycodo, if I can ;)

KnewWX, 2016-07-25:
I appreciate your help!

kizniche, 2016-07-26:
Try this:
```
cd
git clone https://github.com/switchdoclabs/SDL_Pi_TCA9545
cd SDL_Pi_TCA9545
chmod +x ./testSDL_Pi_TCA9545.py
sudo ./testSDL_Pi_TCA9545.py
```

kizniche, 2016-07-26:
I see that I've been reading things wrong this whole time. Mycodo has support for the TCA9548A but you have the TCA9545. It's a different device. I'll have to add a new module to add support for it. When you get a chance to run the above commands, let me know if it works, and I'll integrate the code int Mycodo when I get a chance. I won't be home until this weekend, so I may not get to it until then or shortly after.

KnewWX, 2016-07-26:
This script works, prior to your instructions.  This is the one that did where I could see it checking the busses, but not recognizing the sensors.  I appreciate you incorporating the 9545 into your code.  LEt me know and I'll be happy to test it this weekend.  Thanks!

kizniche, 2016-07-26:
That's good news.

The muliplexer doesn't read the sensors. The multiplexer script merely changes what channel is currently active. For instance, if you have two sensors, both at address 0x40, connected to channels 1 and 2, and you switch to channel 1, then when you communicate with the sensor at address 0x40, it will communicate with the first sensor. Change the multiplexer channel to 2 and then read 0x40, then it will be communicating with the second sensor.

You will still need separate code to communicate with the sensor. Mycodo handles all the channel switching and sensor reading.

kizniche, 2016-07-29:
Can you paste your output from testSDL_Pi_TCA9545.py?

kizniche, 2016-07-29:
Also, I realize now, after having more time to read the docs, that this device most likely acts differently than the Adafruit I2C multiplexer I was confusing it with.

What is the possibility of being able to ssh into your Pi while the bus multiplexer is connected with a sensor or two connected to it?

KnewWX, 2016-08-02:
Yes, you can ssh in and test.  I welcome it!  IS there an email or PM with this site to chat away from board posts?

kizniche, 2016-08-02:
You can message me at http://kylegabriel.com/contact/

kizniche, 2016-08-19:
The sensor i2c bus option I added with commit b483f28 should allow the Grove Mux, TCA9545A, board to work with Mycodo. Give it a test and let me know how it works.

closed 2016-09-08 by kizniche