kizniche/Mycodo #96: daemon don't work after web update

gremlintomsk, 2016-07-22:
After web update to 4.0.8 version daemon does not start. Error:
_ImportError: No module named Ardafruit_Ads1x15_ 
**pip install adafruit-ads1x15** fix it

kizniche, 2016-07-22:
This is a known issue and I believe it is due to pip, not mycodo.

kizniche, 2016-07-22:
I pushed an update that affects the install process of the Python modules. It may not play nicely with current installs. What it does is uninstall any sensor modules before the installation of all required modules, which would ensure there are no incorrect versions of the modules being used.

You can bring your Mycodo install up to speed with the following commands:

```
cd ~/Mycodo
sudo pip uninstall -y Adafruit_ADS1x15
sudo pip uninstall -y Adafruit_BMP
sudo pip uninstall -y Adafruit_GPIO
sudo pip uninstall -y Adafruit_TMP
sudo pip uninstall -y MCP342x
sudo pip uninstall -y Adafruit_DHT
sudo pip uninstall -y Adafruit_Python_DHT
sudo pip install -r requirements.txt --upgrade
```

closed 2016-07-25 by gremlintomsk