kizniche/Mycodo #97: Time is bugged, after power loss

itzik16, 2016-07-24:
Hello, I've just made a fresh install of mycodo and btw the install script is a bit buggy I had to remove all the pip remove commands in order to make fresh install you should check if they are installed before you force uninstall them which cause an error and abort the installation.
any way the main reason I opened this issue is becuase of the time, the system time is correct I've checked in shell and on mycodo tools too.
but the live readings are from says it 5 hours ago and still work as it should, the readings still works and added to the database, I think you should add the option to choose where to grab the time stamp for, for example external time server, which you can select the correct timezone.
thanks in advance mate, keep on the great job. :+1:

kizniche, 2016-07-24:
Ah, you're right! I forgot I put a bit of code in to exit on an error. I'll fix that. I should remove all the uninstall commands... It's really not helpful unless you have an old install on the same system.

I'll also look into the time issue. It should work as long as the Pi is set to the correct time.

itzik16, 2016-07-24:
you are really dedicated for this project :D just wrote this issue and its already being checked haha 
you are awesome mate

kizniche, 2016-07-24:
I've got nothing better to do on the ride from the airport to my hotel, here in New Orleans ;)

kizniche, 2016-07-24:
Seeing your edit, I suspect it may be your time zone on one of the systems.

kizniche, 2016-07-24:
It should be storing the data with the timestamp in UTC, then the web browser should be converting UTC to the timezone of your web browser's computer.

itzik16, 2016-07-24:
my browser may not share the timezone and locale, because of the privacy settings, the problem appears on the graph too by the way.
and I have double checked three times that the pie and the computer are both set to the same time zone.
is there a way to add hard time set option for the live readings time from the configuration panel?

kizniche, 2016-07-24:
There aren't currently any Mycodo time settings. You're the first person that's brought up an issue, besides from one of my systems that the time was incorrectly set.

itzik16, 2016-07-24:
I understand, but since this system controls relays that control lights,pumps and sensors and supposes to allow the user to maintain and monitor artificial environment I think that this option should be added for more reliable work-flow and readings.

kizniche, 2016-07-24:
What would you like to be able to set?

itzik16, 2016-07-24:
either the timestamp or the way it being converted, i think it should have an option to be manually set or set automatically by external server, and if none of the options above would be chosen it should work the same as is works right now. this way it will optional to configure that option in case there is no problems

kizniche, 2016-07-24:
There aren't any options given when the data is sent to be stored in the influx database (influxdb), so influxdb handles creating the timestamp when it's entered (UTC, based on system time). I'm not really sure how to pass settings on that end.

As for the graphs, the only thing time-related that's set is:
```
useUTC: false
```
Which I'm not exactly sure what it does... I added that a long time ago.

wangine, 2016-07-24:
Maybe it help , https://metacpan.org/pod/Device::BCM2835::Timer

kizniche, 2016-09-08:
Any chance of having access to your time-bugged system to diagnose the issue?

itzik16, 2016-09-08:
No I have made a fresh install of linux and mycodo and it seemed to fix it

kizniche, 2016-09-08:
Okay great! Thanks for the update.

closed 2016-09-08 by kizniche