kizniche/Mycodo #98: HTU21 sensor support suggestion

zsole2, 2016-08-05:
I just came across this page:
https://learn.adafruit.com/adafruit-htu21d-f-temperature-humidity-sensor?view=all
Nice little sensor, I2C connection, cheap clones, what's not to like?

kizniche, 2016-08-05:
I agree that Adafruit makes some good boards. But, I would have to either purchase it or find someone with one, to be able to run tests with the new device code to make sure it works with Mycodo.

There are a lot of potential sensors that can be added, so I have to be strategic about which ones to add if I'm going to be purchasing them myself. Since there are already a few temperature/humidity sensors supported, I'm less inclined to purchase another to add support. I'm currently looking to expand support for sensors that measure conditions that currently-supported sensors don't measure, such as the Atlas Scientific line (pH, conductivity, oxidation-reduction potential, etc.).

kizniche, 2016-08-09:
I've been looking around and I've been able to find a lot of the HTU21D breakout boards fairly cheap on eBay. The difference between the HTU21D and HTU21D-F is the filter membrane over the sensor, to protect from particulate and water (light) intrusion. I haven't been able to find any of the HTU21D-F breakouts for cheap. It seems like Adafruit is the only supplier of it with a breakout board.

zsole2, 2016-08-10:
I understand your point on adding different sensors. On the other hand, the alternative I2C sensor (AM2315) is much more expensive, especially if someone can live with the  HTU21D. I was looking around since I am having problems with the DHT sensors' lifetime (none of my units were long-lived in the 95% humid environment)...

kizniche, 2016-08-11:
I do think temperature and humidity is probably the dominant set of conditions people will be wanting to regulate, so a diverse selection of sensors is a good idea. At the moment, though, I don't have plans to buy any new sensors. Though, this could change in a few weeks, when I know more about the fate of my stolen/recovered motorcycle. But if you happen to pick up a sensor, and there's Python code available to read it, I'll try to put it in Mycodo.

zsole2, 2016-08-11:
I plan to get the sensor, and since it takes about a coupe of weeks to get here, I'll have time to search for the python code...

zsole2, 2016-08-11:
This looks like a good starting point:
https://nwgat.ninja/getting-htu21-to-work-with-pi-zero/

zsole2, 2016-09-01:
Kyle,
Just to let you know, I have received this sensor: 
http://www.ebay.com/itm/152187301140?_trksid=p2060353.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT
and can confirm that it works with the instruction cited above.

kizniche, 2016-09-01:
I just pushed a update that includes the new module for the sensor. Can you update your Mycodo and then issue the following command and tell me if it correctly returns the temperature and humidity?

```
sudo python ~/Mycodo/mycodo/sensors/htu21d.py
```

kizniche, 2016-09-01:
I'm going to just go ahead and push the whole sensor module integration update. I just finished and you can test it in Mycodo now.

zsole2, 2016-09-02:
Thanks, that was quick!
Before leaving to work, I updated just now, and the test line you provided gives the following error:

i@chz-test:~ $ sudo python ~/Mycodo/mycodo/sensors/htu21d.py
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/sensors/htu21d.py", line 126, in <module>
    print("Temperature: {}".format(measurements['temperature']))
TypeError: 'NoneType' object has no attribute '__getitem__'

Also, I can add the sensor from the GUI, but the value on the Live page is 0 for everything.

zsole2, 2016-09-02:
I have found two problems with the code:
-  I had to change self. I2C_bus_number to a number instead of a string.
-  when calling from the read-hum function the read_temp function, it returns to the read function with  error and exits, some kind of stack problem must be... Inserting the code for read_temp function, everything goes througf without error. It does not look nice, but works.

kizniche, 2016-09-02:
I just changes the strings to integers in the sensor module. Go ahead and update and test again, please.

Are you saying that was not the root cause of the error? If it's still not working, could I have ssh access to your Pi to perform some further tests?

zsole2, 2016-09-02:
Yes, the integer issue was only a part of the problem. 
I downloaded the pudb debugger, and set a  breakpoint on the line  

temperature = read_temperature()      

in the read_hum function.

When running it after the break it returns from the function, and continues on the next line (except:).

Therefore the nested call of the class is not working.

If I change the code as follows, then it works (only from the terminal, not from the GUI).

    def read_humidity(self):
        handle = self.pi.i2c_open(self.I2C_bus_number, self.address) # open i2c bus
        self.pi.i2c_write_byte(handle, self.rdhumi) # send read humi command
        time.sleep(0.055) # readings take up to 50ms, lets give it some time
        (count, byteArray) = self.pi.i2c_read_device(handle, 3) # vacuum up those bytes
        self.pi.i2c_close(handle) # close the i2c bus
        h1 = byteArray[0] # most significant byte msb
        h2 = byteArray[1] # least significant byte lsb
        humi_reading = (h1 * 256) + h2 # combine both bytes into one big integer
        humi_reading = math.fabs(humi_reading) # I'm an idiot and can't figure out any other way to make it a float
        uncomp_humidity = ((humi_reading / 65536) * 125 ) - 6 # formula from datasheet
        # to get the compensated humidity we need to read the temperature
        handle = self.pi.i2c_open(self.I2C_bus_number, self.address) # open i2c bus
        self.pi.i2c_write_byte(handle, self.rdtemp) # send read temp command
        time.sleep(0.055) # readings take up to 50ms, lets give it some time
        (count, byteArray) = self.pi.i2c_read_device(handle, 3) # vacuum up those bytes
        self.pi.i2c_close(handle) # close the i2c bus
        t1 = byteArray[0] # most significant byte msb
        t2 = byteArray[1] # least significant byte lsb
        temp_reading = (t1 * 256) + t2 # combine both bytes into one big integer
        temp_reading = math.fabs(temp_reading) # I'm an idiot and can't figure out any other way to make it a float 
        temperature = ((temp_reading / 65536) * 175.72 ) - 46.85 # formula from datasheet
        humidity = ((25 - temperature) * -0.15) + uncomp_humidity
        return humidity

kizniche, 2016-09-02:
Oh, I see what I did wrong. Stupid mistake, really. I'll have an update pushed shortly.

kizniche, 2016-09-02:
Oops... still an error. New fix coming shortly.

kizniche, 2016-09-19:
Strangely, I have two DHT21D breakout boards I got from ebay, which appear to be old sparkfun boards that are no longer sold by them. With both boards, Mycodo returns a large negative temperature (-45) and a normalish humidity (30%) (I say normalish because for relative humidity to be properly calculated, it needs an accurate temperature).

Reopening this issue to remind me to try to fix this. Cursory glances to the datasheet show I have the correct equation to calculate humidity and temperature. Also confounding is that @zsole2 has breakout boards with the same HTU21D chip.

zsole2, 2016-09-19:
I coincidentally just installed my HTU21D sensor into my working system this morning.
It seemed working OK then, and now as I checking from work, it looks still OK. When I get back home, I'll put the HT22 back to see how much difference there is between the temp readings. Certainly it is not freezing over, although it is within a fridge...

closed 2017-05-24 by kizniche