kizniche/Mycodo #100: No data from DHT22

pewi81, 2016-08-14:
Have had Mycodo setup and running fine for a few days but suddenly no data was collected from my DHT22 sensor. First time the error occurred was after a reboot of my Pi, there was no data collected at all! I couldn't get around the problem so I did a new install of both Jessie and Mycodo, after that all was fine for about 23 hours when suddenly the data was stuck on 25.1 degrees C and RH at 80.7%, I rebooted my Pi and after that it won't collect data from the sensor at all, show only zeros!

kizniche, 2016-08-14:
Is it actually returning 0 or is it saying no data received in the past 30m? Are you using a 3.3 or 5-volt power source for the sensor? Do you have additional sensors connected or other peripherals that would be using the same power supply?

pewi81, 2016-08-14:
It says no data received in the past 30 mims. I have it connected to 3.3V and gpio 04, nothing else is connected!

kizniche, 2016-08-14:
Try using another GPIO or 5-volt power. 3.3 volts is sometimes not enough if you use a long or resistive wire (I've used 400 feet successfully with the DHT22).

kizniche, 2016-08-14:
Also, what value resistor are you using with the sensor and are all your connections soldered (except where connected to the Pi)?

pewi81, 2016-08-14:
4.7K ohm, all soldered except for connection to Pi. I tried 5V and different gpio's but still same problem!

kizniche, 2016-08-14:
Double-check your connections. If they're all good, it seems like you may have a bad sensor. Do you have anything else that you can read the sensor from to verify it's working (arduino or another Pi)? Do you have another DHT22 to test? I've had a DHT22 connected to my development system running for months now, without issue, so that's what's leading me to believe it's some kind of hardware issue.

pewi81, 2016-08-14:
I will check my connections and resolder if necessary, I will get back once its done! I will also receive a new dht22 next week for another project, I will connect it to Mycodo if the problem continues!

pewi81, 2016-08-14:
Alright, so I re-soldered everything but still got nothing, even tried the sensor with an Arduino but no luck. Turned out the problem probably was the resistor being broken, I replaced it and everything seems to work fine now. Next time I will check the hardware more thoroughly!
Many thanks for the assistance!

closed 2016-08-15 by pewi81

kizniche, 2016-08-15:
Glad to help. Most issues are an easy fix but may not be obvious.