kizniche/Mycodo #101: Timelapse Photos

berserktron3k, 2016-08-21:
Is there a length limit for time lapse photos? I'm trying to do one every 3600 seconds (an hour) for 345600 seconds (5 days). Regardless of the length, it always seems to stop around 11 hours.

kizniche, 2016-08-22:
I've been running a test for the past 10 or so hours. I'll check if there are any errors that I can find tomorrow.

kizniche, 2016-08-23:
My initial test worked fine. The time-lapse didn't end prematurely. I'm going to to a longer test while I'm out of town the next few days and see if it stops.

berserktron3k, 2016-08-23:
Interesting. Can you tell me what settings you used? Frequency/length?

kizniche, 2016-08-23:
Interval of 3600 seconds (1 hour) and length of 172800 seconds (2 days).

berserktron3k, 2016-08-23:
Thanks. I'll set mine the same and see if I get the same results.

Brady Hummel

kizniche, 2016-08-23:
I may be redesigning how the timelapse function works. Currently it does not resume if there's a power outage or if mycodo restarts for some reason. I'd like it to be able to resume the timelapse and the current method makes it difficult to implement that. When/if I do change how it's done, it should clear up any issue you have.

berserktron3k, 2016-08-23:
Sounds good. This may be a non-issue altogether. You won't hurt my feelings
if you close the issue.

As always, thanks for your hard work on all of this.

I've been using the Etekcity 433MHz RF outlets via a python script I wrote
too. Seems to be working pretty reliably. Not sure if you're pursuing that
further or not, but so far I've had good luck.

Brady Hummel

kizniche, 2016-08-23:
I haven't forgotten about the 433MHz outlets. I got mine working but haven't tried to code them into the relay code. It's a little more involved than I initially thought and have been waiting for a good bit of downtime to try and get started on it.

As for the time-lapse, I'd rather not try to diagnose an issue with the current system if I'm going to rewrite it. I'm actually starting to write the code for the new system now. I may have it done by tonight and push an update to v4.0.13. It seems like I have a good idea of how to accomplish it and it appears to be pretty straightforward to code. We shall see!

kizniche, 2016-08-23:
Well, that was a pretty fun little challenge. I managed to rebuild the time-lapse functionality to include all the features I set out to accomplish. Time-lapse can now resume if the Pi reboots or mycodo restarts. It will keep the correct image number order as well. For instance, if you set the following parameters:

```Interval: 10 seconds, Duration: 60 seconds```

It will create the first images:

```
2016-08-23_18-20-54-img-00001.jpg
2016-08-23_18-20-54-img-00002.jpg
```

If the power fails and it reboots in 25 seconds, it will resume with image:

```2016-08-23_18-20-54-img-00005.jpg```

And continue on until it reaches the set end time defined from the duration. An update to git will be incoming in a minute or so. Enjoy and let me know if you find any bugs!

closed 2016-08-23 by kizniche

kizniche, 2016-08-23:
Feel free to reopen the issue if you find a bug. It automatically closes the issue if I mention the issue number in a commit message with the word ```fixes``` before it.

kizniche, 2016-08-26:
My time-lapse that I started on Tuesday is still running. It's using the new time-lapse method I built on Tuesday.

berserktron3k, 2016-08-26:
Awesome. I just kicked one off today. Fingers crossed. Thanks again for all
your work (I won't stop saying that...).

Brady Hummel

commit 5ca46b7b (2016-08-23): v4.0.13: New time-lapse method that can resume if there is a power failure or reboot. Fixes #101